SteveMacenski/slam_toolbox #63: custom lidar source: wrong number of "expected range readings"

mibcat, 2019-08-22:
Hi Steve,

I'm trying to use the software on a small home build diff drive robot. I'm using a X4 YDLIDAR and get the following message after starting the node:

```
process[slam_toolbox-1]: started with pid [14635]
[ INFO] [1566467601.227171238]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[ INFO] [1566467601.227750066]: Using plugin solver_plugins::CeresSolver
Info: clipped range threshold to be within minimum and maximum range!
Registering sensor: [Custom Described Lidar]
LaserRangeScan contains 720 range readings, expected 721
```

In the default configuration the laser /scan topic has the following header:
```
> rostopic echo --noarr /scan -n 1
header: 
  seq: 4756
  stamp: 
    secs: 1566468140
    nsecs: 604043000
  frame_id: "base_laser_link"
angle_min: -3.14159274101
angle_max: 3.14159274101
angle_increment: 0.00872664619237
time_increment: 0.000164166660397
scan_time: 0.11819999665
range_min: 0.10000000149
range_max: 16.0
ranges: "<array type: float32, length: 720>"
intensities: "<array type: float32, length: 720>"
---
```

If changing the lidar parameters e.g. min/max angle the expected number of readings differs always by one ...

SteveMacenski, 2019-08-22:
the 
> Info: clipped range threshold to be within minimum and maximum range!

isn't from my package, where is that from? Try running this without other things going. Is the issue you're reporting that or the scan return count?

mibcat, 2019-08-22:
I refer to the message:
> LaserRangeScan contains 720 range readings, expected 721

But both messages are generated from the **karto_sdk** package which is included in your repo.
Therefore I thought this issue belongs to your code - sorry for not looking into the package.xml.

Nevertheless the issue is caused be the function LaserRangeFinder::Validate() in file **slam_toolbox/lib/karto_sdk/src/Karto.cpp** line 212

SteveMacenski, 2019-08-22:
> Info: clipped range threshold to be within minimum and maximum range!
 
is not from my sdk. It looks like you're using http://docs.ros.org/jade/api/nav2d_karto/html/Sensor_8cpp_source.html which is unrelated to this. 

It looks like your laser scanner isn't correctly implementing its driver. Those min/max angles are inclusive. Its reporting 720 which is `[-pi, pi)` when the driver definition is `[-pi, pi]`. Try just changing the max_angle value from `pi` to `pi-angle_increment`. I've run this package on 5 different lidar models and haven't hit this problem before.

mibcat, 2019-08-22:
> is not from my sdk. It looks like you're using http://docs.ros.org/jade/api/nav2d_karto/html/Sensor_8cpp_source.html which is unrelated to this.

This is confusing because my workspace only contains your repo and the lidar driver.
And the code of the output message can be found at:
https://github.com/SteveMacenski/slam_toolbox/blob/3816b4b022f06a14dbde90e3285cf24384009155/lib/karto_sdk/include/karto_sdk/Karto.h#L3998

I tried your hint by subtracting the `angle_increment` value from `max_angle` but than the driver gives me only 719 readings and the karto_sdk is complaining again about one missing reading.

Do you happen to know why a scan message must have exactly "n + 1" readings.
This means that the first and the last reading are identically.

I've never realized that the scan topic has an _unequal_ number of readings ... 
But the 5:1 ratio implies a driver adjustment ;-)

SteveMacenski, 2019-08-22:
Huh that didnt show up in my search, apologies. These are some weird issues that are all from your laser message that I haven't run into before. 

The clipping error comes from reading the scans and taking the min and max ranges which _appear_ right from what you posted. 

```
range_min: 0.10000000149
range_max: 16.0
```

I don't know if its suspect or not that one of them has a bunch of sig figs and the other does not. Can you check the drivers and make sure those ranges are populated with the same datatype?

>  This means that the first and the last reading are identically.

That's not what that means. That happens to be a special case because you're using a 360 lidar, with another set its bounding the sample range `[N, M]`

mibcat, 2019-08-22:
Yes, you are right: the 360 lidar is a special case and must be fixed in the driver.
Regarding the clipping error I will check the scan topic ...

SteveMacenski, 2019-08-22:
If you have a really quick bag file (~30 seconds) of laser scans with your /tf /tf_static, I can also help debug on my side if its an issue with slam toolbox, but I suspect this is from the data. 

If you're modifying the drivers, try casting the numbers to doubles before populating the message, if you're using dynamic reconfigure its possible that is why the values look odd. 

In terms of the number of returns, because its "360" and not "359.999", I'd copy the first point to the end of the vector of points.

Here its just more important that the number of points in a scan is *consistent*, we could overload this so it'll just work but it seems like the lidar drivers should probably be updated for the next person

mibcat, 2019-08-22:
While looking for reported issues of the lidar driver I realized that I used a "dead" repository which is not maintained anymore. Now after updating the driver the number of readings issue seems to be solved ;-))

The header of the scan topic is slightly different (with full 360° setting):
```
> rostopic echo --noarr /scan -n 1
header: 
  seq: 113
  stamp: 
    secs: 1566508125
    nsecs: 152790560
  frame_id: "base_laser_link"
angle_min: -3.14159274101
angle_max: 3.14159274101
angle_increment: 0.00873878411949
time_increment: 0.00016411683464
scan_time: 0.118000000715
range_min: 0.10000000149
range_max: 16.0
ranges: "<array type: float32, length: 720>"
intensities: "<array type: float32, length: 720>"
---
```

But karto_sdk is now satisfied (I've added an additional output for verification)
```
process[slam_toolbox-1]: started with pid [3803]
[ INFO] [1566508140.586093886]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[ INFO] [1566508140.586652694]: Using plugin solver_plugins::CeresSolver
Info: clipped range threshold to be within minimum and maximum range!
Registering sensor: [Custom Described Lidar]
Kgm: number of lidar readings provided 720
Kgm: number of lidar readings expected 720
 ```
Regarding the range clipping issure I've uploaded a short measurement of my test setup:
[ydlidar.zip](https://github.com/SteveMacenski/slam_toolbox/files/3531986/ydlidar.zip)

And thanks a lot for your support !!!

SteveMacenski, 2019-08-22:
Can you check what the values of `math::Clip(rangeThreshold, GetMinimumRange(), GetMaximumRange())` are? the output, and each of the 3 things. Introspecting it might give you some ideas.

mibcat, 2019-08-22:
Oh - I can see that the readings contains values with 0.0 !!!
Is this a correct behavior of the driver ?

```
header: 
  seq: 200
  stamp: 
    secs: 1566510128
    nsecs: 902494560
  frame_id: "base_laser_link"
angle_min: -3.14159274101
angle_max: 3.14159274101
angle_increment: 0.00873878411949
time_increment: 0.00016383867478
scan_time: 0.11779999733
range_min: 0.10000000149
range_max: 16.0
ranges: [1.5360000133514404, 1.5420000553131104, 1.5440000295639038, 0.0, 1.5440000295639038, 1.5360000133514404, 1.5240000486373901, 0.0, 0.0, 0.0, 1.559000015258789, 1.565999984741211, 0.0, 1.715999960899353, 0.0, 0.0, 0.0, 0.0, 1.7289999723434448, 1.7120000123977661, 0.0, 1.7200000286102295, 1.7209999561309814, 1.7280000448226929, 1.7039999961853027, 1.690000057220459, 0.0, 1.6920000314712524, 1.6970000267028809, 1.7020000219345093, 1.7059999704360962, 1.7130000591278076, 0.0, 1.718999981880188, 1.7230000495910645, 1.7289999723434448, 1.7359999418258667, 1.7419999837875366, 0.0, 1.7489999532699585, 1.75600004196167, 1.7640000581741333, 1.7719999551773071, 1.7799999713897705, 0.0, 1.7879999876022339, 1.7960000038146973, 1.8020000457763672, 1.8109999895095825, 1.8200000524520874, 1.8359999656677246,
...
```

SteveMacenski, 2019-08-22:
Readings can be zero, sure.

mibcat, 2019-08-22:
I hope I get you right and added some outputs to function **SetRangeThreshold()** :
```
    /**
     * Sets the range threshold
     * @param rangeThreshold
     */
    inline void SetRangeThreshold(kt_double rangeThreshold)
    {
      std::cout << "Kgm: rangeThreshold " << rangeThreshold << std::endl;
      std::cout << "Kgm: GetMinimumRange() " << GetMinimumRange() << std::endl;
      std::cout << "Kgm: GetMaximumRange() " << GetMaximumRange() << std::endl;

      // make sure rangeThreshold is within laser range finder range
      m_pRangeThreshold->SetValue(math::Clip(rangeThreshold, GetMinimumRange(), GetMaximumRange()));

      std::cout << "Kgm: GetRangeThreshold " << GetRangeThreshold() << std::endl << std::endl;

      if (math::DoubleEqual(GetRangeThreshold(), rangeThreshold) == false)
      {
        std::cout << "Info: clipped range threshold to be within minimum and maximum range!" << std::endl;
      }
    }
```
Obviously the function is called 3 times:

```
process[slam_toolbox-1]: started with pid [8823]
[ INFO] [1566512286.390026909]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[ INFO] [1566512286.391210814]: Using plugin solver_plugins::CeresSolver
Kgm: rangeThreshold 12
Kgm: GetMinimumRange() 0.1
Kgm: GetMaximumRange() 80
Kgm: GetRangeThreshold 12

Kgm: rangeThreshold 12
Kgm: GetMinimumRange() 0.1
Kgm: GetMaximumRange() 16
Kgm: GetRangeThreshold 12

Kgm: rangeThreshold 20
Kgm: GetMinimumRange() 0.1
Kgm: GetMaximumRange() 16
Kgm: GetRangeThreshold 16

Info: clipped range threshold to be within minimum and maximum range!
Registering sensor: [Custom Described Lidar]
Kgm: number of lidar readings provided 720
Kgm: number of lidar readings expected 720
```

SteveMacenski, 2019-08-22:
where did 20 come from?

mibcat, 2019-08-23:
it comes from the parameter **max_laser_range** defined in the launch files.

`LaserAssistant::makeLaser()` is calling `LaserRangeFinder::SetRangeThreshold()` with this setting.

==> If `max_laser_range` setting is higher than the setting in the laser topic /scan we will get the clipping message.

output with original settings:
```
process[slam_toolbox-1]: started with pid [14149]
[ INFO] [1566543357.991477082]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[ INFO] [1566543357.992051181]: Using plugin solver_plugins::CeresSolver
Kgm: makeLaser --- 
Kgm: SetMinimumRange --- (0.1)
Kgm: SetRangeThreshold --- (12)
Kgm: GetMinimumRange() 0.1
Kgm: GetMaximumRange() 80
Kgm: GetRangeThreshold 12

Kgm: SetMaximumRange --- (16)
Kgm: SetRangeThreshold --- (12)
Kgm: GetMinimumRange() 0.1
Kgm: GetMaximumRange() 16
Kgm: GetRangeThreshold 12

Kgm: read param max_laser_range and call SetRangeThreshold(20)
Kgm: SetRangeThreshold --- (20)
Kgm: GetMinimumRange() 0.1
Kgm: GetMaximumRange() 16
Kgm: GetRangeThreshold 16

Info: clipped range threshold to be within minimum and maximum range!
Registering sensor: [Custom Described Lidar]
Kgm: number of lidar readings provided 720
Kgm: number of lidar readings expected 720
```

output with adjusted settings (max_laser_range = 16m)
```
process[slam_toolbox-1]: started with pid [14356]
[ INFO] [1566544329.713586039]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[ INFO] [1566544329.714164906]: Using plugin solver_plugins::CeresSolver
Kgm: makeLaser --- 
Kgm: SetMinimumRange --- (0.1)
Kgm: SetRangeThreshold --- (12)
Kgm: GetMinimumRange() 0.1
Kgm: GetMaximumRange() 80
Kgm: GetRangeThreshold 12

Kgm: SetMaximumRange --- (16)
Kgm: SetRangeThreshold --- (12)
Kgm: GetMinimumRange() 0.1
Kgm: GetMaximumRange() 16
Kgm: GetRangeThreshold 12

Kgm: read param max_laser_range and call SetRangeThreshold(16)
Kgm: SetRangeThreshold --- (16)
Kgm: GetMinimumRange() 0.1
Kgm: GetMaximumRange() 16
Kgm: GetRangeThreshold 16

Registering sensor: [Custom Described Lidar]
Kgm: number of lidar readings provided 720
Kgm: number of lidar readings expected 720

```

So do we need a parameter setting, when it can be extracted from the /scan topic ?

SteveMacenski, 2019-08-23:
Can you submit a PR? I can't necessarily read a 1,000 line file and know what changed ;-)

My thinking with having the maximum range as a parameter separate from the laser scan message is because professional lidars will often overstate their effective range. So the messages will say 40m, but for slam purposes they're too full of noise or sparse to be helpful so I clip them. 

I think a better set of logic as you have found would be 

```
  double max_laser_range = 25;
  nh_.getParam("max_laser_range", max_laser_range);
  if (max_laser_range > scan_.max_range)
  {
    max_laser_range = scan_.max_range
  }
  laser->SetRangeThreshold(max_laser_range);
```

mibcat, 2019-08-23:
The attached software was only intended for the case if you would like to see the output messages on your system and contains no functional relevant code changes. Just throw it away.

The code snippet you provided will fix my issue for sure (assuming the warning message is only a placeholder for a message like "your setting of the maximum laser range exceeds the  capabilities of the used lidar." ;-) ) 

Again, thank you very much for your support !
Best regards, Michael.

SteveMacenski, 2019-08-23:
Ok, can you submit a PR for it? I think it would be nice to get you on the contributor scoreboard :wink:

mibcat, 2019-08-23:
yes, I can do it ...

SteveMacenski, 2019-08-23:
#66

closed 2019-08-23 by SteveMacenski

ErroujiOussama, 2025-05-05:
@mibcat how did you manage to solve the number of readings problem cuz i have the same issue

Oyefusi-Samuel, 2025-05-10:
> [@mibcat](https://github.com/mibcat) how did you manage to solve the number of readings problem cuz i have the same issue

I'm currently facing this issues, and i dont know what to do. i increased my min range to 0.1, some fix. i am able to map but my lidar is forming maps over maps 
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-04-25-16-40-50-113533-ubuntu-1992
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [async_slam_toolbox_node-1]: process started with pid [2014]
[async_slam_toolbox_node-1] [INFO] [1745599250.885564975] [slam_toolbox]: Node using stack size 40000000
[async_slam_toolbox_node-1] [INFO] [1745599251.640286308] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[async_slam_toolbox_node-1] [INFO] [1745599251.643999308] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[async_slam_toolbox_node-1] [WARN] [1745599251.704956234] [slam_toolbox]: minimum laser range setting (0.0 m) exceeds the capabilities of the used Lidar (0.0 m)
[async_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[async_slam_toolbox_node-1] LaserRangeScan contains 355 range readings, expected 354
[async_slam_toolbox_node-1] LaserRangeScan contains 355 range readings, expected 354
[async_slam_toolbox_node-1] LaserRangeScan contains 353 range readings, expected 354

[image]

mibcat, 2025-05-12:
Oh -  this is a long time ago - and I'm not working with that lidar since several years ...

But as far as I remember it was an issue of an old outdated repository I was using (see [comment](https://github.com/SteveMacenski/slam_toolbox/issues/63#issuecomment-524085973).

**ydlidar** (outdated) vs **ydlidar_ros** but in the year 2019 !

Oyefusi-Samuel, 2025-05-12:
Yeah @mibcat , using YDLIDAR with ROS2 has been quite challenging — that’s actually what pushed me to contribute to a more updated ROS2-compatible node setup. I’ve been working on both the client and driver nodes here: https://github.com/Oyefusi-Samuel/ydlidar_ros2_driver-master.git.

From my experience so far, it seems the published LaserScan data still needs a bit of preprocessing (like reordering, binning, or resampling the ranges) before it's fully compatible with SLAM Toolbox or NAV2 @SteveMacenski . These systems tend to be strict about scan structure, especially the number of points, angle increments, and range noise.

It’s definitely been a tough one to crack, but I’m committed to solving this and making the driver more plug-and-play for ROS2-based SLAM and navigation workflows. Happy to collaborate if anyone else is facing similar issues or has ideas.

mibcat, 2025-05-13:
I think the problem is related to the full 360° range while a min and max angle must be provided.
If they are equal to -PI and +PI  you will get the N+1 warning because these two "end point" are identical and somehow not handled properly.

And I also used a local forked repository for driver enhancement to have a ROS service for starting and stopping the motor because I wanted to extend the operation time of the scanner if the robot is "in idle".

Oyefusi-Samuel, 2025-05-13:
Many thanks @mibcat. Please can you share the repo of the driver enhancement. I hope it’s in ros2

mibcat, 2025-05-13:
Unfortunately not, it's ROS1 from 2019
As said I'm not working with that lidar since the last years so I stopped all work on that repo.

Oyefusi-Samuel, 2025-05-13:
[like]  Oluwakorede Oyefusi reacted to your message:

Oyefusi-Samuel, 2025-06-09:
> So I fork the original driver, and by counting the first 30 data's data counts, and finding the mode of these data counts, try to find out the validation value that slam_toolbox may use, and tuncate or fill the data to the same number of data counts to avoid this problem, the fix of the node at the link below, I also use @Oyefusi-Samuel<https://github.com/Oyefusi-Samuel> 's solution to solve the problem of QoS policy.
>
> https://github.com/Bob-YsPan/ydlidar_ros2_driver/tree/humble
>
> I change the logic from tuncate and fill data to find the nearest point, and new driver update to my repo!

[like]  Oluwakorede Oyefusi reacted to your message: