SteveMacenski/slam_toolbox #73: mapping failed (due to localization lost ?)

mibcat, 2019-08-28:
Hi!
I'm running into trouble during the mapping mode: 
als long as the robot is moving straight forward or backward the mapping is fine but as soon as I add a rotation it seems that the localization gets more and more lost and therefore the mapping process will generate unusable maps.

I assume that I'm using a improper configuration and need some help ...

Here some information about the system:
- diff drive robot
- low cost YDLIDAR X4 
  - mounted upside (rotation of scan unit is clockwise)
  - usage of latest software version [V1.4.1](https://github.com/YDLIDAR/ydlidar_ros)
  - /scan topic header:
```
header: 
  seq: 192
  stamp: 
    secs: 1566986866
    nsecs:   8921560
  frame_id: "laser_link"
angle_min: -1.65806281567
angle_max: 1.65806281567
angle_increment: 0.00874967221171
time_increment: 0.00030976254493
scan_time: 0.117399998009
range_min: 0.10000000149
range_max: 16.0
ranges: "<array type: float32, length: 380>"
intensities: "<array type: float32, length: 380>"
---
```
- tf tree:
[image]
- scans seems to be valid related to distance and angle in relation to **laser_link**:
[image]

- using latest version of slam_toolbox (6281c8c04268d65386a2ba6f3cb1e7e8ff968555)
  - starting `roslaunch slam_toolbox online_sync.launch`
  - adjusting following parameters in **config/mapper_params_online_sync.yaml**:
    - base_frame: base_link
    - laser_frame: laser_link
    - max_laser_range: 16.0
  - running the node will give no error or warning messages:
```
process[sync_slam_toolbox_node-1]: started with pid [5404]
[ INFO] [1566983484.418758725]: Node using stack size 40000000
[ INFO] [1566983485.658208287]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[ INFO] [1566983485.658327096]: Using plugin solver_plugins::CeresSolver
Registering sensor: [Custom Described Lidar]
[ INFO] [1566983491.127153328]: Creating 1 swatches
```

mibcat, 2019-08-28:
Here some screen shots during mapping:

Direct after starting the node:
[image]

drive only forward:
[image]

turn right and dive a little bit forward:
[image]

turn left:
[image]

drive straight backward:
[image]

turning left (180°):
[image]

In this very small and easy environment the resulting map looks not so bad but in a more complex environment the maps will be "destroyed" very easily due to the lost localization ...

mibcat, 2019-08-28:
> Between all of your movements, I’m not seeing the map update at all, so you haven’t moved enough for this package to have triggered an update.

How do you define a map update ? I thought building up a wall or turning unknown space into free space is a "map update".

Regarding the odometry: the engine controller which is creating the /odom topic and the tf transformation `odom` to `base_link` is also home made. Of course, errors can still be found in this firmware. But when using gmapping the same odometry seems not to have such a worse impact to the mapping result.

Regarding the lidar: I think the rotation direction is not configurable - for that I have to mount the lidar upside-down (which is currently not so easy). But what is the influence of the spinning direction ? As said the /scan topic seems to have the objects in the right distance and orientation if displayed in RViz.

My URDF may not be accurate enough (+- 5mm) ...

Nevertheless I'll prepare a bag file with the current setup ...

mibcat, 2019-08-28:
OK- I'll have a deeper look into the driver ...
and here is a bag file: [mapping.zip](https://github.com/SteveMacenski/slam_toolbox/files/3552226/mapping.zip)

mibcat, 2019-08-28:
yes - also a home made device ;-)
But it's not mounted so far ...

mibcat, 2019-08-28:
yes, I agree - this cheap lidar is not best device (but the budget is limited and doesn't include a Sick or Omron lidar unfortunately ;-))

Regarding the scan time: although I'm working with lidars quite a while I'm not aware of the exact definitions of all /scan fields. From the documentation [sensor_msg/LaserScan](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/LaserScan.html)
```
float32 time_increment   # time between measurements [seconds] - if your scanner
                         # is moving, this will be used in interpolating position
                         # of 3d points
float32 scan_time        # time between scans [seconds]
```
For me the fields have the following meaning (and they are also be filled by the driver in this meaning)
**time_increment**: time between each reading in the **ranges** array
**scan_time**: total time duration of the scan (=number of beams * time_increment)

Is this correct ? "time between scans" is not unambiguous.

mibcat, 2019-08-28:
Tried a quick fix in my controller firmware and set the /odom and tf update rate to round about 35 Hz.

During the first mapping I thought that this change makes the mapping result better. It runs quite good for a while but suddenly (maybe after a strong turn) the map starts to degenerate due to misalignment. Stupidly, the cable connection has broken while driving and the ros bag get lost ;-(((

After restart I was not able to get the same good results - strange. Have to check it tomorrow ...

mibcat, 2019-08-28:
just an other idea: the bag was recorded on my desktop and not on the robot (due to limited disk capacity of the raspi). Could this possibly lead to the scan effects you noticed ?

mibcat, 2019-08-28:
and once more: my engine controller is updating the odom topic and tf transformation only at 10 Hz - is that enough ?

SteveMacenski, 2019-08-29:
I'd also try running this in simulation with your URDF and make sure its working as a process of elimination

mibcat, 2019-08-29:
Today I was busy with the lidar driver. I tried to understand how the scan topic is displayed in RViz while the robot is moving. To eliminate all other error sources I only looked at /scan in it's own frame.

From my understanding it's clear that the shapes of the objects in the scan (e.g. a flat wall) must be somehow distorted due to the robot movement because the lidar driver does not know anything about it's movement in the environment.

This is the scan at rest (here is a small gap in the wall which is not a scan distortion):
[image]

Turning the robot at ~1,7 rad/s there is a kink:
[image]
(For the scan duration of about 120ms this rotation speed is equal to about 12 degrees which is in the range of the visible kink angle)

one scan topic later there is also a step:
[image]

It seems that this step/kink is the point where the scan starts and begins (because it's a 360 degree lidar). This point seems to be stable (at least the bag file with a solely rotation which I'm currently lookin on). But I can't see a significant `"swiping back and forth around the 0 degree mark"`  as you described. 
This distortion is systematic and can only be minimized with higher scan rates like it is done in more expensive lidars.

Do you agree or am I wrong ?

SteveMacenski, 2019-08-29:
I agree, but it should be able to be resolved. This is an issue with poorly developed software/firmware not intrinsic about a cheapo lidar. Its a timing issue that they're seemingly just not addressing

mibcat, 2019-08-29:
I'm confused: but to solve this issue the driver / firmware must know the robot's velocities. Correct me if I'm wrong but I don't think this is implemented in the firmware of other high price lidars ...

The other way round: if there is a possibility to lower the scan rate of a well working high price lidar it would be interested to see how its scan topic behaves ...

SteveMacenski, 2019-08-29:
The lidar doesn't need to know anything about the robot velocities if the timestamps on the laser data are correct. 

If the robot is moving at a random high speed, a laser scan cannot be approximated as a static snapshot of the world with no aliasing. The measurements at 10 deg and 60 deg could be meters apart. However that is OK if the *times* that the measurements were recorded are correct, since you can back out that my timestamp for the message is `N` and point 74 was taken `M` cycles into it, and you can back out the approximate time that measurement was taken and transform the point into that frame. That's what the laser projector inside of RVIZ is doing, takes the scan, projects each into a pointcloud by the appropriate time, and display it. If RVIZ is incorrectly displaying the measurements in laser frame, its because the laser projector isn't able to compensate because something in that isn't quite right. Or the data is total trash. One or the other :)

mibcat, 2019-08-29:
Sorry, but my confusion rises more and more. I think I need a coaching like "laser scanner for dummies" ;-))

How can I compensate the scan data in the one and only laser frame? 
A lidar can only give me an angle, distance and a time stamp for each point.

If the laser projector in RViz should compensate a movement of the laser frame than it needs the knowledge about that  laser frame movement or am I on the wrong track ????
(If I provide only the /scan topic to RViz and nothing else than a compensation should not be possible)

SteveMacenski, 2019-08-29:
the transform is map->odom->base->laser, many of those transforms aren't static. So those transforms are stamped at a time. map->odom at t=1 and t=1.1 are different, if you request them. 

With that said this package doesn't do that. But RVIZ does and if RVIZ is displaying incorrectly, that's a big red flag that there's alot of things wrong here that this may be affected by

mibcat, 2019-08-29:
Yes I agree: if I have those transformation (especially odom->base) than it's possible by RViz to compensate the scan topic because the movement of the laser frame is known by tf.

But in this case (without tf)
>(If I provide only the /scan topic to RViz and nothing else than a compensation should not be possible)

it should not be possible ...

SteveMacenski, 2019-08-29:
You are correct. If you are in laser frame and the laser data is doing that, that's the raw aliasing. If you're in odometry frame and its doing it, then there's some timing/encoder issue such that the aliasing isn't being removed. 

Off hand I haven't done that low level of testing with my RPlidar to know if that doesn't also have the same issue. I'm lucky enough to have a SICK laser at home

mibcat, 2019-08-29:
Ok - than I understood it too. And thanks for your patience. ;-)

Because the "raw" laser data says nothing from where the issue in the compete system comes from:
- bad laser time stamps _and/or_
- poor odom->base->.. > laser_link transformation

I have to rethink about how to solve the riddle 

> I'm lucky enough to have a SICK laser at home

unfortunately out of range ... ;-)

SteveMacenski, 2019-08-29:
A suggestion, though annoying:
- Grab the laser data from the SDK itself
- On data reception, transform w/ timestamp
- batch send out the laser scan

That might be able to cancel out a fair amount of it

SteveMacenski, 2019-08-29:
To some amount I don't necessarily know that'll fix your issue. I'm a little surprised you have this issue considering a SICK is only 15hz, and yours is 10hz. That's not some massive difference. I have a feeling like its not reporting data in the correct angle and what you're seeing when you turn is those artifacts.

You have the -1.6-1.6 range, where did that come from? Is there some culling of points maybe that's wrong there?

mibcat, 2019-08-30:
The mounting position of the lidar is not really optimal, the +/- 1.6 range is the "free" field of view in front of the robot. The rest is blocked by the casing.
Maybe for a test I can mount the lidar on top to have full 360°.

SteveMacenski, 2019-08-30:
I wonder if you request from the lidar certain ranges, it does weird things.

SteveMacenski, 2019-08-31:
We're more than welcome to continue this discussion, but if you're not working on it anymore and its not directly related to this project, I'm going to close the ticket.

If you get back on it, we can reopen

closed 2019-08-31 by SteveMacenski