SteveMacenski/slam_toolbox #77: roslaunch error

guorun198959, 2019-09-01:
hello,
     I'm very interested in ur package, and i can compile it successfully. but when i roslaunch slam_toolbox online_async.launch ,each launch file I tried.The error is 
[slam_toolbox-2] process has died [pid 4758, exit code -11, cmd /home/gary/slam_toolbox_ws/install/lib/slam_toolbox/slam_toolbox __name:=slam_toolbox __log:=/home/gary/.ros/log/aa16d8b8-cc9b-11e9-980a-9cb6d06ccefb/slam_toolbox-2.log].
log file: /home/gary/.ros/log/aa16d8b8-cc9b-11e9-980a-9cb6d06ccefb/slam_toolbox-2*.log

   I don't know what's going on? can u help me to work it out,  thank u very much!

jwdinius, 2019-09-01:
Exit code -11 indicates a SIGSEGV (segmentation fault), meaning an invalid access to storage was requested.  The README suggests using `roslaunch slam_toolbox online_sync.launch`, not the command you use above.  Have you tried this?

SteveMacenski, 2019-09-01:
In addition, make sure that your parameters for stack resize are valid for your platform. I think my default is 40MB in async (vs 8 default for C++ code on modern hardware, needed for serialization/deserialization) 

What tag, what OS, what distro, etc. If possible run with gdb and give me a traceback. `-g` flag and in the launch file put `launch-prefix="gdb"` to start the debug process. I've tested against Melodic / 18.04 on sync/async/localization before my last big refactor PR without an issue

guorun198959, 2019-09-02:
jwdinius and SteveMacenski Thank u guys. my problem is solved. it's my ros's error. I changed another pc. problem is done.

SteveMacenski, 2019-09-02:
Awesome, glad you find the bug. Closing the ticket then!

closed 2019-09-02 by SteveMacenski