SteveMacenski/slam_toolbox #80: catkin_make fails to build slam_toolbox in 16.04 docker image

ubtjoe, 2019-09-05:
Thanks for the great package!

When trying to build the toolbox during a docker image build, I receive the following output (and error):

```
Creating symlink "/home/slam/catkin_ws/src/CMakeLists.txt" pointing to "/opt/ros/kinetic/share/catkin
/cmake/toplevel.cmake"                                                                               
-- The C compiler identification is GNU 5.4.0                                                        
-- The CXX compiler identification is GNU 5.4.0                                                      
-- Check for working C compiler: /usr/bin/cc                                                         
-- Check for working C compiler: /usr/bin/cc -- works                                                
-- Detecting C compiler ABI info                                                                     
-- Detecting C compiler ABI info - done                                                              
-- Detecting C compile features                                                                      
-- Detecting C compile features - done                                                               
-- Check for working CXX compiler: /usr/bin/c++                                                      
-- Check for working CXX compiler: /usr/bin/c++ -- works                                             
-- Detecting CXX compiler ABI info                                                                   
-- Detecting CXX compiler ABI info - done                                                            
-- Detecting CXX compile features                                                                    
-- Detecting CXX compile features - done                                                             
-- Using CATKIN_DEVEL_PREFIX: /home/slam/catkin_ws/devel                                             
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic                                                         
-- This workspace overlays: /opt/ros/kinetic                                                         
-- Found PythonInterp: /usr/bin/python (found version "2.7.12")                                      
-- Using PYTHON_EXECUTABLE: /usr/bin/python                                                          
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/slam/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                 
-- ~~  traversing 1 packages in topological order:
-- ~~  - slam_toolbox
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'slam_toolbox'
-- ==> add_subdirectory(slam_toolbox)
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   serialization
--   filesystem
--   thread
--   chrono
--   date_time
--   atomic
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
CMake Error at slam_toolbox/CMakeLists.txt:37 (find_package):
  By not providing "FindCSparse.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "CSparse", but
  CMake did not find one.

  Could not find a package configuration file provided by "CSparse" with any
  of the following names:

    CSparseConfig.cmake
    csparse-config.cmake

  Add the installation prefix of "CSparse" to CMAKE_PREFIX_PATH or set
  "CSparse_DIR" to a directory containing one of the above files.  If
  "CSparse" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/slam/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/slam/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
Base path: /home/slam/catkin_ws
Source space: /home/slam/catkin_ws/src
Build space: /home/slam/catkin_ws/build
Devel space: /home/slam/catkin_ws/devel
Install space: /home/slam/catkin_ws/install
####
#### Running command: "cmake /home/slam/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/slam/catkin_ws/deve
l -DCMAKE_INSTALL_PREFIX=/home/slam/catkin_ws/install -G Unix Makefiles" in "/home/slam/catkin_ws/build"
The command '/bin/bash -c source /opt/ros/kinetic/setup.bash     && cd /home/$username/catkin_ws/src
    && catkin_init_workspace     && cd ..     && catkin_make -j1' returned a non-zero code: 1
```

I have the following dockerfile:

```
FROM ros:kinetic-ros-base-xenial

ARG username=slam
ARG groupid=1000
ARG userid=1000

# USE BASH
SHELL ["/bin/bash", "-c"]

# RUN LINE BELOW TO REMOVE debconf ERRORS (MUST RUN BEFORE ANY apt-get CALLS)
RUN echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections

RUN apt-get update && apt-get install -y --no-install-recommends apt-utils
RUN apt-get install -y ros-kinetic-desktop-full

# setup user env at the end
# -m option creates a fake writable home folder
RUN adduser --disabled-password --gecos '' $username
RUN adduser $username sudo
RUN adduser $username dialout
RUN echo '%sudo ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers

# slam_toolbox
RUN mkdir -p /home/$username/catkin_ws/src \
    && chmod a+rwx /home/$username/catkin_ws \
    && chmod a+rwx /home/$username/catkin_ws/src
RUN cd /home/$username/catkin_ws/src && git clone https://github.com/SteveMacenski/slam_toolbox.git
RUN source /opt/ros/kinetic/setup.bash \
    && cd /home/$username/catkin_ws \
    && rosdep update \
    && rosdep install -q -y -r --from-paths src --ignore-src --rosdistro=kinetic -y

USER $username
RUN echo "source /opt/ros/kinetic/setup.bash" >> /home/$username/.bashrc

RUN source /opt/ros/kinetic/setup.bash \
    && cd /home/$username/catkin_ws/src \
    && catkin_init_workspace \
    && cd .. \
    && catkin_make -j1
RUN echo "source ~/catkin_ws/devel/setup.bash" >> /home/$username/.bashrc

WORKDIR /home/$username
```

SteveMacenski, 2019-09-05:
Try using the kinetic branch, you’re cloning melodic. 

Also PRs are encouraged. I deploy using snaps and debians and building neither has had this particular issue.

SteveMacenski, 2019-09-05:
You also rosdep install with the don’t error on failure flag. Make sure you’re actually installing all the dependencies correctly and theyre up to date

SteveMacenski, 2019-09-05:
Hi,

I just spent a couple hours over cooking dinner looking into it, with this docker file I found that every time I removed another `find_package()`, another one complained (remove CSparse, then its G2O, then its Cholmod, ...) so I think its a path issue with the docker container finding the make files, which is also odd given that G2O and Cholmod I provide the cmake find file and those have never caused me problems.

This leads me to believe that there's some paths not quite hooked up right and/or the build system isn't reading right. I switched to catkin tools (which is better as far as I can tell in all ways) and worked out fine.

A pared down docker file which seems to do the trick:

(btw there's a bunch of stuff in the docker file that isnt necessary. It doesnt hurt but it makes your images much larger than required + build time will take multiple times the length to build)

```
FROM ros:kinetic-ros-base-xenial

# USE BASH
SHELL ["/bin/bash", "-c"]

RUN apt-get update && apt-get upgrade -y && apt-get install -y --no-install-recommends apt-utils

# slam_toolbox
RUN mkdir -p catkin_ws/src
#    && chmod a+rwx /home/$username/catkin_ws \
#    && chmod a+rwx /home/$username/catkin_ws/src
RUN echo "hi"
RUN cd catkin_ws/src && git clone https://github.com/SteveMacenski/slam_toolbox.git
RUN source /opt/ros/kinetic/setup.bash \
    && cd catkin_ws \
    && rosdep update \
    && rosdep install -y -r --from-paths src --ignore-src --rosdistro=kinetic -y

RUN apt install python-catkin-tools -y
RUN source /opt/ros/kinetic/setup.bash \ 
    && cd catkin_ws/src \
    && catkin_init_workspace \
    && cd .. \
    && catkin build

WORKDIR /home/$username
```

SteveMacenski, 2019-09-05:
Have now confirmed that failure is from the changes between kinetic and melodic

```
FROM ros:melodic-ros-base-bionic

# USE BASH
SHELL ["/bin/bash", "-c"]

RUN apt-get update && apt-get upgrade -y && apt-get install -y --no-install-recommends apt-utils

# slam_toolbox
RUN mkdir -p catkin_ws/src
RUN cd catkin_ws/src && git clone https://github.com/SteveMacenski/slam_toolbox.git
RUN source /opt/ros/melodic/setup.bash \
    && cd catkin_ws \
    && rosdep update \
    && rosdep install -y -r --from-paths src --ignore-src --rosdistro=melodic -y

RUN apt install python-catkin-tools -y
RUN source /opt/ros/melodic/setup.bash \ 
    && cd catkin_ws/src \
    && catkin_init_workspace \
    && cd .. \
    && catkin build                   
```

This Melodic docker file works 100%

ubtjoe, 2019-09-05:
@SteveMacenski Awesome, thanks!  I am building the image now and will give feedback if there is anything worthwhile.

ubtjoe, 2019-09-06:
Build process runs smoothly.  Closing the issue.  Thanks @SteveMacenski !

closed 2019-09-06 by ubtjoe

denesh-botsync, 2021-05-27:
I can concur with Steve's point about catkin tools. I faced a similar issue when trying to build the kinetic_devel branch of slam_toolbox. I was using "catkin_make"  and I was getting this error: 

CMake Error at slam_toolbox/CMakeLists.txt:37 (find_package):
  By not providing "FindCSparse.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "CSparse", but
  CMake did not find one.

  Could not find a package configuration file provided by "CSparse" with any
  of the following names:

    CSparseConfig.cmake
    csparse-config.cmake
    
As soon as I switched to "catkin build" the build worked. I did not change anything else. 

System Settings: 
Operating System: Ubuntu 16.04 
Slam Toolbox branch: Kinetic_devel