SteveMacenski/slam_toolbox #81: cannot locali

guorun198959, 2019-09-05:
hello,
        sorry to bother u again.I tried the localization.launch file,and I changed the param "mode: localization" in .config file.but it doesn't work.it always do mapping without localization. my os is ubuntu16.04.

SteveMacenski, 2019-09-05:
Please edit the ticket title and description to accurately and descriptively address what youre seeing and what youre expecting. 

Pictures/figures are good

SteveMacenski, 2019-09-05:
Also make sure you carefully read the readme on localization. I’m going to make a really good guess that its working correctly

guorun198959, 2019-09-07:
my goal is to use ur localization.launch to locate my robot in the map.
here is my understanding:
first, i need to building a map use the online_sync.launch,which is
<launch>

  <node pkg="slam_toolbox" type="sync_slam_toolbox_node" name="sync_slam_toolbox_node" output="screen">
    <rosparam command="load" file="$(find slam_toolbox)/config/mapper_params_online_sync.yaml" />
  </node>

</launch>
and the mapper_params_online_sync.yaml is 

# link_match_minimum_response_fine: 0.4 
# link_scan_maximum_distance: 20.0
# loop_search_maximum_distance: 50.0
# distance_variance_penalty: 1.8             
# angle_variance_penalty: 0.20    
# minimum_angle_penalty: 1.6 #0.9

# Plugin params
solver_plugin: solver_plugins::CeresSolver
ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
ceres_preconditioner: SCHUR_JACOBI
ceres_trust_strategy: LEVENBERG_MARQUARDT
ceres_dogleg_type: TRADITIONAL_DOGLEG
ceres_loss_function: None

# ROS Parameters
odom_frame: odom
map_frame: map
base_frame: base_footprint
mode: mapping #localization

# if you'd like to immediately start continuing a map at a given pose
# or at the dock, but they are mutually exclusive, if pose is given
# will use pose
#map_file_name: test_steve
#map_start_pose: [0.0, 0.0, 0.0]
#map_start_at_dock: true

debug_logging: false
throttle_scans: 1
transform_publish_period: 0.02 #if 0 never publishes odometry
map_update_interval: 5.0
resolution: 0.05
max_laser_range: 20.0 #for rastering images
minimum_time_interval: 0.5
transform_timeout: 0.2
tf_buffer_duration: 10.
stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps

# General Parameters
use_scan_matching: true
use_scan_barycenter: true
minimum_travel_distance: 0.5
minimum_travel_heading: 0.5
scan_buffer_size: 10
scan_buffer_maximum_scan_distance: 10
link_match_minimum_response_fine: 0.1  
link_scan_maximum_distance: 1.5
loop_search_maximum_distance: 3.0
do_loop_closing: true 
loop_match_minimum_chain_size: 10           
loop_match_maximum_variance_coarse: 3.0  
loop_match_minimum_response_coarse: 0.35    
loop_match_minimum_response_fine: 0.45

# Correlation Parameters - Correlation Parameters
correlation_search_space_dimension: 0.5
correlation_search_space_resolution: 0.01
correlation_search_space_smear_deviation: 0.1 

# Correlation Parameters - Loop Closure Parameters
loop_search_space_dimension: 8.0
loop_search_space_resolution: 0.05
loop_search_space_smear_deviation: 0.03

# Scan Matcher Parameters
distance_variance_penalty: 0.5      
angle_variance_penalty: 1.0    

fine_search_angle_offset: 0.00349     
coarse_search_angle_offset: 0.349   
coarse_angle_resolution: 0.0349        
minimum_angle_penalty: 0.9
minimum_distance_penalty: 0.5
use_response_expansion: true

after launched it.the result is
[image]
and the tf tree is
[image]

right now, i need to generate a serialized map,but i don't know how to do it?
secondly,if i had a serialized map , eg:the name is "test_localiztion".
so I will launch the localization.launch file, which is
<launch>
  <node pkg="slam_toolbox" type="localization_slam_toolbox_node" name="localization_slam_toolbox_node" output="screen">
    <rosparam command="load" file="$(find slam_toolbox)/config/mapper_params_localization.yaml" />
  </node>
</launch>

and the mapper_params_localization.yaml is

# link_match_minimum_response_fine: 0.4 
# link_scan_maximum_distance: 20.0
# loop_search_maximum_distance: 50.0
# distance_variance_penalty: 1.8             
# angle_variance_penalty: 0.20    
# minimum_angle_penalty: 1.6 #0.9

# Plugin params
solver_plugin: solver_plugins::CeresSolver
ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
ceres_preconditioner: SCHUR_JACOBI
ceres_trust_strategy: LEVENBERG_MARQUARDT
ceres_dogleg_type: TRADITIONAL_DOGLEG
ceres_loss_function: HuberLoss

# ROS Parameters
odom_frame: odom
map_frame: map
base_frame: base_link #base_footprint
mode: localization #localization 

# if you'd like to start localizing on bringup in a map and pose
map_file_name: /file_path/test_localiztion
map_start_pose: [0.0, 0.0, 0.0]

debug_logging: false
throttle_scans: 1
transform_publish_period: 0.02 #if 0 never publishes odometry
map_update_interval: 5.0
resolution: 0.05
max_laser_range: 20.0 #for rastering images
minimum_time_interval: 0.5
transform_timeout: 0.2
tf_buffer_duration: 10.
stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps

# General Parameters
use_scan_matching: true
use_scan_barycenter: true
minimum_travel_distance: 0.5
minimum_travel_heading: 0.5
scan_buffer_size: 10
scan_buffer_maximum_scan_distance: 10
link_match_minimum_response_fine: 0.1  
link_scan_maximum_distance: 1.5
loop_search_maximum_distance: 3.0
do_loop_closing: true 
loop_match_minimum_chain_size: 10           
loop_match_maximum_variance_coarse: 3.0  
loop_match_minimum_response_coarse: 0.35    
loop_match_minimum_response_fine: 0.45

# Correlation Parameters - Correlation Parameters
correlation_search_space_dimension: 0.5
correlation_search_space_resolution: 0.01
correlation_search_space_smear_deviation: 0.1 

# Correlation Parameters - Loop Closure Parameters
loop_search_space_dimension: 8.0
loop_search_space_resolution: 0.05
loop_search_space_smear_deviation: 0.03

# Scan Matcher Parameters
distance_variance_penalty: 0.5      
angle_variance_penalty: 1.0    

fine_search_angle_offset: 0.00349     
coarse_search_angle_offset: 0.349   
coarse_angle_resolution: 0.0349        
minimum_angle_penalty: 0.9
minimum_distance_penalty: 0.5
use_response_expansion: true

The above is what i know, is it right? if not, please let me know how to correct it . thank u very much!

SteveMacenski, 2019-09-07:
That is correct. You may use the RVIZ plugin to serialize the map, there's an exposed service `serialize_map` to serialize the map. Then if you use the config files you have shown it will try to localize in the map file given at the position given.

You bring up a good point that I don't actually explicitly describe the exposed services / topics in the README. If you'd like to update those that would be great. 

You renamed the ticket to "cannot localize just use localization.launch" but that doesn't describe your issue, it sounds like your problem is that you don't know how to serialize the map. Have you tried to use localization.launch?

guorun198959, 2019-09-08:
I used the RVIZ plugin to serialize the map, and generate four files: test.data, test.pgm,test.posegraph, test.yaml. which file should I put to the map_file_name: /file_path/test.* in file  mapper_params_localization.yaml  ? actually I tried every one. but all failed,and report the messages:
serialization::Read: Failed to open requested file: /home/gary/slam_toolbox_ws/src/slam_toolbox/config/test.yaml.
[ERROR] [1567929437.046213852]: DeserializePoseGraph: Failed to read file: /home/gary/slam_toolbox_ws/src/slam_toolbox/config/test.yaml. need some help , thank u.

SteveMacenski, 2019-09-09:
You do not need to include the file extension. `./` is not a valid path.

guorun198959, 2019-09-13:
I corrected the file and the path,but it also cannot locate. my steps are belows:
1.roslaunch a simulation enviroment.
[image]
2.roslaunch slam_toolbox localization.launch
<launch>

  <node pkg="slam_toolbox" type="localization_slam_toolbox_node" name="localization_slam_toolbox_node" output="screen">
    <rosparam command="load" file="$(find slam_toolbox)/config/mapper_params_localization.yaml" />
  </node>

</launch>
mapper_params_localization.yaml:

//# link_match_minimum_response_fine: 0.4 
//# link_scan_maximum_distance: 20.0
//# loop_search_maximum_distance: 50.0
//# distance_variance_penalty: 1.8             
//# angle_variance_penalty: 0.20    
//# minimum_angle_penalty: 1.6 #0.9

//# Plugin params
solver_plugin: solver_plugins::CeresSolver
ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
ceres_preconditioner: SCHUR_JACOBI
ceres_trust_strategy: LEVENBERG_MARQUARDT
ceres_dogleg_type: TRADITIONAL_DOGLEG
ceres_loss_function: None

//# ROS Parameters
odom_frame: odom
map_frame: map
base_frame: base_link #base_footprint
mode: localization #localization

//# if you'd like to start localizing on bringup in a map and pose
map_file_name: gary_test
map_start_pose: [0.0, 0.0, 0.0]

debug_logging: false
throttle_scans: 1
transform_publish_period: 0.02 #if 0 never publishes odometry
map_update_interval: 5.0
resolution: 0.05
max_laser_range: 20.0 #for rastering images
minimum_time_interval: 0.5
transform_timeout: 0.2
tf_buffer_duration: 30.
stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps

//# General Parameters
use_scan_matching: true
use_scan_barycenter: true
minimum_travel_distance: 0.5
minimum_travel_heading: 0.5
scan_buffer_size: 10
scan_buffer_maximum_scan_distance: 10
link_match_minimum_response_fine: 0.1  
link_scan_maximum_distance: 1.5
loop_search_maximum_distance: 3.0
do_loop_closing: true 
loop_match_minimum_chain_size: 10           
loop_match_maximum_variance_coarse: 3.0  
loop_match_minimum_response_coarse: 0.35    
loop_match_minimum_response_fine: 0.45

//# Correlation Parameters - Correlation Parameters
correlation_search_space_dimension: 0.5
correlation_search_space_resolution: 0.01
correlation_search_space_smear_deviation: 0.1 

//# Correlation Parameters - Loop Closure Parameters
loop_search_space_dimension: 8.0
loop_search_space_resolution: 0.05
loop_search_space_smear_deviation: 0.03

//# Scan Matcher Parameters
distance_variance_penalty: 0.5      
angle_variance_penalty: 1.0    

fine_search_angle_offset: 0.00349     
coarse_search_angle_offset: 0.349   
coarse_angle_resolution: 0.0349        
minimum_angle_penalty: 0.9
minimum_distance_penalty: 0.5
use_response_expansion: true

the result is like this:
[image]
[image]
[image]
I don't know why it cannot locate itself. can u help me to work it out, thank u so much!

SteveMacenski, 2019-09-13:
You're setting your initial pose to 0,0,0. Is that where the robot is? Just like AMCL, you need to give it a reasonable estimation of the actual pose

guorun198959, 2019-09-13:
My start point is 0,0,0. and can this project solve the robot kidnap?

SteveMacenski, 2019-09-14:
No. That is not in the scope for most industrial applications.

closed 2019-09-19 by SteveMacenski

maxxlife, 2022-06-08:
How you solved it? @guorun198959  I have the same problem. @SteveMacenski Why I can't read the file?
[ERROR] [1654719646.414387018]: serialization::Read: Failed to open requested file: /home/maxxlife/Documents/abot_files_maps/maps/june_8/small_room_with_boxes_2_slam_tb_rviz.data.
[ERROR] [1654719646.414441999]: DeserializePoseGraph: Failed to read file: /home/maxxlife/Documents/abot_files_maps/maps/june_8/small_room_with_boxes_2_slam_tb_rviz.data.

Files exists...