SteveMacenski/slam_toolbox #91: Disappearing map?

msadowski, 2019-09-10:
Hi Steve!

Thanks for the great package! I'm currently running some tests and it's the first SLAM package I used that works very well out of the box.

It's very well possible that I've missed some parameter or something else in the docs but is the disappearing map the valid behaviour of slam_toolbox?

Here is how it looks like for me in the simulation (I believe I've seen it happened on real robot too):

[image]

The map shown in the window above is the one appearing on map topic.

Any feedback you might have would be highly appreciated!

SteveMacenski, 2019-09-10:
In localization mode, yes that’s valid behavior. What’s your launch file/config?

The method I implement for localization keeps a rolling window of scans to process and localize against so that you’re localized against both the map itself as well as the local set of features of the real environment rather than throwing out the current state as “junk” to be removed from processing the core map. This rolling window will expire and then the scans and their constraints/nodes will depreciate and disappear from the map. 

But this is unique of localization mode. Slam modes should not remove elements from the graph. 

And thanks! I’ve worked very hard to make that happen :-)

msadowski, 2019-09-11:
Thanks a lot for a quick answer and an explanation! When I wrote the issue I was indeed running the `localization.launch` file. Switching to online_async.launch file the mapping is working well!

I'll be working with slam_toolbox for the next couple of days. If you need any help with then let me know, I'm happy to help! 

I found the gifs to be really good in sharing results (especially in RVIZ). I found peek to be very useful tool for that: https://github.com/phw/peek

Closing this. Once again thanks for help!

closed 2019-09-11 by msadowski

SteveMacenski, 2019-09-12:
@msadowski side note: if you make a nice gif of you mapping a space, I'd love to include it in the readme with a shoutout. I'm always looking for good demonstrations of my work in use.

msadowski, 2019-09-12:
@SteveMacenski It's quite likely that I'll write a blog post about slam_toolbox together with [Ross Robotics](https://rossrobotics.com/) in the near future. When I do I'll make sure to make some nice gifs/videos that you could also use.

msadowski, 2019-09-18:
@SteveMacenski I put together a small article on our first trials with the package! You can find it here: https://www.robosynthesis.com/post/hands-on-with-slam_toolbox . If you find any of the images or gifs usable for you then feel free to use them!