SteveMacenski/slam_toolbox #93: Question about integrating IMU and Odometry sources

joekeo, 2019-09-11:
Hello,

This is a great package, and we are learning how to use it at its full extent. 

We want to know which are the best practices (and if there are examples) on using IMUs and odometry sources (wheels or realsens T265). Is it straight up like remapping the topic to 'shape it' for slam toolbox? Or do we need to use an external package like robot localisation to do it effectively.

Many thanks for the help :)

Pepe

SteveMacenski, 2019-09-11:
This adheres to REP 105 definition of frames (like karto, cartographer, gmapping, etc...) so however you've had it working prior to this would be fine.

But in general, yes you'll need something to filter your IMU/odometry sources into a odom->base_link transform for this to provide the map->odom transform.

SteveMacenski, 2019-09-11:
Closing since its not related to the package, feel free to keep the dialog going if there's more you need

closed 2019-09-11 by SteveMacenski

joekeo, 2019-09-11:
Many thanks! And thanks for the fast reply.
Will do the filtering :)

best,
Pepe