SteveMacenski/slam_toolbox #97: Crash with sync_slam_toolbox_node: devel/lib/slam_toolbox/sync_slam_toolbox_node': free(): invalid pointer: 0x00007fe646ffc170 ***

kajep09, 2019-09-20:
Hi @SteveMacenski 

Thanks for another great package! I'm in the process of trying it out and comparing it to Cartographer, but I'm running into some crashes when trying it out on some of my rosbags.

Summary:
I am on 16.04 / Kinetic, but I cloned melodic-devel and ran install deps / build without problems (at least any problems that I am aware of).

After modifying mapper_params_online_sync.yaml and online_sync.launch I can start the mapping process and have it run for a while, but at some point it crashes on all my bag files (different sensors / robots).

The error, as far as I can tell, summarizes to 
```
`/ws/devel/lib/slam_toolbox/sync_slam_toolbox_node': free(): invalid pointer: 0x00007fe646ffc170 ***
```

I have yet to try async mode and see if the same thing happens there, and I haven't tried your snap either. I will try to do that in parallel with this issue being open.

If you need additional information feel free to ping me back. I look forward to trying out some of the interesting features when it no longer crashes :) 

Full paste:
https://pastebin.com/srrahvaW

Highlights from crash:
```
process[sync_slam_toolbox_node-1]: started with pid [29140]
[ INFO] [1568978332.594228288]: Node using stack size 40000000
[ INFO] [1568978332.657693139]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[ INFO] [1568978332.657756921]: Using plugin solver_plugins::CeresSolver
Registering sensor: [Custom Described Lidar]

Solver Summary (v 1.13.0-eigen-(3.2.92)-lapack-suitesparse-(4.4.6)-cxsparse-(3.1.4)-openmp)

                                     Original                  Reduced
Parameter blocks                           84                       81
Parameters                                 84                       81
Residual blocks                            31                       31
Residual                                   93                       93

Minimizer                        TRUST_REGION

Sparse linear algebra library    SUITE_SPARSE
Trust region strategy     LEVENBERG_MARQUARDT

                                        Given                     Used
Linear solver          SPARSE_NORMAL_CHOLESKY   SPARSE_NORMAL_CHOLESKY
Threads                                    50                       50
Linear solver threads                       1                        1
Linear solver ordering              AUTOMATIC                       81

Cost:
Initial                          2.687018e+06
Final                            1.774825e+00
Change                           2.687016e+06

Minimizer iterations                        9
Successful steps                            9
Unsuccessful steps                          0

Time (in seconds):
Preprocessor                           0.0003

  Residual evaluation                  0.0069
  Jacobian evaluation                  0.0164
  Linear solver                        0.0009
Minimizer                              0.0248

Postprocessor                          0.0002
Total                                  0.0253

Termination:                      CONVERGENCE (Parameter tolerance reached. Relative step_norm: 8.743665e-04 <= 1.000000e-03.)

*** Error in `/home/x/RCW_workspaces/capra_ws/devel/lib/slam_toolbox/sync_slam_toolbox_node': free(): invalid pointer: 0x00007fe646ffc170 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fe666cd07e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7fe666cd937a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fe666cdd53c]
/home/x/RCW_workspaces/capra_ws/devel/lib/libceres_solver_plugin.so(_ZN9__gnu_cxx13new_allocatorIiE10deallocateEPim+0x20)[0x7fe65ce65b4c]
/home/x/RCW_workspaces/capra_ws/devel/lib/libceres_solver_plugin.so(_ZNSt16allocator_traitsISaIiEE10deallocateERS0_Pim+0x2b)[0x7fe65ce63594]
/home/x/RCW_workspaces/capra_ws/devel/lib/libceres_solver_plugin.so(_ZNSt12_Vector_baseIiSaIiEE13_M_deallocateEPim+0x32)[0x7fe65ce6122c]
/home/x/RCW_workspaces/capra_ws/devel/lib/libceres_solver_plugin.so(_ZNSt12_Vector_baseIiSaIiEED2Ev+0x41)[0x7fe65ce5e85f]
/home/x/RCW_workspaces/capra_ws/devel/lib/libceres_solver_plugin.so(_ZNSt6vectorIiSaIiEED1Ev+0x41)[0x7fe65ce5c7b3]
/home/x/RCW_workspaces/capra_ws/devel/lib/libceres_solver_plugin.so(_ZN5ceres6Solver7SummaryD2Ev+0x1e)[0x7fe65ce5ae94]
/home/x/RCW_workspaces/capra_ws/devel/lib/libceres_solver_plugin.so(_ZN14solver_plugins11CeresSolver7ComputeEv+0x7f5)[0x7fe65ce5537f]
======= Memory map: ========
00400000-0053b000 r-xp 00000000 08:05 425235                             /home/x/RCW_workspaces/capra_ws/devel/.private/slam_toolbox/lib/slam_toolbox/sync_slam_toolbox_node
0073a000-0073b000 r--p 0013a000 08:05 425235                             /home/x/RCW_workspaces/capra_ws/devel/.private/slam_toolbox/lib/slam_toolbox/sync_slam_toolbox_node
0073b000-0073c000 rw-p 0013b000 08:05 425235                             /home/x/RCW_workspaces/capra_ws/devel/.private/slam_toolbox/lib/slam_toolbox/sync_slam_toolbox_node
0073c000-0073d000 rw-p 00000000 00:00 0 
02504000-02975000 rw-p 00000000 00:00 0                                 
 [heap]
...
 [stack]
...
[sync_slam_toolbox_node-1] process has died [pid 29140, exit code -6, cmd /home/x/RCW_workspaces/capra_ws/devel/lib/slam_toolbox/sync_slam_toolbox_node scan:=/os1/scan odom:=/camera/odom/sample __name:=sync_slam_toolbox_node __log:=/home/x/.ros/log/76759c88-db82-11e9-89d9-1c1bb5890ad0/sync_slam_toolbox_node-1.log].
log file: /home/x/.ros/log/76759c88-db82-11e9-89d9-1c1bb5890ad0/sync_slam_toolbox_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

kajep09, 2019-09-20:
Async fails with a similar error.

kajep09, 2019-09-20:
I tried changing stack sizes in the launch file (to match my standard ulimit -s of 8192) as well as increasing my ulimit to 40000 to match the default settings of the configuration file. However, it doesn't seem to have had any effect and I still get the same crash.

Interestingly (perhaps?), with both the bag files I tried (different robots) it always crashes around the same time in the bag, but I doesn't seem related to the size of the map or the time it has been running (~52 sec for one bag and 620 sec in another bag, these timestamps for the crashes are pretty consistent over multiple runs).

SteveMacenski, 2019-09-20:
Correct me if I'm wrong, that traceback looks like it originates from the Ceres library (?) while processing the summary, which is... strange. Did you try turning off debg_logging so that isn't being printed? either way that shouldn't happen. 

What platform is this? If its not related (you think) to the direction, if you run the rosbag starting lets say 4 seconds before the time it usually crashes, does it still crash? I have run this over bags of data ~10,000 seconds long and I've never seen a crash like that - that would have been all red alarms firing in my head to fix ASAP.

SteveMacenski, 2019-09-23:
Followup - can you compile with gdb and give me the traceback if you can replicate this? Make sure Ceres is current. If building from source, clean the workspace and try again -- maybe some strange issue.

kajep09, 2019-09-23:
Thank you for the quick response.

My platform is a standard Intel corei7 laptop (Thinkpad P51) with Ubuntu 16.04, ROS Kinetic and slam_toolbox built from source(melodic-devel at commit f028d62).

I tested the bags again with different start times and duration and it doesn't look like the crash is correlated to a specific time in the bag. However, I enabled debug_logging in your node, and it seems like the crash is always right after it prints `[DEBUG] [1569225229.980395192, 1544781373.766992301]: CeresSolver: Setting first node as a constant pose.`  which seems to originate from: 

```   // populate contraint for static initial pose
  if (!was_constant_set_ && first_node_ != nodes_->end())
  {
    ROS_DEBUG("CeresSolver: Setting first node as a constant pose.");
    problem_->SetParameterBlockConstant(&first_node_->second(0));
    problem_->SetParameterBlockConstant(&first_node_->second(1));
    problem_->SetParameterBlockConstant(&first_node_->second(2));
    was_constant_set_ = !was_constant_set_;
  }
```

Unfortunately, I am really inept with gdb and I can't seem to get any core dumps generated.

kajep09, 2019-09-23:
I tried to build from source on a mostly clean 18.04 / Melodic boot on the same laptop, and there I can run the same bag file all the way through without any issues.

I think that you are correct that the problem is probably isolated to Ceres. On my 16.04 / Kinetic boot I have compiled Cartographer from source with ceres-solver 1.13 and I'm not sure if there is something conflicting here.

Does your package require ceres-solver 1.14?

kajep09, 2019-09-23:
I recompiled ceres-solver at the newest version (1.14 - but master branch) and recompiled slam_toolbox afterwards and now it works on my 16.04 / Kinetic boot as well. So my take-away is that the minimum ceres-solver version required is 1.14 (you did say "Make sure Ceres is current.", but I guess that went in one ear (eye?) and out the other).

I guess other people might run into the same issue because they want to compare slam_toolbox with cartographer, and the wstool .rosinstall file for cartographer lists ceres-solver at 1.13 by default.

I'll leave the issue open for you to have a chance to digest and comment on all my spam, but in the end all that is required is probably a note somewhere in the readme that ceres-solver should be at 1.14?

Now I can finally get around to testing some of the interesting aspects of your package which I look very much forward to. Also good luck at RosCon2019!

SteveMacenski, 2019-09-23:
This is all from compiling ceres from scratch, which cartographer also doesn’t require (I assume?). Its assumed your getting from the apt package manager where there are rosdep keys for ceres I sync against. 

The thread warning you get is also from the flags you set on compile time and don’t represent a general issue. 50 is more or less a random number, but “alot” was my goal. Let ceres use as many threads as it deems appropriate, and obviously 50 is absurdly high.

SteveMacenski, 2019-09-23:
I’m going to close this out but glad you found your issue! Also I might recommend doing a “crash” course in gdb, it’ll save you a ton of time in the long run if you deal with memory management at all

closed 2019-09-23 by SteveMacenski