SteveMacenski/slam_toolbox #98: Snap running issue plugin

Middyeasy, 2019-09-20:
New to snap and this package so please forgive me but im getting this error whenever i try and run the snap version dev and regular. 
process[sync_slam_toolbox_node-1]: started with pid [3374]
[ INFO] [1568987315.882619235]: Node using stack size 40000000
[FATAL] [1568987316.062751959]: Failed to create solver_plugins::CeresSolver, is it registered and built? Exception: Failed to load library /snap/slam-toolbox-melodic/3/opt/ros/melodic/lib/libceres_solver_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = liblapack.so.3: cannot open shared object file: No such file or directory).
[sync_slam_toolbox_node-1] process has died [pid 3374, exit code 1, cmd /snap/slam-toolbox-melodic/3/opt/ros/melodic/lib/slam_toolbox/sync_slam_toolbox_node __name:=sync_slam_toolbox_node __log:=/root/snap/slam-toolbox-melodic/3/ros/log/bddb5af8-dbab-11e9-9058-00155d011907/sync_slam_toolbox_node-1.log].
log file: /root/snap/slam-toolbox-melodic/3/ros/log/bddb5af8-dbab-11e9-9058-00155d011907/sync_slam_toolbox_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

any help would be great as i expected the snap to just work and im not sure how to even get in there and mess with it.

SteveMacenski, 2019-09-20:
You're right, there's a bad link there. I thought I'd fixed that, I'll look into it today

SteveMacenski, 2019-09-20:
https://github.com/SteveMacenski/slam_toolbox/pull/99

will resolve, thanks for the patience. Sorry about that

Middyeasy, 2019-09-20:
Thank you!!!

Sent from my Verizon, Samsung Galaxy smartphone

#99<https://github.com/SteveMacenski/slam_toolbox/pull/99>

SteveMacenski, 2019-09-20:
New version is 1.1.3, should be up momentarily

SteveMacenski, 2019-09-20:
ugh, Canonical... ok I can't release it to the snap store due to their internal auto-rejection buffer. Its going to take awhile to work around them. In the meantime, you can compile the snap yourself with the generate_snap script

Middyeasy, 2019-09-20:
I really appreciate it guy!!!

Sent from my Verizon, Samsung Galaxy smartphone

SteveMacenski, 2019-09-22:
For clarity: the issue was that canonical changed a bunch of stuff from core to core18 and now the strict confinement doesn't give pluginlib access to the libraries it needs, so it needs to be used as a devmode or classic confinement snap, and snap store requires strict confinement for the stable release channels also starting this year... so I have to apply for an exception for technical grounds, I can release to the beta/edge channels but then you have to install via alot of flags that look dangerous but are actually safe.

SteveMacenski, 2019-10-17:
I am (for the moment) dropping support for snapstore, snaps are still fully supported but will require building yourself so devmode or  classic confinement (`./snap/generate_snap` will do this for you). 

This is now released to the ROS build farm, so you should be able to apt-get install this package as an alternative method of installation.

Middyeasy, 2019-10-17:
That's fantastic! I cant wait to try it. Thanks for making it so easy.

Sent from my Verizon, Samsung Galaxy smartphone

closed 2019-10-17 by SteveMacenski