SteveMacenski/slam_toolbox #107: [Question]Can slam_toolbox deal with scans when the robot not on a flat surface?

joekeo, 2019-10-01:
My question is if there is a way to filter out the scans when the robot is tilted in roll or pitch, or if this is a filter that we need to do ourselves to the /scan data that goes into the package?

SteveMacenski, 2019-10-01:
Yes, it can deal with that. It uses TF to transform the pointcloud of the scan into the local version of the frame. If that frame is tilted, so be your laser!

Closing since it was a pretty clear-cut answer. Feel free to keep the discussion going if you have further clarifications

closed 2019-10-01 by SteveMacenski

SteveMacenski, 2019-10-01:
But lets say the robot is tilted "a little bit", you should still be able to map without much of a concern. I've done some tests of this on the sidewalk (cracks, going over lumps, driveways, etc) in a major US city and showed it worked OK.

If you're asking "will this work on a drone" my answer will be no better or worse than gmapping / karto since they both make the same assumption and handle it the same way I do.