SteveMacenski/slam_toolbox #108: [Question]Dynamic environments

joekeo, 2019-10-01:
Hello, it is me again,
I would like to know if the system is able to cope with dynamic environments.
As an example i have a door that some times is open and some times closed and sometimes in the middle of the two.

The image below explains a little bit the situation.
[image]

Having a way to remove the old state of the door would help immensely for auto navigation.

SteveMacenski, 2019-10-01:
If you'd like to help by adding some kind of door classifier and remover, I'd be more than happy to include it.

However this is done by rastering a probablistic grid of values. So as you see the current state of the environment, it will update the state of the map, that's the whole point of SLAM. If the door is closed when you first start mapping, it'll show closed. If you later open it and rays can get through, the grid will be updated and that door should be removed and the new will be gone. 

Now this usually seems like a problem if you're using the map for something other than positioning (ei you have it setup with the static layer or something for global planning). I typically recommend people a) don't do that for reasons you address, as well as random noise for any particular mapping system resulting in odd behaviors around 1 bad pixel b) if you *must* do that, you create 1 "scrubbed" version of the map for the static layer to use that removes things like moving obstacles, doors, etc.

joekeo, 2019-10-02:
Hello Steve, 

Before anything, let me tell you: thanks for your help :)
I realise what happened in my experiment, I left the robot idle near the open door, now that I re made the experiment i moved the robot around different states of the door and the map updated properly.

One thing that caught my eye is the 'decay' time for the objects that are not there anymore. Is there any way to adjust this? Also is it possible to make the algorithm update the map while in static position?

SteveMacenski, 2019-10-02:
> 'decay' time for the objects that are not there anymore

So this isn't something like STVL where I'm just assigning some clever heuristic to grid values. A decay time may make sense for something where you're balancing computation with accurate representation, but for mapping, the whole game is accuracy. 

You can see the snippet that you're probably going to look for:

```
    virtual void UpdateCell(kt_int8u* pCell, kt_int32u cellPassCnt, kt_int32u cellHitCnt)
    {
      if (cellPassCnt > m_pMinPassThrough->GetValue())
      {
        kt_double hitRatio = static_cast<kt_double>(cellHitCnt) / static_cast<kt_double>(cellPassCnt);

        if (hitRatio > m_pOccupancyThreshold->GetValue())
        {
          *pCell = GridStates_Occupied;
        }
        else
        {
          *pCell = GridStates_Free;
        }
      }
    }
```

So if we dissect this, you could in practice tune both the minimum count and ratio parameters. Those might be the **only** 2 parameters in this project I didn't expose. If you're so motivated, please submit a PR to expose ROS params & then in `smapper_`'s `getOccupancyGrid` set those parameters with their respective set methods

SteveMacenski, 2019-10-02:
I'd be careful playing with these too much, but honestly I'm really interested if you play with those for a few days if you're able to substantially improve on it.

If you find values you like better, feel free to submit new defaults in your PR and I can run them over my private datasets to make sure everything seems reasonable

joekeo, 2019-10-03:
I am going to start working on this, Will hit you up if needed, and will make a PR wen I think it is ready

joekeo, 2019-10-07:
Hello Steve,
I am looking into this at the moment. Let me know if i understand this code correctly:

If a cell in the map has been 'seen' by the laser more than a threshold we calculate the ratio between the times it has been 'hit' and the times it has been 'missed'. If the ratio is above certain value, we declare it 'occupied' if not free.

What it comes to my mind immediately is that if a cell is hit a for a long time, then the object that is in that cell is removed, it is going to take a time proportional to the time it was there for the map to be updated with the current state of the real environment. 

Changing the ratio parameter will only change the time it will take for the 'new' environment to be represented in the map. But that time will still be proportionate right? So it will be fast for an object that is in constant change (people moving about) but much longer for objects that remain in the same place then change (like a door).

I think then my question to you is if it is worth exploring to solve this issue approaching it a little bit different? I am thinking that maybe having a maximum possible number of `cellPassCn` and `cellHitCnt` so that the number of hits does not grow too high, so a change in the cell state will be reflected faster. Alternatively maybe something like a rolling window for the last `n` cell passes and hits.

Because I don't have much knowledge of the rest of the code, I don't know if this is going to have a domino effect on the rest of the system.

SteveMacenski, 2019-10-08:
I think what you're proposing is reasonable, if you're seeing it be a problem more than it seems like it could theoretically be a problem. I had intended at implementing exactly what you proposed but after extensive testing I found that it wasn't really necessary, at least for what I am doing. I definitely would evaluate any PR you submit on this topic.

The ratio though should implement that to some level if you think about it. If the ratio of hit to miss is low, then you don't need that many new versions of hits to include into the map, though still may take a little while to remove a long standing object. In contrast, with the proposed method, over very long running (or lifelong multisession) sessions, the hit would get saturated in alot of places because you may have many and many observation

SteveMacenski, 2019-10-17:
@joekeo any update here? Even just starting with exposing the parameters would be a substantial improvement!

joekeo, 2019-10-17:
Hello Steve,
I haven't been able to make any progress on this topic as I have been very busy. I will aim to get a PR at some point next week, closer to the end of it.

joekeo, 2019-11-05:
@SteveMacenski I am actively working in this this week. And as you can see from above comments and the rest of this post, I am not a hard core programmer, not even a medium core, more like a softie. Anyways, I am trying my best to do some modifications and to test them on my robot.
I am interested in this bit of the code:
```
virtual kt_bool RayTrace(const Vector2<kt_double>& rWorldFrom,
                             const Vector2<kt_double>& rWorldTo,
                             kt_bool isEndPointValid,
                             kt_bool doUpdate = false)
    {
      assert(m_pCellPassCnt != NULL && m_pCellHitsCnt != NULL);

      Vector2<kt_int32s> gridFrom = m_pCellPassCnt->WorldToGrid(rWorldFrom);
      Vector2<kt_int32s> gridTo = m_pCellPassCnt->WorldToGrid(rWorldTo);

      CellUpdater* pCellUpdater = doUpdate ? m_pCellUpdater : NULL;
      m_pCellPassCnt->TraceLine(gridFrom.GetX(), gridFrom.GetY(), gridTo.GetX(), gridTo.GetY(), pCellUpdater);

      // for the end point
      if (isEndPointValid)
      {
        if (m_pCellPassCnt->IsValidGridIndex(gridTo))
        {
          kt_int32s index = m_pCellPassCnt->GridIndex(gridTo, false);

          kt_int32u* pCellPassCntPtr = m_pCellPassCnt->GetDataPointer();
          kt_int32u* pCellHitCntPtr = m_pCellHitsCnt->GetDataPointer();

          pCellPassCntPtr[index]++;
          pCellHitCntPtr[index]++;

          if (doUpdate)
          {
            (*m_pCellUpdater)(index);
          }
        }
      }
```
As I understand it, this is where each of the grid "tiles" get either a pass counter or a pass counter and a hit, right? I can clearly see where it get the pass and hit counter incremented
```
pCellPassCntPtr[index]++;
pCellHitCntPtr[index]++;
```
but I cannot yet find where is that a grid "tile" just gets its pass counter incremented. I would have guesses I would see some lonely `pCellPassCntPtr[index]++;` somewhere else, instead I have the feeling that this happens in `m_pCellPassCnt->TraceLine(gridFrom.GetX(), gridFrom.GetY(), gridTo.GetX(), gridTo.GetY(), pCellUpdater);` by `pCellUpdater`. 

`TraceLine` is bit more abstract, but if my understanding is correct, is doing exactly that, traicing all the grid "tiles" that are between the emitted laser and the end point. I am assuming that all those grid "tiles" are assumed to be empty, therefore in this function the tiles should be updated to say that there was a ray pass but no hit.

I think that this bit of code does it
```
Vector2<kt_int32s> gridIndex(pointX, pointY);
        if (IsValidGridIndex(gridIndex))
        {
          kt_int32s index = GridIndex(gridIndex, false);
          T* pGridPointer = GetDataPointer();
          pGridPointer[index]++;

          if (f != NULL)
          {
            (*f)(index);
          }
        }
```
This is where I was expecting the lonely `pCellPassCntPtr[index]++;` in some sort of shape or form, instead is am finding that the trail leads to ` class KARTO_EXPORT CellUpdater : public Functor` which is in the limit of my abstraction level. 

Do you have any pointers if I am going in the right direction and where I should go from here? Any extra help would be appreciated.

Best,
pepe

joekeo, 2019-11-05:
I just made the first prototype of the mod, it is horrible, and needs a lot of work, but if you have a chance, have a loot at it to see if it makes any sense. 
i will make an initial PR so it is easier to check the code

SteveMacenski, 2019-11-06:
I'm a little confused -- what specifically are you trying to accomplish? 

You shouldn't need to mess with that bit of code. The last we chatted you wanted to change the parameter values of the pass vs hit threshold to mark a cell as occupied or free (or having enough information to make a declaration). What is your goal in modifying that bit of information? That's building the actual grid table of values to be later used to decide whether its occupied or free, this function itself isn't making any decision about free-ness or not-free-ness

nickvaras, 2020-04-07:
Although this issue is closed and I'm not proposing any specific implementation, I have the unsubstantiated feeling that incorporating concepts of decay and frequency of observation are probably a step in the right direction toward a truly intelligent robot navigation system. Some of these concepts are powerful and I see their value in STVL every day. Clearing (as a means of forgetting) obstacles via raytracing is inefficient with the current implementation of laserscans, and something like a "quicker" local (within laser range) and "slooow" global (beyond that range) decay rates  could make a huge difference in lifelong mapping, *I think. [This paper](http://strands.acin.tuwien.ac.at/publications/y4/krajnik_pulido_IROS16.pdf)  seemed to me to be taking that next step beyond static slam, but bringing groundbreaking concepts into production-ready software is a huge challenge and that's where you Steve have really shined. I'm sure this will only keep getting better into the future.

nickvaras, 2020-04-07:
Another shortcoming of raycasting/raytracing-only clearing is that, in the events of wonky loopclosure, the ghost markings of previous passes will never clear since they are behind the new observed obstacle, as seen in the wall on the right in the image:
[image]

SteveMacenski, 2020-04-07:
> I have the unsubstantiated feeling that incorporating concepts of decay and frequency of observation are probably a step in the right direction toward a truly intelligent robot navigation system

Agreed, that's where the work on lifelong mapping node came into play. While its a (generally non-functioning) example, it does work as a POC that its possible. The major issue facing that implementation is how to scale over the long, long duration. In order to scale on the order of months, you need to both cull data (laser scans) and well as graph nodes (poses). When removing graph poses, you're going to create some disorder in the structure of the graph. There's probably a way to augment the remaining graph recursively for these updates to retain some structure while removing old nodes, but I haven't gotten to it. If that interests you, more than welcome to work on it together and I can help as able. For shorter-ish lifelong mapping terms, we could potentially only remove the data and any graph nodes with data removed keeps the structure but no longer effects the generated occupancy map while "holding down" the pose graph. Over time though, your number of nodes will continue to grow but depending on your CPU and timescale, that may be acceptable. I generally look for "one shot kill" solutions over incremental improvements, but this could be a valuable addition to the non-functioning demo to make a more functioning. 

> bringing groundbreaking concepts into production-ready software is a huge challenge and that's where you Steve have really shined

I appreciate that, that's how I think of it too. 

Dynamic environments "do work" because we're using an occupancy grid - as new measurements come in the old are less of the total for a cell resulting in lowering probability. You are right though that if you can't ray cast out, you can't remove. But that issue comes along probably because you need to tune parameters. Double walls probably shouldn't happen.

nickvaras, 2020-04-09:
> In order to scale on the order of months, you need to both cull data (laser scans) and well as graph nodes (poses). When removing graph poses, you're going to create some disorder in the structure [...]

I do need do dig deeper into the inner workings of karto and the general algorithm, so with only a superficial, observational idea of how this thing may work, I ask:

- Can nodes be "voxelized" (for lack of a better work) and merged? That is, can one set a policy where only one node is allowed within a 1m^2 space, and if two distinct nodes fall within that space, we replace them with a single, synthetic node in their stead.
- How are the laserscans stored? as the usual distance-angle arrays? Could they instead be kept as something closer to pointclouds, that would better fit with the synthetic node approach described above.

> probably because you need to tune parameters. Double walls probably shouldn't happen. Given your laser scan and box, you probably just need to drive around a little more for another loop closure to fix it actually.

Yes, tuning is probably in order, but so far its been relatively easy to make the loop closure fail. I need to dig deeper into the scanmatching-augmented odometry, but in my general experience it can improve rotational odometry significantly, but in featureless aisles it will be worse than wheel odometry. It would be nice to treat rotation and translation enhancement from scanmatching separately. I've been using some deliberately mapping-hostile gazebo environments and I'm sure that it would work better on the real robot. Will try that soon enough.

SteveMacenski, 2020-07-31:
Reviewing old tickets - closing this from inactivity and no direct action items. If you feel this should be reopened let me know and I'll do so.

closed 2020-07-31 by SteveMacenski