SteveMacenski/slam_toolbox #109: Proper usage of `save_map` service?

ubtjoe, 2019-10-03:
Hey @SteveMacenski ,

While observing mapping data from my robot (which is running as the ros master) I am trying to save the map (on my laptop) with the following call:

```
rosservice call /sync_slam_toolbox_node/save_map "name:
  data: './map.pgm'"
```

and the behavior is quite bizarre.  Sometimes the command hangs and sometimes it completes but there is no `map.pgm` file anywhere to be found.  Meanwhile, `rosrun map_server map_saver` does _exactly_ what I want.

You mention in the `README` that the aforementioned service is just a wrapper for `map_saver`, but this doesn't appear to  be the case.  Any thoughts?

SteveMacenski, 2019-10-03:
It's just a simple wrapper, if you look at the map saver, its just calling the bash for map saver. Did you check your `.ros`? If you don't specify a full path they'll get dumped there. If you add `/` but isnt a full path, then it will also fail as would it with calling from the terminal.

I'd need more information about what you're trying to do. What do your logs say?

ubtjoe, 2019-10-03:
See below:

```
ros@joeslaptop:~/.ros/log$ rosservice call /sync_slam_toolbox_node/save_map "name:
  data: 'maptest'" 

ros@joeslaptop:~/.ros/log$ cat rosservice_
rosservice_570_1570126106524.log  rosservice_604_1570126368162.log
rosservice_581_1570126209337.log  rosservice_612_1570126398903.log
rosservice_595_1570126345634.log  
ros@joeslaptop:~/.ros/log$ cat rosservice_612_1570126398903.log 
[rospy.client][INFO] 2019-10-03 11:13:18,922: init_node, name[/rosservice_612_1570126398903], pid[612]
[xmlrpc][INFO] 2019-10-03 11:13:18,922: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2019-10-03 11:13:18,922: Started XML-RPC server [http://192.168.0.139:38783/]
[rospy.init][INFO] 2019-10-03 11:13:18,923: ROS Slave URI: [http://192.168.0.139:38783/]
[rospy.impl.masterslave][INFO] 2019-10-03 11:13:18,923: _ready: http://192.168.0.139:38783/
[rospy.registration][INFO] 2019-10-03 11:13:18,923: Registering with master node http://192.168.0.178:11311
[xmlrpc][INFO] 2019-10-03 11:13:18,923: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2019-10-03 11:13:19,023: registered with master
[rospy.rosout][INFO] 2019-10-03 11:13:19,024: initializing /rosout core topic
[rospy.rosout][INFO] 2019-10-03 11:13:19,105: connected to core topic /rosout
[rospy.simtime][INFO] 2019-10-03 11:13:19,132: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.internal][INFO] 2019-10-03 11:13:19,339: topic[/rosout] adding connection to [/rosout], count 0
[rospy.core][INFO] 2019-10-03 11:13:21,236: signal_shutdown [atexit]
[rospy.internal][INFO] 2019-10-03 11:13:21,268: topic[/rosout] removing connection to /rosout
[rospy.impl.masterslave][INFO] 2019-10-03 11:13:21,269: atexit
```

ubtjoe, 2019-10-03:
The `signal_shutdown` line appears ominous...  What is the log output supposed to look like?

SteveMacenski, 2019-10-03:
This is client side code, this isnt going to tell you anything. Whats the actual node saying? 

Also if you’re on another machine calling it, are you certain the request is going through? Again looking at server logs.

ubtjoe, 2019-10-03:
Issue resolved... Closing now.

The maps are being saved on the server in `~/.ros`.  Naively, I thought that the service call would save the map locally on the caller's, in this case the client's, side.  I will just save the map via `map_server` directly.  

FWIW: It would be a nice feature to be able to use the rviz plugin and services from the client-side, as I don't want to run anything graphical on my robot pc, whether through vnc or other.

Regarding, your comment about mixing distros:  What's wrong with this?  I am using docker and the individual containers are isolated.  Any exposure is only over http, right?  Has roscore changed that much between kinetic and melodic?  A couple days of testing and I've not had any issues between the Toolbox and my other ros stack; the whole setup runs like a champ.

closed 2019-10-03 by ubtjoe

SteveMacenski, 2019-10-04:
Its bad because there’s no promise that the message hashes are the same between versions, or that the core ros libraries are compatible. If you’re seeing it work then you’re probably fine, but I wouldn’t expect that to work in the general case. I wouldn’t expect the errors to manifest randomly, so if you can use all the services/topics without an issue you’re in the clear. 

If you’re trying to save maps locally, calling the service to the slam toolbox server doesn’t really make sense, you need to run map saver on your second machine. Only way the toolbox could do it is if you also provided ssh keys and info for the server to scp the data over, which seems over the top. 

The server is being called to perform a remote action, I dont think its in scope to call a server to perform a local action. But glad you found it!

Mohit-Ak, 2020-11-27:
@SteveMacenski 
Can you please post the command for ROS2 usage as well. I tried the following and they did not work.

- ros2 service call /save_map slam_toolbox/srv/SaveMap "name: data: './map.pgm'"
```yaml.scanner.ScannerError: mapping values are not allowed here
  in "<unicode string>", line 1, column 11:
    name: data: './map.pgm'
```
-  ros2 service call /save_map slam_toolbox/srv/SaveMap "name: ./map.pgm"
```Failed to populate field: Value './map.pgm' is expected to be a dictionary but is a str```
- ros2 service call /save_map slam_toolbox/srv/SaveMap "{'name':"knox"}"
```Failed to populate field: Value 'knox' is expected to be a dictionary but is a str```

Mohit-Ak, 2020-11-28:
I actually solved it. There is a hidden newline character.
```ros2 service call /save_map slam_toolbox/srv/SaveMap "name:
  data: './map.pgm'"
waiting for service to become available...
requester: making request: slam_toolbox.srv.SaveMap_Request(name=std_msgs.msg.String(data='./map.pgm'))

response:
slam_toolbox.srv.SaveMap_Response()
```
But strangely, I don't see any map getting saved.

akela1101, 2022-11-22:
One liner would be:
```
rosservice call /slam_toolbox/save_map "name: {data: 'your_map_name'}"
```