SteveMacenski/slam_toolbox #118: [Dashing] Mapping problems

maxlein, 2019-10-29:
Hi,

I recently found your toolbox and was very excited to try it out as I have only experience with gmapping ;)  

Setup was no big problem, just some wondering on some of the parameters like frame names.  

So I just give it a try and mapped one part of a warehouse, and as you can see I got some wrong map updates.  
  
[image]
[image]

Don't let you irritate you by the scan points on the 2nd picture, they are some old points.

I remember such a problem from gmapping but can't remember what fixed it, but maybe you have seen this behavior on your tests recently? ...  

First my setup:  

Core i5, Ubuntu 18.04, ROS2 Dashing, no navigation stack, tested with joystick driver and mapping
Lidar: Pepperl & Fuchs R2000 ( 360° 2D)

Because you have a scan matcher, I was wondering if you publish a odom corrected frame or something similar, but I don't think so, so the tf frames should look something like that:  

scan_0 -> base_link -> odom_wheel -> map 

````
osg_1:
  slam_toolbox:
    ros__parameters:

      use_sim_time: false

      # link_match_minimum_response_fine: 0.4 
      # link_scan_maximum_distance: 20.0
      # loop_search_maximum_distance: 50.0
      # distance_variance_penalty: 1.8             
      # angle_variance_penalty: 0.20    
      # minimum_angle_penalty: 1.6 #0.9

      # Plugin params
      solver_plugin: solver_plugins::CeresSolver
      ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
      ceres_preconditioner: SCHUR_JACOBI
      ceres_trust_strategy: LEVENBERG_MARQUARDT
      ceres_dogleg_type: TRADITIONAL_DOGLEG
      ceres_loss_function: None

      # ROS Parameters
      odom_frame: odom_wheel
      map_frame: map
      base_frame: base_link
      scan_topic: scan_0
      mode: mapping

      # lifelong params
      lifelong_search_use_tree: false
      lifelong_minimum_score: 0.1
      lifelong_iou_match: 0.85
      lifelong_node_removal_score: 0.04
      lifelong_overlap_score_scale: 0.06
      lifelong_constraint_multiplier: 0.08
      lifelong_nearby_penalty: 0.001
      lifelong_candidates_scale: 0.03

      # if you'd like to immediately start continuing a map at a given pose
      # or at the dock, but they are mutually exclusive, if pose is given
      # will use pose
      #map_file_name: test_steve
      #map_start_pose: [0.0, 0.0, 0.0]
      #map_start_at_dock: true

      debug_logging: true
      throttle_scans: 1
      transform_publish_period: 0.01 #if 0 never publishes odometry
      map_update_interval: 5.0
      resolution: 0.05
      max_laser_range: 10.0 #for rastering images
      minimum_time_interval: 0.5
      transform_timeout: 0.3
      tf_buffer_duration: 10.
      stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps

      # General Parameters
      use_scan_matching: true
      use_scan_barycenter: true
      minimum_travel_distance: 0.5
      minimum_travel_heading: 0.5
      scan_buffer_size: 10
      scan_buffer_maximum_scan_distance: 10.0
      link_match_minimum_response_fine: 0.1  
      link_scan_maximum_distance: 1.5
      loop_search_maximum_distance: 3.0
      do_loop_closing: true 
      loop_match_minimum_chain_size: 10           
      loop_match_maximum_variance_coarse: 3.0  
      loop_match_minimum_response_coarse: 0.35    
      loop_match_minimum_response_fine: 0.45

      # Correlation Parameters - Correlation Parameters
      correlation_search_space_dimension: 0.5
      correlation_search_space_resolution: 0.01
      correlation_search_space_smear_deviation: 0.1 

      # Correlation Parameters - Loop Closure Parameters
      loop_search_space_dimension: 8.0
      loop_search_space_resolution: 0.05
      loop_search_space_smear_deviation: 0.03

      # Scan Matcher Parameters
      distance_variance_penalty: 0.5      
      angle_variance_penalty: 1.0    

      fine_search_angle_offset: 0.00349     
      coarse_search_angle_offset: 0.349   
      coarse_angle_resolution: 0.0349        
      minimum_angle_penalty: 0.9
      minimum_distance_penalty: 0.5
      use_response_expansion: true
```
Also little problem not showing laser scans in rviz. I am not exactly sure where the problem lies, if its rviz itself or some tf publishing.  
The only thing I saw, was some connection between transform_publish_period param and the time difference shown below.   
The delay changes when changing transform_publish_period.

Requested time 1572337875.48494 but the latest data is at time 1572337875.45044, when looking up transform from frame [scan_0] to frame [map]

SteveMacenski, 2019-10-29:
> I remember such a problem from gmapping but can't remember what fixed it

This sounds like an issue in your setup and not in the package if you’ve seen this in other packages.

SteveMacenski, 2019-10-30:
Are there mirrors or something in your space? 

Can't say I've ever seen this outside of gmapping in simulation -- I used gmapping for years and I don't recall ever seeing this with a properly configured hardware asset

maxlein, 2019-10-31:
So, I now tried against a gazebo simulation with my robot urdf and got the same results.

The only thing whats not like a "normal" robot is, that the scanner is bottom up, so the joint roll axis is 180°.  Switched this to zero, and mapping works.  

For today I am finished.  
And I also don't think your part is the problem except you assume something about the scan data without taking the tf tree into account.

SteveMacenski, 2019-11-01:
This does take that into account so I think there may be an issue with your TF tree. 

All of my robots also have their lasers upside down- so at least in the ROS1 version I know this works. There may be something going on with the TF lookup in dashing, I’ll file a new ticket to validate since I validated the dashing port using another robot that had it facing up.

closed 2019-11-01 by SteveMacenski