SteveMacenski/slam_toolbox #119: Check ROS2 TF laser orientation lookup

SteveMacenski, 2019-11-01:


maxlein, 2019-11-02:
I made a little urdf with two lidars in gazebo, one inverted.  
```
ros2 run tf2_ros tf2_echo base_link scanner_link

At time 0.0
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
  
ros2 run tf2_ros tf2_echo base_link inverted_scanner_link
At time 0.0
- Translation: [0.000, 0.000, 0.500]
- Rotation: in Quaternion [1.000, 0.000, 0.000, -0.000]
```

Running rviz with static link `base_link` and it looks like no transformation.  
Green is default lidar and red the inverted one..

[image]
[image]

SteveMacenski, 2019-11-03:
RVIZ they look suspiciously like they're on the same plane even though your transforms show they're 0.5m offset in Z, is that true?

Looking at the frames of the links in RVIZ, it does look inverted. The flipping of data does seem surprising. Is the global base_link frame somewhere in the RVIZ window I'm not seeing? Maybe something odd with the tree?

This implies it's not actually this package's fault but TF/projections? That also seems to be a surprising result if you say gmapping was working in ROS2 given that it uses the same mechanics to determine orientation.

maxlein, 2019-11-03:
Base_link has same pose like scanner_link, so i unchecked display for better readability.

Cant say if z is also wrong, I thought it was ok when looking around in rviz.
Bit cant check for the next 30hours or so, have a long flight.

And no, gmapping was a few years back. In Ros2, yours was the only mapping solution I tried.

So tf seems to be the problem, but a bit questionable how this bug can hide for so long..

SteveMacenski, 2019-11-03:
30 flight to.. IROS? I'm in China myself. 

I'll take another look at this hopefully today, though admittedly my list of things to do today has been growing

SteveMacenski, 2019-11-03:
If you can provide me a simulation environment with a robot and an inverted laser, it would speed up my ability to debug. Right now the only ros2-able simulation that I know of with all the stuff I need is the TB3 and that's not inverted

maxlein, 2019-11-03:
> 30 flight to.. IROS? I'm in China myself.

Ja, saw your talk at roscon 😉👍
But not going to IROS, back home to austria.

> I'll take another look at this hopefully today, though admittedly my list of things to do today has been growing

I can try to give you the simulation, but that will take some time and I also dont know if I can exclude the proprietary stuff from my company that easy. Dont think they will allow me to. 

But is a simulation with mapping really needed if we think tf is the problem?
Just a little test node and spawning lidars which do a tranform and compare returned points was my thinking. 

I will have a look, lidar Simulation test could be easier to extract.

maxlein, 2019-11-03:
This is the urdf for the upper lidar test.  Maybe it saves you some time.

[sensor_sim_test.urdf.txt](https://github.com/SteveMacenski/slam_toolbox/files/3801237/sensor_sim_test.urdf.txt)

But if you have the turtlebot sim, can't you just invert the lidar in urdf?   
Should be the easiest option.

SteveMacenski, 2019-11-04:
Test I did:

1) launch navigation2 stack with TB3 in simulation. I commented out the actual navigation part, but it brings up the TB3 waffle, gazebo, rviz.
2) launch online sync launch slam toolbox on dashing devel branch as is
3) launch teleop for turtlebot3

Built map, no problem and added prints confirms it sees it as right side up. All updates the print also shows it goes correctly through the pipeline. This is what I expect given this is how I tested the port to begin with.

Repeat with laser inverted in URDF. Prints from TF also confirm its upside down and shows correctly going through the pipeline. I suspect what you were seeing was something odd to do with your configuration (or that your laser scanner isnt spinning the correct direction?) 

[image]

closed 2019-11-04 by SteveMacenski

maxlein, 2019-11-04:
Found the problem in simulation:  
I copyied the laser urdf and in there was a comment:  
`                    
`  

So frame name was not actively set, and I assumed it defaults to its own link name.  
But thats not the case, it takes its parents link name! And thats base link :(
I think  this is a bug in the gazebo plugin, makes no sense to me...  

But sorry for wasting your time for this. 
Now will look to the real robot again and whats the problem there...

SteveMacenski, 2019-11-04:
Its ok. Given its a new port to ROS2 I’d rather find problems fast or find non-problems faster to make sure I did everything right. 

Not sure what you mean by “update topic”

maxlein, 2019-11-04:
In rviz you have the map display and there you have the topic "map" and the map updates topic "map updates".  

I only have the default map topic.  

[image]