SteveMacenski/slam_toolbox #122: Slam Toolbox and Cartographer

RonaldEnsing, 2019-11-07:
Thanks a lot for the great talk at ROScon. Unfortunately, there was no opportunity for questions at the end of the talk, so that's why I open a discussion here.

I am very much interested in the relation between Slam Toolbox and Cartographer. What were the most important reasons that you chose to write a new SLAM package, instead of extending cartographer? The Slam Toolbox readme states: '_This project contains the ability to do most everything any other available SLAM library, both free and paid, and more._' However, I am not convinced, especially since cartographer is able to do SLAM in 3D and is able to include the GNSS in the pose graph optimization.

SteveMacenski, 2019-11-07:
There is basically no relationship between this work and cartographer. 

By in large, cartographer does not work for most indoor industrial applications. This is a result I’ve come to from my experiences as well as polling members of the community. By in large, no one has gotten the performance or stability out of cartographer required for a serious product. Of course there are a few notable exceptions to that, but I’m not trying to solve the exceptional cases but the majority of cases. 

I provide nearly all interfaces needed for industrial large-scale slam. It is not my intention with this work to extend to 3D lidars.

RonaldEnsing, 2019-11-07:
Thank you for this explanation. I am interested in outdoor 3D SLAM for self-driving cars. Since Slam Toolbox is particularly suitable for 2D indoor, I'll stick with cartographer (3D) for now. I get decent results with cartographer in offline mode. However, since the computations are quite heavy, I haven't got it fast enough for _online_ mode. Because of that, I am still interested in any other SLAM package.

There are so many open source SLAM packages available. I think that an effort to work towards a general SLAM package seems reasonable and desirable to me. I'll keep an eye out on the discourse as well: https://discourse.ros.org/t/supporting-maintaining-slam-in-ros2-input-requested/10986/39

SteveMacenski, 2019-11-07:
Well I certainly support the concept of that discussion, but we also need more people (like you!) to get involved in putting boots on the ground to make it happen. 

This work is not intended or geared to autonomous driving. Frankly I think its dangerous for anyone to use Cartographer or this project in autonomous driving for countless reasons, but that’s up to you and your risk management team.

RonaldEnsing, 2019-11-07:
> Well I certainly support the concept of that discussion, but we also need more people (like you!) to get involved in putting boots on the ground to make it happen. I’m more than willing to spin up an official TSC sponsored ROS2 working group to make it happen, but it can’t just be me +/- a part time Amazon engineer. 3D support is something I believe strongly in.
> 

I would be happy to see where I can join. Shall we proceed the discussion on the discourse?

> This work is not intended or geared to autonomous driving. Frankly I think its dangerous for anyone to use Cartographer or this project in autonomous driving for countless reasons, but that’s up to you and your risk management team.

Yeah, we are not using cartographer or any other SLAM package for experiments where the car actually drives. We use cartographer for data sets that were recorded in regular, manual driving mode.

SteveMacenski, 2019-11-08:
Let me figure that out, but yeah posting on that thread is a good way to keep all the information in one spot.