SteveMacenski/slam_toolbox #125: [Eloquent][Bug] Timer interface not set before waitForTransform

maxlein, 2019-11-20:
I found a bug building toolbox from source in eloquent and running the `lifelong_slam_toolbox` node:  
I know its not released for eloquent yet, but for your interest:  

```
[slam_toolbox-9] terminate called after throwing an instance of 'tf2_ros::CreateTimerInterfaceException'
[slam_toolbox-9]   what():  timer interface not set before call to waitForTransform
```

SteveMacenski, 2019-11-20:
Ah, I believe I just support Dashing right now, the timer interface is a new Eloquent thing- feel free to add a PR if you can!

maxlein, 2019-11-20:
For now I moved back to Dashing, too many missing packages and things which lead to problems...  
The mapping stuff is working, but I think there is another issue coming regarding localization only and loadPoseGraphByParams()...

SteveMacenski, 2019-11-20:
OK, I'm going to go ahead and close this then since I don't have an eloquent branch yet (and also isn't technically out yet). 

See other ticket for the load pose graph by params conversation

closed 2019-11-20 by SteveMacenski

SteveMacenski, 2019-11-22:
@maxlein quick note, there's now an eloquent branch with the timer set.

CleopatraDion, 2020-04-17:
@SteveMacenski where exactly did you fix this issue? I'm working on the iris_lama slam package, eloquent-devel, and I am getting a similar issue. Would be nice to see what you did, to see if I can adapt it to this package...?

SteveMacenski, 2020-04-17:
Look at TF2 Instantiation. You must set a timer now. Eloquent branch and forward had it.