SteveMacenski/slam_toolbox #126: [Dashing][Bug] Seg fault in loadPoseGraphByParams()

maxlein, 2019-11-20:
So here I am again...  

So I am just trying to localize on an existing map.  
Therefore I thought I take the localization node and pass the map in yaml.  

But I always get a segmentation fault when loading the pose graph.  

Right now I am at `void SlamToolbox::loadPoseGraphByParams()` line 287:  

` req->initial_pose = pose;`

First I thought pose is assigned to an uninitialized pointer ->  
`std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph::Request> req;`  

but this also didn't help:  

`std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph::Request> req = std::make_shared<slam_toolbox::srv::DeserializePoseGraph::Request>();`  

I guess the default constructor is called without calling make_shared but until now I thought you have to call make_shared.  

But either way, I don't know why, but assigning the pose leads to a seg fault...

SteveMacenski, 2019-11-20:
Can you compile with gdb flags and get a traceback?

maxlein, 2019-11-20:
So after building in a shell (running inside CLion with linking to libraries which are build by the IDE and also having installed libs by colcon can cause troubles ) with `make_shared` on req and resp I get:  

```  
[ERROR 1574270648.468606872] [slam_toolbox]: LocalizationSlamToolbox: Incorrect number of arguments for map starting pose. Must be in format: [x, y, theta]. Starting at the origin
terminate called after throwing an instance of 'std::bad_weak_ptr'
  what():  bad_weak_ptr

Thread 1 "lifelong_slam_t" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) backtrace
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff606f801 in __GI_abort () at abort.c:79
#2  0x00007ffff66c4957 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff66caab6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff66caaf1 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff66cad24 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7883162 in std::__throw_bad_weak_ptr () at /usr/include/c++/8/bits/exception.h:63
#7  0x00007ffff7877015 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count (__r=..., this=0x7fffffffa238) at /home/kin/playground_ws/src/slam_toolbox/src/slam_toolbox_common.cpp:851
#8  std::__shared_ptr<rclcpp::Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<rclcpp::Node, void> (__r=..., this=0x7fffffffa230) at /usr/include/c++/8/bits/shared_ptr_base.h:1191
#9  std::shared_ptr<rclcpp::Node>::shared_ptr<rclcpp::Node, void> (__r=std::weak_ptr<rclcpp::Node> (empty) = {...}, this=0x7fffffffa230) at /usr/include/c++/8/bits/shared_ptr.h:276
#10 std::enable_shared_from_this<rclcpp::Node>::shared_from_this (this=0x5555558204d8) at /usr/include/c++/8/bits/shared_ptr.h:655
#11 slam_toolbox::SlamToolbox::deserializePoseGraphCallback(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph_Request_<std::allocator<void> > >, std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph_Response_<std::allocator<void> > >) () at /home/kin/playground_ws/src/slam_toolbox/src/slam_toolbox_common.cpp:822
#12 0x00007ffff7bab84f in slam_toolbox::LifelongSlamToolbox::deserializePoseGraphCallback(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph_Request_<std::allocator<void> > >, std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph_Response_<std::allocator<void> > >) () at /usr/include/c++/8/ext/atomicity.h:96
#13 0x00007ffff786eda5 in slam_toolbox::SlamToolbox::loadPoseGraphByParams() () at /usr/include/c++/8/bits/shared_ptr_base.h:614
#14 0x00007ffff7ba9de0 in slam_toolbox::LifelongSlamToolbox::LifelongSlamToolbox(rclcpp::NodeOptions) () at /home/kin/playground_ws/src/slam_toolbox/src/experimental/slam_toolbox_lifelong.cpp:41
#15 0x000055555559f62d in __gnu_cxx::new_allocator<slam_toolbox::LifelongSlamToolbox>::construct<slam_toolbox::LifelongSlamToolbox, rclcpp::NodeOptions&> (this=<optimized out>, __p=0x5555558204d0) at /usr/include/c++/8/new:169
#16 std::allocator_traits<std::allocator<slam_toolbox::LifelongSlamToolbox> >::construct<slam_toolbox::LifelongSlamToolbox, rclcpp::NodeOptions&> (__a=..., __p=0x5555558204d0) at /usr/include/c++/8/bits/alloc_traits.h:475
#17 std::_Sp_counted_ptr_inplace<slam_toolbox::LifelongSlamToolbox, std::allocator<slam_toolbox::LifelongSlamToolbox>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<rclcpp::NodeOptions&> (__a=..., this=0x5555558204c0)
    at /usr/include/c++/8/bits/shared_ptr_base.h:545
#18 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<slam_toolbox::LifelongSlamToolbox, std::allocator<slam_toolbox::LifelongSlamToolbox>, rclcpp::NodeOptions&> (__a=..., __p=<synthetic pointer>: <optimized out>, 
    this=<synthetic pointer>) at /usr/include/c++/8/bits/shared_ptr_base.h:677
#19 std::__shared_ptr<slam_toolbox::LifelongSlamToolbox, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<slam_toolbox::LifelongSlamToolbox>, rclcpp::NodeOptions&> (__tag=..., this=<synthetic pointer>)
    at /usr/include/c++/8/bits/shared_ptr_base.h:1342
#20 std::shared_ptr<slam_toolbox::LifelongSlamToolbox>::shared_ptr<std::allocator<slam_toolbox::LifelongSlamToolbox>, rclcpp::NodeOptions&> (__tag=..., this=<synthetic pointer>) at /usr/include/c++/8/bits/shared_ptr.h:359
#21 std::allocate_shared<slam_toolbox::LifelongSlamToolbox, std::allocator<slam_toolbox::LifelongSlamToolbox>, rclcpp::NodeOptions&> (__a=...) at /usr/include/c++/8/bits/shared_ptr.h:706
#22 std::make_shared<slam_toolbox::LifelongSlamToolbox, rclcpp::NodeOptions&> () at /usr/include/c++/8/bits/shared_ptr.h:722
#23 main () at /home/kin/playground_ws/src/slam_toolbox/src/experimental/slam_toolbox_lifelong_node.cpp:45
#24 0x00007ffff6050b97 in __libc_start_main (main=0x55555559f2d0 <main>, argc=2, argv=0x7fffffffc6b8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffc6a8) at ../csu/libc-start.c:310
#25 0x000055555559fa2a in _start () at /home/kin/playground_ws/src/slam_toolbox/src/experimental/slam_toolbox_lifelong_node.cpp:32
(gdb) 
```  

And if I run with localization_node:  

````
[... cut ...]
#8  std::make_shared<slam_toolbox::LocalizationSlamToolbox, rclcpp::NodeOptions&> () at /usr/include/c++/8/bits/shared_ptr.h:722
#9  main () at /home/kin/playground_ws/src/slam_toolbox/src/slam_toolbox_localization_node.cpp:45
#10 0x00007ffff6042b97 in __libc_start_main (main=0x55555559f2f0 <main>, argc=2, argv=0x7fffffffc768, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffc758) at ../csu/libc-start.c:310
#11 0x000055555559fa4a in _start () at /home/kin/playground_ws/src/slam_toolbox/src/slam_toolbox_localization_node.cpp:32
(gdb) 
```

SteveMacenski, 2019-11-20:
Ah thanks, I know exactly the issue then. 

You can't do `shared_from_this()` in the constructor of a Node class because the node hasn't yet been fully constructed. Therefore the reference counter on `this` is still 0 and a shared from this requires it to have already been in use. 

Basically this call needs to be refactored out of the constructor

SteveMacenski, 2019-11-20:
Maybe will work quick fix, change from

```
bool SlamToolbox::deserializePoseGraphCallback(
  const std::shared_ptr<rmw_request_id_t> request_header,
  const std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph::Request> req,
  std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph::Response> resp)
```

to 

```
bool SlamToolbox::deserializePoseGraphCallback(
  const std::shared_ptr<rmw_request_id_t> /*request_header*/,
  const std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph::Request> req,
  std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph::Response> resp)
```

I doubt it'll fix it, but worth a quick shot. Otherwise the solution to this is to make another deserialize method that isn't registered as a callback (assuming that you're right and its not the missing `make_shared<>()`

SteveMacenski, 2019-11-20:
https://github.com/SteveMacenski/slam_toolbox/pull/127

@maxlein give this a shot, it should fix your issues

maxlein, 2019-11-21:
It fixes this seg fault but leads to another one 💣  

```
Mapper <- Module
Mapper <- m_pSequentialScanMatcher
Mapper <- m_pGraph
MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
Mapper <- m_pMapperSensorManager
MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
Mapper <- m_Listeners
**Finished serializing Mapper**
Load From File
**Serializing Dataset**
Dataset <- m_SensorNameLookup
Dataset <- m_Data
Dataset <- m_Lasers
Dataset <- m_pDatasetInfo
**Finished serializing Dataset**

Thread 1 "localization_sl" received signal SIGSEGV, Segmentation fault.
0x00007ffff7863fc1 in slam_toolbox::SlamToolbox::loadSerializedPoseGraph(std::unique_ptr<karto::Mapper, std::default_delete<karto::Mapper> >&, std::unique_ptr<karto::Dataset, std::default_delete<karto::Dataset> >&) ()
    at /usr/include/c++/8/bits/shared_ptr_base.h:1018
1018          _M_get() const noexcept
(gdb) backtrace
#0  0x00007ffff7863fc1 in slam_toolbox::SlamToolbox::loadSerializedPoseGraph(std::unique_ptr<karto::Mapper, std::default_delete<karto::Mapper> >&, std::unique_ptr<karto::Dataset, std::default_delete<karto::Dataset> >&) ()
    at /usr/include/c++/8/bits/shared_ptr_base.h:1018
#1  0x00007ffff7867a66 in slam_toolbox::SlamToolbox::deserializePoseGraphCallback(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph_Request_<std::allocator<void> > >, std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph_Response_<std::allocator<void> > >) () at /home/kin/playground_ws/src/slam_toolbox/src/slam_toolbox_common.cpp:835
#2  0x00007ffff7ba445f in slam_toolbox::LocalizationSlamToolbox::deserializePoseGraphCallback(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph_Request_<std::allocator<void> > >, std::shared_ptr<slam_toolbox::srv::DeserializePoseGraph_Response_<std::allocator<void> > >) () at /usr/include/c++/8/ext/atomicity.h:96
#3  0x00007ffff7ba47e8 in slam_toolbox::LocalizationSlamToolbox::loadPoseGraphByParams() () at /usr/include/c++/8/bits/shared_ptr_base.h:614
#4  0x000055555559f6e9 in main () at /home/kin/playground_ws/src/slam_toolbox/src/slam_toolbox_localization_node.cpp:46
#5  0x00007ffff6041b97 in __libc_start_main (main=0x55555559f360 <main>, argc=2, argv=0x7fffffffc608, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffc5f8) at ../csu/libc-start.c:310
#6  0x000055555559faba in _start () at /home/kin/playground_ws/src/slam_toolbox/src/slam_toolbox_localization_node.cpp:32

```

SteveMacenski, 2019-11-21:
Please post your config file & serialized files

maxlein, 2019-11-21:
There you go:  
[map.tar.gz](https://github.com/SteveMacenski/slam_toolbox/files/3875356/map.tar.gz)

I am used to problems with ROS2 ;) I was suprised mapping worked that good :)

SteveMacenski, 2019-11-21:
Also you’re free to take a look and see if you can find the issue ;-)

I don’t immediately know the cause of that one

maxlein, 2019-11-21:
> Edit: it may be related to nullptr being passed in rather than the shared ptr of the rwm header template.

Tried that already, and didn't change output. removed it completly as its never used anyway.

SteveMacenski, 2019-11-21:
Well you cant remove it. That’s needed for the ROS service sharing the Cb.

SteveMacenski, 2019-11-22:
Found the issue, I added the load graph before the solver plugin was loaded so its an uninitialized shared ptr. Fixing now.

SteveMacenski, 2019-11-22:
I updated the branch/PR. Let me know if you run into any other crashes

maxlein, 2019-11-22:
And I wake up and get bugs fixed over night ;)  

Your fix works, but then you forgot to spin while waiting for the scan msg.  
`rclcpp::spin_some(this->get_node_base_interface());`  

So now my last logs are  

````
[slam_toolbox-8] [INFO 1574412190.336280846] [slam_toolbox]: Waiting for incoming scan to get metadata...
[slam_toolbox-8] Registering sensor: [Custom Described Lidar]
```
But nothing is going on now, no map published, no reaction on /initialpose call.

SteveMacenski, 2019-11-22:
Try just replacing ‘[scan]’ with ‘[&]’ and scan = msg; in the lambda. I didnt pass by reference

maxlein, 2019-11-22:
> Try just replacing ‘[scan]’ with ‘[&]’ and scan = msg; in the lambda.

Already did that, added also log output but lambda is never called.
Only the other subscriber i mentioned.

Will try to get a bag tomorrow. Didnt try that yet in ros2...

SteveMacenski, 2019-11-22:
@maxlein the issue was just spinning (plus lambda mess ups). See PR. The spinning would also let scans come in through the other method, however we have the smapper mutex locked up so this one will be processed first and once its done the queue of new scans will firehose in.

SteveMacenski, 2019-11-22:
Ok **now** i'll be impressed if something messes up

SteveMacenski, 2019-11-22:
Merged updates since they objectively fix some issues - let me know if you run into any other trouble

maxlein, 2019-11-23:
> however we have the smapper mutex locked

Still not working for me, it seems to hang in the addScan function at the mutex.  

Trying to make a bag, but bag recording is also not working... Some sqlite3 storage error.. give me some time...

maxlein, 2019-11-23:
Got it working:  
[rosbag2_2019_11_23-19_15_28.tar.gz](https://github.com/SteveMacenski/slam_toolbox/files/3882790/rosbag2_2019_11_23-19_15_28.tar.gz)

SteveMacenski, 2019-11-23:
What’s hanging exactly?

I’ve shown while publishing an empty laser scan that it gets called with the subscriber and tries to create the laser object (I get errors at that point bc its a fake laser scan and there’s no real TF tree to transform, but its being received and worked with). Are you saying that once the laser scan is registered in the while loop you see that no-TF-transformation error, the loop spins forever, or after the deserialization is done the main processing is blocked?

maxlein, 2019-11-24:
I never see `RCLCPP_INFO(get_logger(), "Waiting for incoming scan to get metadata...");`  
It gets till line 763 `rclcpp::spin_some(this->get_node_base_interface());`  

This call blocks. Maybe it goes like that:  

`loadSerializedPoseGraph` locks `smapper_mutex_`, calling spin processes scans which in return calls `addScan` which also wants the `smapper_mutex_` and now we are waiting forever ...

SteveMacenski, 2019-11-24:
Mhm. I do see how that could be a problem, and I wouldnt have seen it how I tested because I didnt have valid TF for it to get through the message filter. 

I’ll have to think about this then. Probably have to introduce a new class member to cache the last scan

SteveMacenski, 2019-11-25:
@maxlein try this: https://github.com/SteveMacenski/slam_toolbox/tree/cb_group

This branch removes the scan callback. I think this should work fine. If it doesn't I know the way to definitely fix it but its less clean so I'd rather this work out.

Essentially rather than adding the laser object once to the dataset, we add it twice and overrided the second time in the case of the same sensor. For different sensors they're loaded once each. I liked the idea of new data being required before continuing but I think this is equivalent. 

if you don't find this works, I can fix it the hard way

maxlein, 2019-11-26:
Well, it's running! I have a map.  
Thanks for the help and fixes!  

One thing I don't get right now:  
I have `mode: localization` set, but it's still mapping. In README you say it's just performance optimizations, so maybe there is no "localization only" mode?  

On the picture you see that I started the toolbox without a starting pose ( 0,0,0 ) and then localized it.  

[image]

SteveMacenski, 2019-11-26:
The mode is just for resource optimization in the optimizer. It doesn’t effect which node you utilize (whether you’re in the localization node or one of the mapping nodes). 

If you’re actually in the localization node, move around, you should see those old scans disappear. See readme/roscon/ros developers podcast talk for more details.

SteveMacenski, 2019-11-26:
#132 #133

closed 2019-11-26 by SteveMacenski