SteveMacenski/slam_toolbox #140: drift in mapping mode during rotation

Benjaminmar8, 2019-12-05:
Thank you for the great package, I have been experimenting your package in the past week. I was able to produce some results, but there are some issues that I can resolve no matter how i tune the parameters. I have a robot  installed with P&F 360 degree LIDAR, the  odometry pof the robot is bad. I am using online async mapping mode. When the robot is moving straight, it is generally ok, but whenever the robot starts to rotate (yaw), the scan often cannot match back to the map. Please see the pics for illustration. The result is after rotation, the robot will end up at a wrong pose, and the map produced is also wrong (last pic). How can I tune parameters to handle the bad odometry case? Thanks a lot for the help. 
[image]
[image]
[image]

SteveMacenski, 2019-12-05:
Can you give me a bag file and your parameters? What’s the laser?

This is now the second time I’ve heard from someone with a full 360 degree lidar having a similar issue. I wonder if I have a 360 lidar bug or if drivers aren’t using the right notation. 

Does this work with other slam systems (ei did you try this on Karto? If not, try it and let me know)? Want to separate issues with potentially the robot from the lidar from this package.

Benjaminmar8, 2019-12-05:
You can download the bagfile here: https://www.dropbox.com/s/gkyps2sogutckht/record.bag?dl=0
The LIDAR used is Pepperl & Fuchs R2000, range 30 meters, 23 Hz scan rate, 360 degree, 0.1 degree angular resolution.

The LIDAR scan had a bit of issue with karto library, karto produced one error "LaserRangeScan contains 3600, range readings, expected 3601" and mapping can not be started. So I modified the code (a bit hackish I admit that), in carto.h, update() function, I added 
```
if(GetType()==0)
	m_NumberOfRangeReadings = static_cast<kt_int32u>(math::Round((GetMaximumAngle() - GetMinimumAngle()) / GetAngularResolution()));
```
to just get rid of the error so that at least mapping can start. 

I also have a static tf between base_link and lidar frame, the tf between them is <0 0 0 3.14159 0 0>. For this bag file, I have verified using other slam systems such as carto. Its performance is fine. Thanks a lot for looking into this.

maxlein, 2019-12-05:
I also mapped with a R2000 and it did work quite well but my robots odometry is pretty nice.  
Your tf tree is correct?

Benjaminmar8, 2019-12-05:
tf should be ok as it is tested with other slam systems.

SteveMacenski, 2019-12-05:
Please provide a link to your actual yaml file. 

@maxlein suggestion makes me believe then it may be in your setup. That error you commented out also isn't probably a good sign. Not all SLAM algorithms are going to take into account the laser's pose relative to the robot frame as much as this does. 

You mentioned trying Karto, what were your results? 
Have you tried gmapping? 

Your problems seem consistent with @maxlein ticket here https://github.com/SteveMacenski/slam_toolbox/issues/118. This may be an issue with this specific lidar driver or the TF around it/robot

maxlein, 2019-12-05:
My tests in #118 were done with dashing and opensplice which led to hickups in the running system. I may try something tomorrow.

Benjaminmar8, 2019-12-06:
Here is my yaml file:
`# Plugin params
solver_plugin: solver_plugins::CeresSolver
ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
ceres_preconditioner: SCHUR_JACOBI
ceres_trust_strategy: LEVENBERG_MARQUARDT
ceres_dogleg_type: TRADITIONAL_DOGLEG
ceres_loss_function: HuberLoss 

# ROS Parameters
odom_frame: odom
map_frame: map
base_frame: base_link 
scan_topic: /scan 
mode: mapping 

debug_logging: true
throttle_scans: 1
transform_publish_period: 0.02 
map_update_interval: 5.0
resolution: 0.05
max_laser_range: 30.0 
minimum_time_interval: 0.5
transform_timeout: 0.2
tf_buffer_duration: 30.
stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps
enable_interactive_mode: true

# General Parameters
use_scan_matching: true
use_scan_barycenter: true 
minimum_travel_distance: 0.5 
minimum_travel_heading: 0.5 
scan_buffer_size: 200 # default 10
scan_buffer_maximum_scan_distance: 30 #default 10
link_match_minimum_response_fine: 0.1  #default 0.1
link_scan_maximum_distance: 1.5
loop_search_maximum_distance: 3.0
do_loop_closing: true 
loop_match_minimum_chain_size: 10           
loop_match_maximum_variance_coarse: 3.0  
loop_match_minimum_response_coarse: 0.35    
loop_match_minimum_response_fine: 0.45

# Correlation Parameters - Correlation Parameters
correlation_search_space_dimension: 0.5      
correlation_search_space_resolution: 0.01     
correlation_search_space_smear_deviation: 0.1 

# Correlation Parameters - Loop Closure Parameters
loop_search_space_dimension: 8.0 
loop_search_space_resolution: 0.05 
loop_search_space_smear_deviation: 0.03 

# Scan Matcher Parameters
distance_variance_penalty: 0.5   
angle_variance_penalty: 1.0 

fine_search_angle_offset: 0.00349  
coarse_search_angle_offset: 0.349   
coarse_angle_resolution: 0.0349      
minimum_angle_penalty: 0.9
minimum_distance_penalty: 0.5
use_response_expansion: true`

I did test on Cartographer, it is fine for both mapping and localization methods.

SteveMacenski, 2019-12-06:
What were your results with karto and gmapping, specifically. I dont know cartographer.

Benjaminmar8, 2019-12-06:
here is my result of cartographer, when the robot is rotating, the lidar scan matches with the map too. 
[image]

SteveMacenski, 2019-12-06:
Again, I'm not asking about cartographer

Benjaminmar8, 2019-12-07:
I tried on gmapping, the final result is ok as shown in the pic below, but when the robot is rotating, I can see the scan did not match the map, it is only matched back after the robot moved for some time. 
[image]

#118 is a different issue, the user has an inverted LIDAR, this is not my case. I guess it is odom issue, as my odom drifts a lot. Are there any parameter that i can tune to mitigate the effect of bad odom?  Thanks a lot.

SteveMacenski, 2019-12-11:
Add a huber loss function if its bad, you may want to update the map more often (lower the threshold distance and angle to update) so there's less distance to drift between updates (but dont go too wild)

but really all the parameters 35-70 have an effect for poor odometry. the Penalty parameters will add a penalty for more distance or angle traversed to find a match, so you may want to relax those. The smear parameters will "smear" the probability distribution around the cells of readings to help smooth out a response, increase that _slightly_.  Lower the search resolution, etc. There's no one shot magic thing.

SteveMacenski, 2019-12-11:
@Benjaminmar8 can you link to a zip or tar of the bag file? I cannot access dropbox.

Benjaminmar8, 2019-12-12:
You may download it here: [https://drive.google.com/file/d/1BSwPJjnwHfuHvwLyinHLpE6SExUsRihx/view?usp=sharing](url)
I think to have a is_360_lidar option is great,  maybe after I solve the rotation drift issue. Thanks for looking into the problem.

SteveMacenski, 2019-12-12:
Drive is going to have the same issues. I need the actual files.

Benjaminmar8, 2019-12-12:
Please see if you can download as an attachment. Thank you.
 record.tar.xz
<https://drive.google.com/file/d/1W5ROHSE29aS3YOSprYpaatk2hVkmvQ7T/view?usp=drive_web>

SteveMacenski, 2019-12-17:
The video below is showing the laser scan and the robot position in the odometry frame. This is without SLAM or any external corrections. As you can see, you're drifting quite alot while turning. That is a pretty resounding answer to the "what's broken here?" question. I don't know how you're computing it, but if you're using a filter, you should really tune that filter. If this is something you've built yourself, I'm a little suspect of your wheel radius and bias calculation for the turning angle calculation. It seems like whenever you rotate it reacts really really poorly. 

[image]

I am relatively surprised with how well it actually works in correcting for those issues with default settings, as you can see below. I don't see any issues at all until the very very end when you rotate very fast then back up. Again, this is all to do with your odometry. 

[image]

Given these results, I'm inclined to close the ticket since the issue isn't in the software package, but your robot's odometric estimation. However, it may be possible to tune this package to work better for your issue, but I think your time is better spent fixing the issue that will effect other systems as well.

closed 2019-12-17 by SteveMacenski

maxlein, 2020-01-16:
@Benjaminmar8 do you have your r2000 working now?  
And do you have filtering enbaled on your scanner?  
Because I have, and then mapping is complaining `[slam_toolbox-1] LaserRangeScan contains 2937 range readings, expected 3601`.

Benjaminmar8, 2020-01-21:
@maxlein I have to hack the code a bit in order to get it working. I did not enable filtering.