SteveMacenski/slam_toolbox #143: Dynamic Map Size and Costmap Resizing for Lifelong Mapping

kw90, 2019-12-17:
Hi Steve and Contributors,

Thanks for your work on offering this awesome toolbox. Since some time now I have been working on the navigation for the [Mobility Base from Dataspeed](https://robots.ros.org/mobility-base/) which is working really good on "static" maps built with Cartographer, Gmapping, Hector SLAM. However, I discovered the Slam Toolbox a few months back and experimented with it, as I was searching for a robust solution for lifelong mapping and laser odometry in hopes to get a better final pose accuracy. This package offers to kill two birds with one stone. :)

Now i have read that the lifelong mapping is highly experimental ATM, but I think my problem has a different cause. Problematic for me is the global costmap when incrementally building a map as seen in these [screenshots](https://imgur.com/a/HlrnhKc). Use case for this would be e.g. exploration of a new environment. The problem seems only to occur to the global costmap, when the map is growing in size.

As can be seen in the last screenshot and in the output from `move_base` below, it tries resizing the static layer to the new map size. Sometimes it does that successfully but most of the times the global costmap gets corrupted as seen in the penultimate and ultimate screenshots.

```zsh
[ INFO] : Resizing static layer to 129 X 111 at 0.050000 m/pix
[ INFO] : Resizing static layer to 129 X 110 at 0.050000 m/pix
```
Once the full map is built and not growing in size anymore, this error does not happen anymore. 

In an example video from the `frontier_exploration` ROS package on [youtube](https://youtu.be/WMcQRpBpN1s) it can be seen that the global costmap is much bigger in size than the actual map. I have also observed this behavior with Hector and Gmapping SLAM.

1. Do you think this is related to the map size?
2. Does this have to be fixed in `slam_toolbox` or is this a misconfiguration of `move_base` from my side?

The used launch and configuration files for `move_base` and `slam_toolbox` can be found on [my GitHub group](https://github.com/mt-kawa).

If you read that far, thank you and I am looking forward to reading your feedback. :)
If not:
TL;DR: [screenshots](https://imgur.com/a/HlrnhKc) say more than a thousand words.

Cheers

SteveMacenski, 2019-12-17:
You should direct the question to ROS answers

But it doesn’t appear anything is wrong. You’re building a map while navigating, that’s not a warning, its an info, totally fine.

closed 2019-12-17 by SteveMacenski

kw90, 2019-12-17:
Thanks for that quick reply :+1: 

I think you're totally right - it's probably a `move_base` related issue. Will try my luck over at ROS answers.