SteveMacenski/slam_toolbox #146: Publishing Pose Topic

David9696, 2020-01-11:
Hi,

Thanks for your great package. Is it possible to have pose topic of the robot being published in localization mode? Something like amcl_pose (geometry_msgs/PoseWithCovarianceStamped) in amcl package.

Thanks.

SteveMacenski, 2020-01-12:
I didn't add it intentionally since I think it was really misused in ROS1 AMCL as a provider of state estimation rather than the debug / signal of an update that it was meant to be.

However, if you had a genuine need for it, I wouldn't be opposed to approving a PR implementing it.

David9696, 2020-01-14:
Hi,

Thanks for your reply. Normally, I use the topic to keep track the movement of the robot.

I am not very good in programming as i am still a beginner. However, I would be more than happy to help you stress test the package if there is any update. 

Once again, thanks for your great package.

SteveMacenski, 2020-01-14:
Feel free to submit a PR

SteveMacenski, 2020-01-21:
@David9696 is this something you're motivated to submit a PR for?

David9696, 2020-01-22:
Hi,

I am interested to work on this. But, I need to spend some time to understand the whole picture. I am thinking to work on it by using the tf transform and remap.  Then, publish the pose as a topic. Working on a node that is independent of this package so that I would not screw up the whole package. Anyway, I am still learning programming and ROS. This may take a while. Still hope I can contribute a bit.

Thanks.

SteveMacenski, 2020-01-22:
You could certainly use another node to just have a TF listener that publishes it.

Another option is you can in https://github.com/SteveMacenski/slam_toolbox/blob/eloquent-devel/src/slam_toolbox_common.cpp#L552 take the pose and just publish it to `/current_pose` or something after the reprocessing transform. 

The downside to the SLAM Toolbox option is that the current pose will only be updated when the slam problem is updated (after some time, rotation, or translation thresholds are met) so they will not be consistent at `N` Hz. The first option will be more consistent in timing.

Generally, I'd say that adding a new node would be a good idea, but if you just need the `current_pose` as a general awareness that the transformation has been updated or general awareness of where you are, I don't think that's a bad thing to add.

closed 2020-05-13 by SteveMacenski