SteveMacenski/slam_toolbox #147: Question on tracking distiguished locations while mapping

blablebli-robots, 2020-01-13:
Hello,

Excellent job!

I'd like to know if there are any facilities already to track selected locations during mapping to be used later for navigation purposes. My concern is that after loop closures/optimizations these selected points will change coordinates due to pose optimizations and possible loop closures.

If no such facilities exist, and if at all possible, Could you please point me at right location(s) of the code I can adapt to start tracking selected, say robot estimated poses at given time t.

Cheers

SteveMacenski, 2020-01-15:
Hi,

That seems like a pretty involved feature. Is there a reason you want to mark points while mapping rather than waiting until complete to mark?

This is creating a graph of sensor constraints and optimizing over it. I think what you may be proposing could be technically implemented as artificial nodes with static constraints relative to local nodes with your metadata you’d like to store in them as a “marked point”. I think you’d find that challenging, but technically possible, though you’d need to find a way to distinguish between real nodes and decorator nodes just storing info. 

PRs welcome ;-)

SteveMacenski, 2020-01-17:
Any other questions on this?

SteveMacenski, 2020-01-21:
Will reopen if any other topics for discussion.

closed 2020-01-21 by SteveMacenski