SteveMacenski/slam_toolbox #148: Is there a reason for non namespaced topics?

maxlein, 2020-01-15:
Is there a reason why some topics don't take the nodes namespace if there is one?  
Like map, initialpose..

SteveMacenski, 2020-01-15:
Because they are traditionally global topics across many implementations. Not much other reason. Makes this drop in replaceable.

closed 2020-01-15 by SteveMacenski

maxlein, 2020-01-16:
Ok, because amcl is taking the namespace, so not drop in replacable.

SteveMacenski, 2020-01-16:
That's a bug then in whatever version of AMCL you're using. What distro are you on?

maxlein, 2020-01-17:
I am on eloquent-devel 6442c35, was in dashing also.  

I start node like this:  
```  
            Node(
            package='nav2_amcl',
            node_executable='amcl',
            node_name='amcl',
            node_namespace=LaunchConfiguration('namespace'),
            output='screen',
            parameters=[ ...
```

Topics: initialpose and map, basically I think it's every pub/sub topic.  
Also scan topic has no parameter, but that could also be as intended and one has to remap the topic before.

SteveMacenski, 2020-01-17:
```
s.macenski@reese:~/Documents/nav/navigation2_ws$ ros2 topic list 
/amcl/transition_event
/amcl_pose
/bt_navigator/transition_event
/cmd_vel
/controller_server/transition_event
/cost_cloud
/evaluation
/global_costmap/costmap
/global_costmap/costmap_raw
/global_costmap/costmap_updates
/global_costmap/footprint
/global_costmap/global_costmap/transition_event
/global_costmap/parameter_events
/global_costmap/published_footprint
/global_costmap/rosout
/goal_pose
/initialpose
/local_costmap/costmap
/local_costmap/costmap_raw
/local_costmap/costmap_updates
/local_costmap/footprint
/local_costmap/local_costmap/transition_event
/local_costmap/parameter_events
/local_costmap/published_footprint
/local_costmap/rosout
/local_plan
/map
/map_server/transition_event
/marker
/odom
/parameter_events
/particlecloud
/plan
/planner_server/transition_event
/received_global_plan
/recoveries_server/transition_event
/rosout
/scan
/tf
/tf_static
/transformed_global_plan
/waypoint_follower/transition_event
```

Inside of the default navigation2 bringups, I dont see that behavior

maxlein, 2020-01-17:
the default bringups have no namespace.
I run the nodes in a robot namespace to distinguish between them.

SteveMacenski, 2020-01-17:
Ok, I think that answers your question then -- you should remap the launch file for your robot namespace

maxlein, 2020-01-19:
I think we are talking past each other :)  
> you should remap the launch file for your robot namespace  

If I understand that correctly, then I am doing that already.  

I try a last time to describe the problem otherwise I fix it internally.

I have a launch file, and in there a namespace is passed to all nodes run by the launch file.  
And in this launch file I can switch between running amcl and ST.  
So both nodes started with same namespace, parameters, etc...  

But when looking at `initialpose `and `map` topic, they are not drop in replacable.  

Launch file run with amcl activated:  
```
ros2 topic list
/clock
/osg_1/amcl/transition_event
/osg_1/amcl_pose
/osg_1/bt_navigator/transition_event
/osg_1/cmd_vel
/osg_1/controller_server/transition_event
/osg_1/cost_cloud
/osg_1/custom_camera/custom_points
/osg_1/evaluation
/osg_1/global_costmap/costmap
/osg_1/global_costmap/costmap_raw
/osg_1/global_costmap/costmap_updates
/osg_1/global_costmap/footprint
/osg_1/global_costmap/global_costmap/transition_event
/osg_1/global_costmap/parameter_events
/osg_1/global_costmap/published_footprint
/osg_1/global_costmap/rosout
/osg_1/global_costmap/voxel_grid
/osg_1/goal_pose
/osg_1/initialpose
/osg_1/local_costmap/costmap
/osg_1/local_costmap/costmap_raw
/osg_1/local_costmap/costmap_updates
/osg_1/local_costmap/footprint
/osg_1/local_costmap/local_costmap/transition_event
/osg_1/local_costmap/parameter_events
/osg_1/local_costmap/published_footprint
/osg_1/local_costmap/rosout
/osg_1/local_costmap/voxel_grid
/osg_1/local_plan
/osg_1/map
/osg_1/map_server/transition_event
/osg_1/marker
/osg_1/odom
/osg_1/parameter_events
/osg_1/particlecloud
/osg_1/plan
/osg_1/planner_server/transition_event
/osg_1/received_global_plan
/osg_1/recoveries_server/transition_event
/osg_1/rosout
/osg_1/scan
/osg_1/transformed_global_plan
/osg_1/transformed_virtual_complete_path
/osg_1/transformed_virtual_path
/osg_1/waypoint_follower/transition_event
/parameter_events
/rosout
/tf
/tf_static

```

Launch file run with ST activated:    
```
ros2 topic list
/clock
/initialpose
/map
/map_metadata
/osg_1/bt_navigator/transition_event
/osg_1/cmd_vel
/osg_1/controller_server/transition_event
/osg_1/cost_cloud
/osg_1/custom_camera/custom_points
/osg_1/evaluation
/osg_1/global_costmap/costmap
/osg_1/global_costmap/costmap_raw
/osg_1/global_costmap/costmap_updates
/osg_1/global_costmap/footprint
/osg_1/global_costmap/global_costmap/transition_event
/osg_1/global_costmap/parameter_events
/osg_1/global_costmap/published_footprint
/osg_1/global_costmap/rosout
/osg_1/global_costmap/voxel_grid
/osg_1/goal_pose
/osg_1/graph_visualization
/osg_1/local_costmap/costmap
/osg_1/local_costmap/costmap_raw
/osg_1/local_costmap/costmap_updates
/osg_1/local_costmap/footprint
/osg_1/local_costmap/local_costmap/transition_event
/osg_1/local_costmap/parameter_events
/osg_1/local_costmap/published_footprint
/osg_1/local_costmap/rosout
/osg_1/local_costmap/voxel_grid
/osg_1/local_plan
/osg_1/map
/osg_1/marker
/osg_1/odom
/osg_1/parameter_events
/osg_1/plan
/osg_1/planner_server/transition_event
/osg_1/received_global_plan
/osg_1/recoveries_server/transition_event
/osg_1/rosout
/osg_1/scan
/osg_1/transformed_global_plan
/osg_1/transformed_virtual_complete_path
/osg_1/transformed_virtual_path
/osg_1/waypoint_follower/transition_event
/parameter_events
/rosout
/tf
/tf_static
```