SteveMacenski/slam_toolbox #151: New Default SLAM In ROS2: Maintainership, Process, Release, Testing, Oh My!

SteveMacenski, 2020-01-16:
Per discussions in the official ROS2 TSC Navigation working group (https://github.com/ros-planning/navigation2/issues/1389), Matt and I with input from the rest of the working group and navigation community have decided to make this project the new default ROS2 SLAM implementation to ship with ROS2 and Navigation.

As part of this, we need to create more formality in process and maintainership moving forward to make sure its not a one-man show of Steve.

On this discussion, we need the following:
- [ ] Additional maintainers with expressed interest and use of SLAM Toolbox. This should include at least 1 additional company using SLAM Toolbox and a member of OSRF with administration rights in case other maintainers are needing to be added due to maintainers abandoning the project. 
- [ ] Process around reviewing and merging pull requests and issue tickets
- [ ] Process or consistency around the release cycles and careful not to break API or functionality after a distribution has been released
- [ ] Adding additional testing capabilities for units and integration to increase the reliability of PRs behind builds and checking some basic math.

LeroyR, 2020-01-31:
Consider to Split the package into e.g. msg and viz.

This enables just using messages (or rviz) from other hosts without building the code or not building the rviz plugins for hosts without display.

SteveMacenski, 2020-01-31:
PRs always appreciated :-)

LeroyR, 2020-01-31:
I could try, but would do it on the melodic branch if thats ok?

SteveMacenski, 2020-03-05:
Closing, opened a ticket for the remaining ROS2 work, but its no longer motivational on this higher level task.

closed 2020-03-05 by SteveMacenski

safdark, 2021-05-26:
> Closing, opened a ticket for the remaining ROS2 work, but its no longer motivational on this higher level task.

@SteveMacenski Would you mind posting the new ticket(s) so viewers could follow along this trail?