SteveMacenski/slam_toolbox #154: Dropping previously mapped area

thack5, 2020-01-20:
I am really hoping that this an 'I missed something issue'. I am running a RPLIDAR S1 (10HZ) and trying to map an area. As I move around the map looks great but as soon as I move too far away from my starting position the area that I have previously mapped disappears from my map. As I come back to the original location the robot begins mapping the area all over again and then drops where I just came from. I have the param mode set to mapping and I have no idea why it keeps dropping the map. The robot position and map to odom transform don't seem to be affected by the mapped area disappearing. Any help would be great.

SteveMacenski, 2020-01-20:
What is your launch file, parameters, etc

thack5, 2020-01-20:
Launch File
<launch>

  <node pkg="tf" type="static_transform_publisher" name="world_to_map"
    args="0.0 0.0 0.0 0.0 0.0 0.0 /world /map 1000" />

  <node pkg="tf" type="static_transform_publisher" name="odom_to_base"
    args="0.0 0.0 0.0 0.0 0.785398 0.0 /camera_pose_frame /base_link 1000" />

  <node pkg="tf" type="static_transform_publisher" name="base_to_laser"
    args="0.0 0.0 0.0 0.0 0.0 0.0 /base_link /laser 1000" />

<node pkg="laser_filters" type="scan_to_scan_filter_chain" name="laser_filter">
    <rosparam command="load" file="/opt/ros/melodic/lib/laser_filters/my_laser_filter.yaml" />
</node>

  <node name="rplidarNode"          pkg="rplidar_ros"  type="rplidarNode" output="log">
  <param name="serial_port"         type="string" value="/dev/rplidar"/>  
  <param name="serial_baudrate"     type="int"    value="256000"/>
  <param name="frame_id"            type="string" value="laser"/>
  <param name="inverted"            type="bool"   value="false"/>
  <param name="angle_compensate"    type="bool"   value="true"/>
  </node>

  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default="t265"/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>

  <arg name="fisheye_width"       default="848"/> 
  <arg name="fisheye_height"      default="800"/>
  <arg name="enable_fisheye1"     default="true"/>
  <arg name="enable_fisheye2"     default="true"/>

  <arg name="fisheye_fps"         default="30"/>

  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="62"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_sync"           default="false"/>

  <arg name="linear_accel_cov"      default="0.01"/>
  <arg name="initial_reset"         default="false"/>
  <arg name="unite_imu_method"      default="linear_interpolation"/>

  <arg name="publish_odom_tf"     default="true"/>
  
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_sync"              value="$(arg enable_sync)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye1"          value="$(arg enable_fisheye1)"/>
      <arg name="enable_fisheye2"          value="$(arg enable_fisheye2)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>

      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>

	  
    </include>

  </group>

  
<node pkg="rviz" type="rviz" name="rviz"/>

  <node pkg="slam_toolbox" type="localization_slam_toolbox_node" name="slam_toolbox" output="screen">
    <rosparam command="load" file="$(find slam_toolbox)/config/mapper_params_localization_v1.yaml" />
  </node>

</launch>

thack5, 2020-01-20:
param file

# link_match_minimum_response_fine: 0.4 
# link_scan_maximum_distance: 20.0
# loop_search_maximum_distance: 50.0
# distance_variance_penalty: 1.8             
# angle_variance_penalty: 0.20    
# minimum_angle_penalty: 1.6 #0.9

# Plugin params
solver_plugin: solver_plugins::CeresSolver
ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
ceres_preconditioner: SCHUR_JACOBI
ceres_trust_strategy: LEVENBERG_MARQUARDT
ceres_dogleg_type: TRADITIONAL_DOGLEG
ceres_loss_function: None

# ROS Parameters
odom_frame: camera_odom_frame
map_frame: map
base_frame: base_link
scan_topic: /scan_filtered
mode: continued mapping #localization

# if you'd like to start localizing on bringup in a map and pose
#map_file_name: test_steve
#map_start_pose: [5.0, 1.0, 0.0]

debug_logging: false
throttle_scans: 1
transform_publish_period: 0.02 #if 0 never publishes odometry
map_update_interval: 5.0
resolution: 0.05
max_laser_range: 40.0 #for rastering images
minimum_time_interval: 0.5
transform_timeout: 0.2
tf_buffer_duration: 30.
stack_size_to_use: 800000000 #// program needs a larger stack size to serialize large maps

# General Parameters
use_scan_matching: true
use_scan_barycenter: true
minimum_travel_distance: 0.5
minimum_travel_heading: 0.5
scan_buffer_size: 10
scan_buffer_maximum_scan_distance: 10
link_match_minimum_response_fine: 0.1  
link_scan_maximum_distance: 1.5
loop_search_maximum_distance: 3.0
do_loop_closing: true 
loop_match_minimum_chain_size: 10           
loop_match_maximum_variance_coarse: 3.0  
loop_match_minimum_response_coarse: 0.35    
loop_match_minimum_response_fine: 0.45

# Correlation Parameters - Correlation Parameters
correlation_search_space_dimension: 0.5
correlation_search_space_resolution: 0.01
correlation_search_space_smear_deviation: 0.1 

# Correlation Parameters - Loop Closure Parameters
loop_search_space_dimension: 8.0
loop_search_space_resolution: 0.05
loop_search_space_smear_deviation: 0.03

# Scan Matcher Parameters
distance_variance_penalty: 0.5      
angle_variance_penalty: 1.0    

fine_search_angle_offset: 0.00349     
coarse_search_angle_offset: 0.349   
coarse_angle_resolution: 0.0349        
minimum_angle_penalty: 0.9
minimum_distance_penalty: 0.5
use_response_expansion: true

thack5, 2020-01-20:
I am using a realsense camera for odometry the camera is angled so it is pointing at the ceiling (this is due to the purpose of the robot). In a small area(one room) everything works great but as soon as i move into the hallway for example and move further from the start point my original room map just disappears.

thack5, 2020-01-20:
sorry looks like the launch file got cut off.

<launch>

  <node pkg="tf" type="static_transform_publisher" name="world_to_map"
    args="0.0 0.0 0.0 0.0 0.0 0.0 /world /map 1000" />

  <node pkg="tf" type="static_transform_publisher" name="odom_to_base"
    args="0.0 0.0 0.0 0.0 0.785398 0.0 /camera_pose_frame /base_link 1000" />

  <node pkg="tf" type="static_transform_publisher" name="base_to_laser"
    args="0.0 0.0 0.0 0.0 0.0 0.0 /base_link /laser 1000" />

<node pkg="laser_filters" type="scan_to_scan_filter_chain" name="laser_filter">
    <rosparam command="load" file="/opt/ros/melodic/lib/laser_filters/my_laser_filter.yaml" />
</node>

  <node name="rplidarNode"          pkg="rplidar_ros"  type="rplidarNode" output="log">
  <param name="serial_port"         type="string" value="/dev/rplidar"/>  
  <param name="serial_baudrate"     type="int"    value="256000"/>
  <param name="frame_id"            type="string" value="laser"/>
  <param name="inverted"            type="bool"   value="false"/>
  <param name="angle_compensate"    type="bool"   value="true"/>
  </node>

  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default="t265"/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>

  <arg name="fisheye_width"       default="848"/> 
  <arg name="fisheye_height"      default="800"/>
  <arg name="enable_fisheye1"     default="true"/>
  <arg name="enable_fisheye2"     default="true"/>

  <arg name="fisheye_fps"         default="30"/>

  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="62"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_sync"           default="false"/>

  <arg name="linear_accel_cov"      default="0.01"/>
  <arg name="initial_reset"         default="false"/>
  <arg name="unite_imu_method"      default="linear_interpolation"/>

  <arg name="publish_odom_tf"     default="true"/>
  
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_sync"              value="$(arg enable_sync)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye1"          value="$(arg enable_fisheye1)"/>
      <arg name="enable_fisheye2"          value="$(arg enable_fisheye2)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>

      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>

	  
    </include>

  </group>

  
<node pkg="rviz" type="rviz" name="rviz"/>

  <node pkg="slam_toolbox" type="localization_slam_toolbox_node" name="slam_toolbox" output="screen">
    <rosparam command="load" file="$(find slam_toolbox)/config/mapper_params_localization_v1.yaml" />
  </node>

</launch>

thack5, 2020-01-20:
here are two pictures during the same scan.
[image]
[image]

SteveMacenski, 2020-01-20:
Please provide me the exact **slam toolbox** launch file you are using. The one you posted was just a bunch of arguments for another launch file.

thack5, 2020-01-21:
Sorry here is the launch file for the slam. I will also include the yaml file and full launch file as txt docs. Sorry I am pretty new to slam toolbox so please bear with me.

  <node pkg="slam_toolbox" type="localization_slam_toolbox_node" name="slam_toolbox" output="screen">
    <rosparam command="load" file="$(find slam_toolbox)/config/mapper_params_lifelong.yaml" />
  </node>

[launch_file.txt](https://github.com/SteveMacenski/slam_toolbox/files/4088543/launch_file.txt)

[mapper_params_lifelong_yaml.txt](https://github.com/SteveMacenski/slam_toolbox/files/4088556/mapper_params_lifelong_yaml.txt)

SteveMacenski, 2020-01-21:
Please look at my previous comment. You're using the localization mode. Please read the documentation in the `README`

thack5, 2020-01-21:
That solved my issue I am so sorry to have wasted your time with a trivial error on my part.

closed 2020-01-21 by SteveMacenski