SteveMacenski/slam_toolbox #155: Multirobot cloud SLAM

abylikhsanov, 2020-01-30:
Hi,
I was reading the README and the issues related to the cloud mapping, where several robots can either deliver local maps (like in the Cartographer) or laserscans (like I guess is implemented here, based on these enhancements: #https://github.com/SteveMacenski/slam_toolbox/issues/76)

Does it mean that the multirobot SLAM using a cloud is done in a ROS fashion where we have multiple robots with unique namespaces and cloud simply subscribes to all of them and tries to SLAM? If so, don't you think that Cartographer way of making a gRPC server is better as very often cloud can run outside the LAN?

SteveMacenski, 2020-01-30:
> takes care to allow for the application of operating in the cloud

You are required to handle the cloud communications element. This was designed to run in the cloud, but the method you get data to it, is left to the user and the users application.

There are dozens of existing communication methods from a robot (client) to the cloud (server). ROS is not a suitable method so that was not directly implemented for security, reliability, speed, latency, ... reasons.

abylikhsanov, 2020-01-30:
@SteveMacenski Thanks. Are there any specific tools/methods you can advice?

SteveMacenski, 2020-01-30:
I do not use or particularly support RoboMaker. I think this is a technology decision that is best left to a trade study -- I don't think my anecdotal opinions here are productive.

abylikhsanov, 2020-01-30:
Ok but am I correct saying that multiple fleet Slam is configured by just adding the all scan topics to the config file of lifetime Slam?

SteveMacenski, 2020-01-30:
No, please read the documentation and reference my [roscon talk](https://vimeo.com/378682207)

abylikhsanov, 2020-01-30:
I did read the documentation (Readme file) and did watch your video. Unfortunately I couldn't spot where you mention the multirobot session...

But looking at this:
https://github.com/SteveMacenski/slam_toolbox/compare/multi_laser
It looks like having multiple laserscan topics under different names paces enables the multirobot session. What am I missing?

SteveMacenski, 2020-02-04:
There are other tickets open about multirobot slam.

abylikhsanov, 2020-02-04:
There are only 2 tickets, this https://github.com/SteveMacenski/slam_toolbox/issues/76 and this https://github.com/SteveMacenski/slam_toolbox/issues/68

SteveMacenski, 2020-02-04:
Yes, and multiple laser scanners are used for multiple robots, as the ticket's name provides. This capability has not been completed yet. If you're interested  in helping finish that work, the branch is shown in that ticket with a check list to complete.

closed 2020-02-04 by SteveMacenski