SteveMacenski/slam_toolbox #157: [Melodic] Localization mode crashes. pthread mutex/priority assertion failures.

kajep09, 2020-01-31:
Hi @SteveMacenski 

I've been running into some problems when using localization mode and I wanted to get your take on the errors/crashes and see if you can point me in the right direction to fix them. 

FYI I am using the current Melodic branch build on a 16.04 / Kinetic computer and  the dependencies installed fine and it build without problems.

I successfully mapped multiple times with the default parameters of `mapper_params_online_sync.yaml` except the topic names / frames of course.

However, when I try to load that posegraph file and localize with it, after a while it always crashes, but with different errors. 
For localization I used the `mapper_params_localization.yaml` file, but changed mode from mapping to localization - is this correct or is it actually supposed to be in the mapping mode?

Below are the parameters and below those I have included the 2 different crashes I experienced.

```
PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /slam_toolbox/angle_variance_penalty: 1.0
 * /slam_toolbox/base_frame: base_link
 * /slam_toolbox/ceres_dogleg_type: TRADITIONAL_DOGLEG
 * /slam_toolbox/ceres_linear_solver: SPARSE_NORMAL_CHO...
 * /slam_toolbox/ceres_loss_function: None
 * /slam_toolbox/ceres_preconditioner: SCHUR_JACOBI
 * /slam_toolbox/ceres_trust_strategy: LEVENBERG_MARQUARDT
 * /slam_toolbox/coarse_angle_resolution: 0.0349
 * /slam_toolbox/coarse_search_angle_offset: 0.349
 * /slam_toolbox/correlation_search_space_dimension: 0.5
 * /slam_toolbox/correlation_search_space_resolution: 0.01
 * /slam_toolbox/correlation_search_space_smear_deviation: 0.1
 * /slam_toolbox/debug_logging: False
 * /slam_toolbox/distance_variance_penalty: 0.5
 * /slam_toolbox/do_loop_closing: True
 * /slam_toolbox/fine_search_angle_offset: 0.00349
 * /slam_toolbox/link_match_minimum_response_fine: 0.1
 * /slam_toolbox/link_scan_maximum_distance: 1.5
 * /slam_toolbox/loop_match_maximum_variance_coarse: 3.0
 * /slam_toolbox/loop_match_minimum_chain_size: 3
 * /slam_toolbox/loop_match_minimum_response_coarse: 0.35
 * /slam_toolbox/loop_match_minimum_response_fine: 0.45
 * /slam_toolbox/loop_search_maximum_distance: 5.0
 * /slam_toolbox/loop_search_space_dimension: 8.0
 * /slam_toolbox/loop_search_space_resolution: 0.05
 * /slam_toolbox/loop_search_space_smear_deviation: 0.03
 * /slam_toolbox/map_file_name: /home/sdu/maps/20...
 * /slam_toolbox/map_frame: map
 * /slam_toolbox/map_update_interval: 5.0
 * /slam_toolbox/max_laser_range: 20.0
 * /slam_toolbox/minimum_angle_penalty: 0.9
 * /slam_toolbox/minimum_distance_penalty: 0.5
 * /slam_toolbox/minimum_time_interval: 0.5
 * /slam_toolbox/minimum_travel_distance: 0.5
 * /slam_toolbox/minimum_travel_heading: 0.5
 * /slam_toolbox/mode: localization
 * /slam_toolbox/odom_frame: odom
 * /slam_toolbox/resolution: 0.05
 * /slam_toolbox/scan_buffer_maximum_scan_distance: 10
 * /slam_toolbox/scan_buffer_size: 3
 * /slam_toolbox/scan_topic: /front_scan
 * /slam_toolbox/solver_plugin: solver_plugins::C...
 * /slam_toolbox/stack_size_to_use: 40000000
 * /slam_toolbox/tf_buffer_duration: 30.0
 * /slam_toolbox/throttle_scans: 1
 * /slam_toolbox/transform_publish_period: 0.02
 * /slam_toolbox/transform_timeout: 0.5
 * /slam_toolbox/use_response_expansion: True
 * /slam_toolbox/use_scan_barycenter: True
 * /slam_toolbox/use_scan_matching: True
```

Crash 1
```
[ INFO] [1580398406.349837695]: Node using stack size 40000000
[ INFO] [1580398406.599108463]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[ INFO] [1580398406.599190298]: Using plugin solver_plugins::CeresSolver
Load From File /home/sdu/maps/2020_01_30_sdu_lab.posegraph 
Mapper <- Module
Mapper <- m_pSequentialScanMatcher
Mapper <- m_pGraph
MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
Mapper <- m_pMapperSensorManager
MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
Mapper <- m_Listeners
**Finished serializing Mapper**
Load From File
**Serializing Dataset**
Dataset <- m_SensorNameLookup
Dataset <- m_Data
Dataset <- m_Lasers
Dataset <- m_pDatasetInfo
**Finished serializing Dataset**
Registering sensor: [Custom Described Lidar]
[ INFO] [1580398406.784702904]: Waiting for incoming scan to get metadata...
[ INFO] [1580398407.796686491]: Waiting for incoming scan to get metadata...
[ INFO] [1580398408.809902091]: Waiting for incoming scan to get metadata...
[ INFO] [1580398409.823049826]: Waiting for incoming scan to get metadata...
[ INFO] [1580398410.836324491]: Waiting for incoming scan to get metadata...
[ INFO] [1580398411.849916680]: Waiting for incoming scan to get metadata...
[ INFO] [1580398412.862915339]: Waiting for incoming scan to get metadata...
[ INFO] [1580398413.145308362]: Got scan!
localization_slam_toolbox_node: tpp.c:84: __pthread_tpp_change_priority: Assertion `new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)' failed.
[slam_toolbox-1] process has died [pid 18980, exit code -6, cmd /home/xx_ws/devel/lib/slam_toolbox/localization_slam_toolbox_node __name:=slam_toolbox __log:=b5198c22-435d-11ea-9c41-00044ba0f9e2/slam_toolbox-1.log].
log file: b5198c22-435d-11ea-9c41-00044ba0f9e2/slam_toolbox-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Crash 2
```
[ INFO] [1580398772.310853981]: Node using stack size 40000000
[ INFO] [1580398772.577551839]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[ INFO] [1580398772.577716731]: Using plugin solver_plugins::CeresSolver
Load From File /home/sdu/maps/2020_01_30_sdu_lab.posegraph 
Mapper <- Module
Mapper <- m_pSequentialScanMatcher
Mapper <- m_pGraph
MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
Mapper <- m_pMapperSensorManager
MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
Mapper <- m_Listeners
**Finished serializing Mapper**
Load From File
**Serializing Dataset**
Dataset <- m_SensorNameLookup
Dataset <- m_Data
Dataset <- m_Lasers
Dataset <- m_pDatasetInfo
**Finished serializing Dataset**
Registering sensor: [Custom Described Lidar]
[ INFO] [1580398772.805815097]: Waiting for incoming scan to get metadata...
[ INFO] [1580398772.825553988]: Got scan!
[ INFO] [1580398799.335457897]: LocalizePoseCallback: Localizing to: (-1.13 0.75), theta=-2.55
localization_slam_toolbox_node: ../nptl/pthread_mutex_lock.c:352: __pthread_mutex_lock_full: Assertion `INTERNAL_SYSCALL_ERRNO (e, __err) != ESRCH || !robust' failed.
[slam_toolbox-1] process has died [pid 19503, exit code -6, cmd /home/xx_ws/devel/lib/slam_toolbox/localization_slam_toolbox_node __name:=slam_toolbox __log:=b5198c22-435d-11ea-9c41-00044ba0f9e2/slam_toolbox-1.log].
log file: b5198c22-435d-11ea-9c41-00044ba0f9e2/slam_toolbox-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

Can you point me in the right direction or give me some clues so I can do some more controlled testing and see if I can generate a rosbag that reproduces the problem? The crashes where on an actual robot at a remote location, but I can see if I can reproduce it on some old bags or in simulation.

As I said earlier, I am unsure if using the localization mode has anything to do with the crash, or if that crash might as well have happened in mapping mode, and the reason I didn't see it is just the small sample size.

Thanks in advance!

SteveMacenski, 2020-01-31:
I haven’t seen that before, but those errors look very low level. I wonder if its related to your platform. 

What’s your architecture, OS, how have you compiled/installed STVL/Costmap2D?

Also, please compile with gdb and get a traceback. This error by itself is un-diagnosable without knowing where (or what type of thing) is throwing those assertions. 

Are these errors happening immediately or after some time running? Crashes don’t give you timestamps ;-)

SteveMacenski, 2020-02-18:
@kajep09 are you still seeing this issue or have you resolved it?

SteveMacenski, 2020-03-02:
Closing from inactivity and no other reports.

closed 2020-03-02 by SteveMacenski