SteveMacenski/slam_toolbox #162: snap: slam-toolbox optimizations

antoan, 2020-02-06:
I came across the following statement in the readme:

> For running on live production robots, I recommend using the snap: slam-toolbox, it has optimizations in it that make it about 10x faster. 

I was wondering if those optimizations are now available in the deb installs since snaps are no longer updated as mentioned in the conversations below?
https://github.com/SteveMacenski/slam_toolbox/issues/98

Alternatively, can I achieve the performance speedup if I simply generate the snap my self?

Cheers

SteveMacenski, 2020-02-06:
Its not in the debians, no. I don't fully understand how snaps sped things up. I chatted with Canonical about it and we came up with a few guesses but nothing we could easily test to figure out exactly the cause. We suspect it has to do with choosing the libraries to use (BLAS, LAPACK, SuiteSparse, etc) and because they were confined the checks took some time but were immediately rejected in snaps.

There may be some games that can be played outside of the snaps but I haven't investigated. 

You would get that if you generated the snap yourself. Every time I went to generate a new snap every few months, the snap ecosystem would change behaviors or break APIs and I'd spend days trying to get it to build (or if building, rejected from the snap store for pedantic reasons). I mostly removed support for snaps because it was taking so much time and it didn't seem that anyone other than me were using them

antoan, 2020-02-07:
>  I'd spend days trying to get it to build (or if building, rejected from the snap store for pedantic reasons)
That sounds like living hell (or the secular equivalent) , so sorry you had to go through that. Snaps sound like they are more trouble than they are worth at the present moment.

Thank you for your insight and for slam toolbox, I really appreciate it.

I was trying to ho

closed 2020-02-07 by antoan

SteveMacenski, 2020-02-07:
> Snaps sound like they are more trouble than they are worth at the present moment.

I agree. I saved so much energy and frustration when I left them behind. 

Thanks!