SteveMacenski/slam_toolbox #169: Message Filter dropping message

cblesing, 2020-02-21:
Hello,

we are trying to get the slam toolbox running in combination with ros2. Unfortunately we receive an Info message and we have no clue whats going wrong. The message is:

    ros2 launch slam_toolbox online_sync_launch.py 
    [INFO] [launch]: All log files can be found below /home/admin/.ros/log /2020-02-21-12-32-13-316458-CTV10-29494
    [INFO] [launch]: Default logging verbosity is set to INFO
    [INFO] [slam_toolbox-1]: process started with pid [29505]
    [slam_toolbox-1] [INFO] [ctv_10.slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
    [slam_toolbox-1] [INFO] [ctv_10.slam_toolbox]: Message Filter dropping message: frame 'ctv_10/laser_front_link' at time 1582284733.680 for reason 'Unknown'
    [slam_toolbox-1] [INFO] [ctv_10.slam_toolbox]: Message Filter dropping message: frame 'ctv_10/laser_front_link' at time 1582284733.730 for reason 'Unknown'`

Our setup: First of all, our robot is running in a namespace called "/ctv_xx/. We have several ros1 nodes that are not ported to ros2 at the moment. They are responsible for publishing laserscans and odometry sensor information. Therefore we're using the ros1_bridge. The bridge works fine. Here's the bridge output:

    ros2 run ros1_bridge dynamic_bridge --bridge-all-topics
    created 1to2 bridge for topic '/ctv_10/back_scan' with ROS 1 type 'sensor_msgs/LaserScan' and ROS 2 type 'sensor_msgs/msg/LaserScan'
    created 1to2 bridge for topic '/ctv_10/front_scan' with ROS 1 type 'sensor_msgs/LaserScan' and ROS 2 type 'sensor_msgs/msg/LaserScan'
    created 1to2 bridge for topic '/ctv_10/map' with ROS 1 type 'nav_msgs/OccupancyGrid' and ROS 2 type 'nav_msgs/msg/OccupancyGrid'
    created 1to2 bridge for topic '/ctv_10/odom' with ROS 1 type 'nav_msgs/Odometry' and ROS 2 type 'nav_msgs/msg/Odometry'
    created 1to2 bridge for topic '/ctv_10/slam_toolbox/karto_graph_visualization' with ROS 1 type 'visualization_msgs/MarkerArray' and ROS 2 type 'visualization_msgs/msg/MarkerArray'
    created 1to2 bridge for topic '/ctv_10/slam_toolbox/karto_scan_visualization' with ROS 1 type 'sensor_msgs/LaserScan' and ROS 2 type 'sensor_msgs/msg/LaserScan'
    created 1to2 bridge for topic '/ctv_10/slam_toolbox/update' with ROS 1 type 'visualization_msgs/InteractiveMarkerUpdate' and ROS 2 type 'visualization_msgs/msg/InteractiveMarkerUpdate'
    created 1to2 bridge for topic '/ctv_10/slam_toolbox/update_full' with ROS 1 type 'visualization_msgs/InteractiveMarkerInit' and ROS 2 type 'visualization_msgs/msg/InteractiveMarkerInit'
    created 1to2 bridge for topic '/map_metadata' with ROS 1 type 'nav_msgs/MapMetaData' and ROS 2 type 'nav_msgs/msg/MapMetaData'
    created 1to2 bridge for topic '/rosout' with ROS 1 type 'rosgraph_msgs/Log' and ROS 2 type 'rcl_interfaces/msg/Log'
    created 1to2 bridge for topic '/rosout_agg' with ROS 1 type 'rosgraph_msgs/Log' and ROS 2 type 'rcl_interfaces/msg/Log'
    created 1to2 bridge for topic '/tf' with ROS 1 type 'tf2_msgs/TFMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage'
    created 1to2 bridge for topic '/tf_static' with ROS 1 type 'tf2_msgs/TFMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage'
    Created 2 to 1 bridge for service /ctv_10/slam_toolbox/dynamic_map
    [INFO] [ros_bridge]: Passing message from ROS 1 sensor_msgs/LaserScan to ROS 2 sensor_msgs/msg/LaserScan (showing msg only once per type)
    [INFO] [ros_bridge]: Passing message from ROS 1 rosgraph_msgs/Log to ROS 2 rcl_interfaces/msg/Log (showing msg only once per type)
    [INFO] [ros_bridge]: Passing message from ROS 1 nav_msgs/Odometry to ROS 2 nav_msgs/msg/Odometry (showing msg only once per type)
    [INFO] [ros_bridge]: Passing message from ROS 1 visualization_msgs/InteractiveMarkerInit to ROS 2 visualization_msgs/msg/InteractiveMarkerInit (showing msg only once per type)
    [INFO] [ros_bridge]: Passing message from ROS 1 tf2_msgs/TFMessage to ROS 2 tf2_msgs/msg/TFMessage (showing msg only once per type)
    [INFO] [ros_bridge]: Passing message from ROS 1 visualization_msgs/InteractiveMarkerUpdate to ROS 2 visualization_msgs/msg/InteractiveMarkerUpdate (showing msg only once per type)
    created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log'

All listed topics are available under ros2. Data on each topic is received. We just don't understand why the slam toolbox gives us that above mentioned INFO message. 

Here are the configuration files:

    slam_toolbox:
      ros__parameters:

        # Plugin params
        solver_plugin: solver_plugins::CeresSolver
        ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
        ceres_preconditioner: SCHUR_JACOBI
        ceres_trust_strategy: LEVENBERG_MARQUARDT
        ceres_dogleg_type: TRADITIONAL_DOGLEG
        ceres_loss_function: None

        # ROS Parameters
        odom_frame: ctv_10/odom
        map_frame: ctv_10/map
        base_frame: ctv_10/base_link
        scan_topic: /ctv_10/back_scan
        mode: mapping #localization

        # if you'd like to immediately start continuing a map at a given pose
        # or at the dock, but they are mutually exclusive, if pose is given
        # will use pose
        #map_file_name: test_steve
        #map_start_pose: [0.0, 0.0, 0.0]
        #map_start_at_dock: true

        debug_logging: true
        throttle_scans: 1
        transform_publish_period: 0.02 #if 0 never publishes odometry
        map_update_interval: 5.0
        resolution: 0.05
        max_laser_range: 20.0 #for rastering images
        minimum_time_interval: 0.5
        transform_timeout: 0.2
        tf_buffer_duration: 30.
        stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps
        enable_interactive_mode: true

        # General Parameters
        use_scan_matching: true
        use_scan_barycenter: true
        minimum_travel_distance: 0.5
        minimum_travel_heading: 0.5
        scan_buffer_size: 10
        scan_buffer_maximum_scan_distance: 10.0
        link_match_minimum_response_fine: 0.1  
        link_scan_maximum_distance: 1.5
        loop_search_maximum_distance: 3.0
        do_loop_closing: true 
        loop_match_minimum_chain_size: 10           
        loop_match_maximum_variance_coarse: 3.0  
        loop_match_minimum_response_coarse: 0.35    
        loop_match_minimum_response_fine: 0.45

        # Correlation Parameters - Correlation Parameters
        correlation_search_space_dimension: 0.5
        correlation_search_space_resolution: 0.01
        correlation_search_space_smear_deviation: 0.1 

        # Correlation Parameters - Loop Closure Parameters
        loop_search_space_dimension: 8.0
        loop_search_space_resolution: 0.05
        loop_search_space_smear_deviation: 0.03

        # Scan Matcher Parameters
        distance_variance_penalty: 0.5      
        angle_variance_penalty: 1.0    

        fine_search_angle_offset: 0.00349     
        coarse_search_angle_offset: 0.349   
        coarse_angle_resolution: 0.0349        
        minimum_angle_penalty: 0.9
        minimum_distance_penalty: 0.5
        use_response_expansion: true

Does anyone has an idea?

SteveMacenski, 2020-02-21:
Hi,

This is a ROS2/TF warning, not something from slam toolbox. I think your best bet is filing a question on ros answers. It may relate to the bridge, it may not, but I can say I don’t use it and I haven’t seen this before in this project. 

I have seen it before in Navigation2 from TF due to startup networking traffic but stops immediately after all the piping is setup.

closed 2020-02-21 by SteveMacenski

rakeshshrestha31, 2021-05-06:
Hi,

I was also getting a continuous stream of these messages, not just during startup. I found this PR: https://github.com/ros-planning/navigation2/pull/1574 from @SteveMacenski which solved this problem for navigation2 package which I thought also applied here. After making the corresponding changes the messages aren't there anymore. My PR: https://github.com/SteveMacenski/slam_toolbox/pull/391