SteveMacenski/slam_toolbox #171: Melodic deb fails to install because conflicts with open_karto

mikaelarguedas, 2020-02-25:
**Required Info:**

- Operating System:
  - Ubuntu Bionic
- Installation type:
  - deb
- ROS Version
  - melodic
- Version or commit hash:
  - N/A
#### Steps to reproduce issue

```
docker run -it --rm osrf/ros:melodic-desktop-full
apt -qq update && apt -qq upgrade -y
apt install -y ros-melodic-open-karto
apt install -y ros-melodic-slam-toolbox
```

#### Expected behavior

slam_toolbox installs successfully OR explicitly conflicts with open_karto

#### Actual behavior

Crash at unpacking time:
```
Unpacking ros-melodic-slam-toolbox (1.1.2-1bionic.20200121.145618) ...
dpkg: error processing archive /tmp/apt-dpkg-install-UvGDbF/36-ros-melodic-slam-toolbox_1.1.2-1bionic.20200121.145618_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/melodic/lib/libkarto.so', which is also in package ros-melodic-open-karto 1.2.0-0bionic.20191210.214159
```

#### Additional information

It appears that this has been copied from the originally released [open_karto](https://github.com/ros-perception/open_karto) without preserving history and is now released under another name but overwriting the one installed by open_karto

SteveMacenski, 2020-02-25:
It is copied into the slam toolbox project under the same license because of how much of it I changed, it no longer made sense as a fork or as a separate project since it was only usable by essentially this project. 

It is still at its core open_karto- and you’re correct I didn’t change the library name. 

What do you think is the best course of action?

mikaelarguedas, 2020-02-25:
Thanks for the context
That's a good question,. I guess one way to do it is to rename it so that the name makes it obvious it's provided by and usable only by `slam_toolbox`, not sure it's the recommended way.
I know some packages in ROS 2 install custom builds of library under their own package directory as a subfolder of `opt` folder, for example `rviz_ogre_vendor`
```
ls /opt/ros/eloquent/opt/rviz_ogre_vendor/lib
OGRE                               libOgrePaging.so.1.12.1
libOgreGLSupport.a                 libOgreProperty.so
libOgreHLMS.so                     libOgreProperty.so.1.12.1
libOgreHLMS.so.1.12.1              libOgreRTShaderSystem.so
libOgreMain.so                     libOgreRTShaderSystem.so.1.12.1
libOgreMain.so.1.12.1              libOgreTerrain.so
libOgreMeshLodGenerator.so         libOgreTerrain.so.1.12.1
libOgreMeshLodGenerator.so.1.12.1  libOgreVolume.so
libOgreOverlay.so                  libOgreVolume.so.1.12.1
libOgreOverlay.so.1.12.1           pkgconfig
libOgrePaging.so
```

@tfoote may have more precise recommendations for these scenarios

SteveMacenski, 2020-02-25:
I have no issue with renaming it to some arbitrary other name, at the end of the day, to a user and most developers, they're not going to be seeing or using the `.so` file. 

I do wonder what the best course of action is relative to ABI breaking changes since we're changing a library name. This is also, as you mention, not a great problem to mark as "wontfix" since it gives users a poor out of box experience. We could also move the install from the `/opt/ros/<distro>/lib` to `/opt/ros/<distro>/pkg/lib` with a change [here](https://github.com/SteveMacenski/slam_toolbox/blob/eloquent-devel/CMakeLists.txt#L179)

mikaelarguedas, 2020-02-25:
Yes, you could install it to `CATKIN_PACKAGE_LIBEXEC_DESTINATION`.

Not sure what version of the library people will get if they `find_package(slam_toolbox)` and there is one available in `/opt/ros/melodic/lib` as well as `/opt/ros/melodic/lib/slam_toolbox`

SteveMacenski, 2020-02-25:
Mhm. Maybe then not a good idea.

We could just rename it. `libkartoSlamToolbox.so` or something.

mikaelarguedas, 2020-02-25:
Yeah if the library renaming is acceptable for consumers of this package (which seems the case as this is used mostly for internal use), it seems like a sensible approach

SteveMacenski, 2020-02-25:
Want to submit a PR with that for melodic / kinetic?

SteveMacenski, 2020-03-02:
We resolved.

closed 2020-03-02 by SteveMacenski