SteveMacenski/slam_toolbox #175: crash on kinetic when trying to use serialization and deserialization

27Apoorva, 2020-02-28:
**Required Info:**

- Operating System:
Ubuntu 16.04
  - 
- Installation type:
from souurce
  - 
- ROS Version
ROS1
  - 
- Version or commit hash:
Commit hash 9b14cf8369a63264fd9935be4cca5cf14386bfc5
  - 
git branch: kinetic
#### Steps to reproduce issue

Run serialization and deserialization on kinetic branch
#### Expected behavior

#### Actual behavior
When I try to use serialization feature (online_sync.launch) on kinetic branch, slam_toolbox crashes. I had to comment line 6606 in Karto.h to make it save the .posegraph and .data file. 

Another bug is when I run derializationn, the service call keeps waiting for /scan topic. There was a bunch of code I found in function `DeserializePoseGraphCallback` which shouldn't be there. Following lines shouldn't be there:
while (true)
      {
        ROS_INFO("Waiting for scan to get metadata...");
        boost::shared_ptr<sensor_msgs::LaserScan const> scan = ros::topic::waitForMessage<sensor_msgs::LaserScan>(std::string("/sensors/lidar/scan"), ros::Duration(1.0));
        if (scan)
        {
          ROS_INFO("Got scan!");
          lasers_[laser_frame_] = laser_assistant_->toLaserMetadata(*scan);
          break;
        }
      }

#### Additional information

SteveMacenski, 2020-02-28:
First, I recommend Melodic branch. It will work on Kinetic as well but is far, far more organized and many bugs patched.

Second, you didn’t provide a crash backtrace. That’s necessary. Make sure you set the stack memory parameter high enough, see defaults.

27Apoorva, 2020-02-28:
I will try using melodic branch. I removed code after understanding and comparing with master branch. I will provide stack trace. I think stack memory parameter might be obsolete for kinetic branch. Thanks. I update this thread with stack trace later on.

27Apoorva, 2020-02-28:
Log when crash happens for serialization (tried on melodic as well)

Registering sensor: [Custom Described Lidar]
Save To File .posegraph 
Mapper <- Module
Mapper <- m_pSequentialScanMatcher
Mapper <- m_pGraph
MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
Mapper <- m_pMapperSensorManager
MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
Mapper <- m_Listeners
**Finished serializing Mapper**
Save To File
**Serializing Dataset**
Dataset <- m_SensorNameLookup
Dataset <- m_Data
Dataset <- m_Lasers
Dataset <- m_pDatasetInfo
[slam_toolbox-2] process has died [pid 14970, exit code -11, cmd /mfp_workspace/devel/lib/slam_toolbox/sync_slam_toolbox_node __name:=slam_toolbox __log:=/home/apoorvagupta/.ros/log/0f801e86-5a58-11ea-aa4a-001c421d330c/slam_toolbox-2.log].
log file: /home/apoorvagupta/.ros/log/0f801e86-5a58-11ea-aa4a-001c421d330c/slam_toolbox-2*.log

SteveMacenski, 2020-02-28:
Please provide a traceback & configuration file.

SteveMacenski, 2020-03-05:
Hi,

Any update?

27Apoorva, 2020-03-05:
I pulled in latest melodic branch and it stopped happening. Thanks Closing it now

closed 2020-03-05 by 27Apoorva