SteveMacenski/slam_toolbox #177: [ROS2] Complete porting of rviz plugin and interactive markers

SteveMacenski, 2020-03-05:
#### Feature description

In ROS1, I had an RViz plugin for interacting with slam toolbox for testing and novice users. That was not fully ported at the time because RViz wasn't feature complete yet and interactive markers weren't in Crystal.

This is no longer the case and it can be completed to be fully feature-complete with ROS1's version.

#### Implementation considerations

* Implement commented out code in the loop closure assistant, add interactive markers as a dependency, and look at the registration of the rviz plugin with `rviz_common`.

shoemakerlevy9, 2020-03-18:
@SteveMacenski I'm evaluating this for Rotu to work on. Can you describe a little more about what this plugin did.

SteveMacenski, 2020-03-18:
If you look at https://github.com/SteveMacenski/slam_toolbox#rviz-plugin, you'll see the rviz plugin window. Essentially, its just an interface to work with tools exposed. It calls the various services and gives interactive tool commands.

The interactive markers are slightly different, being used by the loop closure assistant for people to visually drag graph nodes around to cause manual loop closures. But also tied into the interactive tools.

nachoampu, 2020-06-09:
Hi, so I am trying to implement the slam_toolbox to a project I have in ROS2, and I have been unable to install or use the rviz plugin. I am kind of a noob with all of these. I am using rviz2, maybe that is the issue.

SteveMacenski, 2020-06-09:
If you read this ticket and the readme, its not yet ported to ROS2 for use.

AndreasAZiegler, 2020-07-28:
May I ask about the current state? I might be able to contribute a bit of time for this. However, I don't like to make redundant work.

SteveMacenski, 2020-07-28:
@timotheos was working on it (I believe that is his github handle) for a bit but I'm not sure his status on it. I don't see a fork on his account.

This should also be really straight forward since I just commented out the code, so its just uncommenting, porting some, and testing.

timotheos, 2020-07-29:
Hi @AndreasAZiegler @SteveMacenski.

I pulled it straight from the repository, and I'm still in the middle of porting it. I should get it to testing state by the end of the week or early next week 🙂.

AndreasAZiegler, 2020-07-29:
@timotheos Thanks for the info. Let me know when it is ready to test. I'm happy to give it a try.

timotheos, 2020-08-04:
Hi again, 

I have some issues trying to port the `rviz_common::Panel` _referenced below_ and I was going to test it some more. However, after rebase-ing my commit to current `foxy-devel` branch, some of my simulations are stopped working.  

If you want to take a look at the progress, I have it forked [here](https://github.com/timotheos/slam_toolbox/tree/foxy-rviz-plugin).

Will continue to pour more time into this throughout the week :)

[image]

SteveMacenski, 2020-08-04:
https://github.com/ros2/rviz/issues/469 see this ticket with the fix - things changed from ROS1 to 2 on this front.

timotheos, 2020-08-05:
Yup, I saw the issue, but still wasn't able to fix it. I'll recheck my code to see if something was overlooked. I also stumbled on [this](https://answers.ros.org/question/341214/custom-panel-plugin-not-being-found-when-running-rviz2/) the other day.

SteveMacenski, 2020-08-05:
Yeah that's what it links to. I think the issue is just in not using the `rviz_common` registration or something. In the ticket someone shows theirs works

WilliamWoo45, 2020-11-19:
Hi @timotheos !
I'm kind of having a similar issue with you, in which the `SlamToolboxPlugin` panel failed to show up in `Rviz`.
Can you please take a look at my detailed description in [pull #234](https://github.com/SteveMacenski/slam_toolbox/pull/234#issuecomment-729399409) ?

Thanks and all the best!
William

wienans, 2021-02-13:
Hey thanks for the fix for the first issue. But after Rviz detected the Plugin i run directly into the next Plugin Factory Error.

[image]

So i checked the directory and the libSlamToolboxPlugin.so is available. Do you have some ideas what is happening?

SteveMacenski, 2021-02-13:
I'm not 100% sure from that error / description. Pluginlib is picky at the best of times (fairly so, but alot of moving parts) so it could be one of a number of things that I can't narrow down much without experiencing it myself and playing around. I know that's not very helpful :frowning_face:

wienans, 2021-02-13:
Thanks for your response. But the most/all features of the plug-in are also available with services too? Actually I tried to save the posegraph and didn’t got a response from the service call. So I thought my cli input was wrong. Actually the tab auto complete of the service call looked really wired with all the backslashes in it. And that’s why I wanted to true the plug-in.
Maybe you have an example of the ros2 service call to save the posegraph. Or do you have examples and I didn’t found them 🤔 I only found the documentation which service are offered. 
thanks in advance

SteveMacenski, 2021-03-03:
Working on this now - about half way done. Should be in before the end of the week

SteveMacenski, 2021-03-04:
Have it all compiling and rviz plugin visualizing, PR  incoming shortly. Will test tomorrow

closed 2021-03-04 by SteveMacenski

HazemAbidi, 2021-03-08:
Hey @SteveMacenski, I'm using ros foxy and slam toolbox in my project.
I'm facing an error when I open the rviz config file provided in the config directory.  
please check the errors screenshot in these links [rviz_error1](https://ibb.co/hsz9Pm3), [rviz_error2](https://ibb.co/wBVbmbd), and [cmd_errors](https://ibb.co/c2r11Fg).
Any idea please.
I thank you in advance