SteveMacenski/slam_toolbox #188: Error slam_toolbox:: cannot locate rosdep definition for [ament_lin_auto]

alexx100-cmd, 2020-03-27:
**Required Info:**

- Operating System:
 Ubuntu 16.04
- Installation type:
 from source  
- ROS Version
- ROS1, MELODIC 

#### Steps to reproduce issue
RUN mkdir -p catkin_ws/src
RUN cd catkin_ws/src && git clone https://github.com/SteveMacenski/slam_toolbox.git
RUN source /opt/ros/melodic/setup.bash
&& cd catkin_ws
&& rosdep update
&& rosdep install -y -r --from-paths src --ignore-src --rosdistro=melodic -y

RUN apt install python-catkin-tools -y
RUN source /opt/ros/melodic/setup.bash \
&& cd catkin_ws/src
&& catkin_init_workspace
&& cd ..
&& catkin build

When I make this 

>  rosdep install -y -r --from-paths src --ignore-src --rosdistro=melodic -y

 had the error.

#### Expected behavior

Build the package slam_toolbox 

#### Actual behavior

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
slam_toolbox: Cannot locate rosdep definition for [ament_lint_auto]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

SteveMacenski, 2020-03-27:
The clone didn’t clone the melodic branch, the default branch is Eloquent, which requires ament. 

A PR to fix the dockerfile would be appreciated

SteveMacenski, 2020-03-29:
You just need to update the `melodic-devel`'s docker file to checkout the proper branch, and then submit a PR on `melodic-devel`. The `git clone` doesn't specify a branch and the default repo branch happens to be `eloquent-devel`, but you don't require anything from it. You just need to make sure the dockerfile checks out the right branch.

alexx100-cmd, 2020-03-29:
Ok, i do it. Thanks Steve.
Just one question more, perhaps it's very basic but how can i load the plugin in rivz??

SteveMacenski, 2020-03-29:
Through docker? You’ve have to Rviz from the container I suppose.

alexx100-cmd, 2020-03-29:
Yes i have, i follow the basic tutorials to load but i cant`open from panels of Rviz

SteveMacenski, 2020-03-29:
Cant help. Dont know your setup & out of the scope of this repo. I suggest filing a ticket on ros answers.

SteveMacenski, 2020-03-30:
https://github.com/SteveMacenski/slam_toolbox/pull/190 resolves

closed 2020-03-30 by SteveMacenski