SteveMacenski/slam_toolbox #192: Mapping Performance for Small Robots

lukicdarkoo, 2020-04-02:
I am working on a rather special case. It is a small robot (7cm in diameter), equipped with an array of infra sensors with a maximum distance of 5cm. The result is the LaserScan topic that has a resolution of 15 degrees and an FoV of 300 degrees. The sensors return values greater than 5cm if there is no object around or close to zero if the reading is invalid.

I suspected that the LaserScan resolution and range are too low for mapping. Therefore, I compared to `gmapping` and it gives significantly better results (more details in _Additional information_). Changing parameters didn't help, but maybe I was changing the wrong ones.

- Operating System:
  - Ubuntu 18.04 (also tried on Ubuntu 20.04)
- Installation type:
  - source
- ROS Version
  - ROS2 Eloquent
- Version or commit hash:
  - 349a1d269287183056297e3ac0e9547362382658

#### Expected behavior
Here is output from `gmapping` (ROS1 running with `ros1_bridge`) after a few seconds of the robot moving around:
[image]

#### Actual behavior
Here is output from `slam_toolbox`:
[image]

#### Additional information
In comparison with `gmapping` I noticed the following issues:
Issues are the following (key performance differences compared to `gmapping`):
- There are frequent jumps in position. It seems that `slam_toolbox` doesn't "trust" to odometry, but I couldn't find a parameter to change it.
- If a range is greater than the maximum range it is not considered. Our sensors have a very short range and very often we can "see" there is no object in the radius of 5cm. This could be valuable in map building. `gmapping` does it very well.
- Map updates are slow. It can be changed to send a new map more frequently but there are no new features in the map (it is not as fast as `gmapping`).

Do you have any suggestion on how to tackle those 3 issues?

SteveMacenski, 2020-04-02:
If you don't have enough measurement readings to track any external features, yes, you're going to have a problem with any type of graph-based slam. What you're seeing is the effect of particle filter slam that gmapping provides: that it can charge forward into the unknown without any particular information and clear out theoretically free space, but relative to what? No one knows.

This is from a fundamental difference in slam problem formulation for which your application is unforunately on the down-side of. The entire field has moved away from particle filter based mapping approaches because they cannot have their state saved and re-initiated, it is impossible to make large enough spaces to be useful consistently, and sensors with significantly more range have become extremely cheap (for the cost of a 360 sonar ring, you can today buy a cheap lidar with 5x the range). There are a bunch of other reasons, but those are the greatest-hits I bring up. 

Overall, if your goal is to support a platform that will continue to be relevant, you should not choose that sensor configuration. Less and less techniques are going to be available for you to work with. Gmapping isn't even maintained anymore nor are there plans to port to ROS2.

lukicdarkoo, 2020-04-04:
> for the cost of a 360 sonar ring, you can today buy a cheap lidar with 5x the range

Besides the fact I cannot change design of the robot, the purpose of the robot is to be very small so it can fit on the palm of a hand. Putting LiDAR that is bigger than the robot destroys its purpose.

I have to find an alternative solution :/ Maybe a small note about the hardware requirements (eg. LiDAR resolution, range etc) would be useful for the other users of `slam_toolbox`.

SteveMacenski, 2020-04-04:
Its not a function of absolute resolution or range. Its a function of relative values. If you have a 10cm lidar in a warehouse, you’ll have trouble. If you have a 100m lidar in an apartment, you won’t (on the extremes). It seems the relative range for your needs is far too low to track functional features. There are users using this with the lowest end of available lidars so I’m confident users can get a good result for the supported sensor suite.

closed 2020-04-04 by SteveMacenski