SteveMacenski/slam_toolbox #193: merge map, rotation

sunrentao, 2020-04-02:
- Operating System:
  - Ubuntu 18.04
- Installation type:
 - from source 
- ROS Version
  - ROS1
- Version or commit hash:
  -9fd560ab1f56187e2a4397c8a9440da9b2dfa8c1

#### Steps to reproduce issue
Launch merge_maps_kinematic.launch 
Add 2 submaps and manipulate one of them to merge 

#### Expected behavior

generate a map as indicated

#### Actual behavior

generate a wrong map, problem happens about rotation

#### Additional information

also another problem is that for example if I call rosservice merge_map twice, the map will make another transformation respect to last position

SteveMacenski, 2020-04-02:
> generate a wrong map, problem happens about rotation

Is not a helpful description. Please add detail and examples.

sunrentao, 2020-04-03:
[image]

hello, for example, this is the first step to add submap, then , if I rotate this map, then the generated map is as following, 

[image]

and if I only move the map without rotation, first time generated map will be ok, but if I generate map twice without any moving , it will be like this 

[image]

I read the code, that every time the map tool give a new data, it will update "submap_marker_transform_" with " previous_submap_correction.inverse() " I think this is the problem, but I am not sure.

SteveMacenski, 2020-04-03:
I did not consider trying to hit the merge button multiple times as if you merge your N maps, you should be done. 

However a PR would be appreciated that resolves your issue.

closed 2020-04-06 by sunrentao