SteveMacenski/slam_toolbox #199: Mapping freeze with Gazebo

lukaszmitka, 2020-04-27:
**Required Info:**

- Operating System:
 Ubuntu 18.04
- Installation type:
 Binaries
- ROS Version
  ROS2 Dashing
- Version or commit hash:
  commit hash: 094e9aa3bd914b56a78c3e7d3de1433ea9d6d85a

#### Steps to reproduce issue
Start Gazebo simulation with custom robot model.
Model and world used are from [rosbot_description package](https://github.com/husarion/rosbot_description/tree/ros2), simulation started with [rosbot_sim.launch.py](https://github.com/husarion/rosbot_description/blob/ros2/launch/rosbot_sim.launch.py)

Start slam (`async_slam_toolbox_node` or `sync_slam_toolbox_node`) with parameters:
```
slam_toolbox:
  ros__parameters:

    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: None

    # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_link
    scan_topic: /scan
    mode: mapping #localization

    debug_logging: false
    throttle_scans: 1
    transform_publish_period: 0.1 #if 0 never publishes odometry
    map_update_interval: 1.0
    resolution: 0.025
    max_laser_range: 25.0 #for rastering images
    minimum_time_interval: 0.5
    transform_timeout: 5.0
    tf_buffer_duration: 30.
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps
    enable_interactive_mode: false

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.05
    minimum_travel_heading: 0.05
    scan_buffer_size: 50
    scan_buffer_maximum_scan_distance: 0.25
    link_match_minimum_response_fine: 0.1  
    link_scan_maximum_distance: 0.25
    loop_search_maximum_distance: 3.0
    do_loop_closing: true
    loop_match_minimum_chain_size: 10           
    loop_match_maximum_variance_coarse: 3.0  
    loop_match_minimum_response_coarse: 0.35    
    loop_match_minimum_response_fine: 0.45

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.5
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.1 

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03

    # Scan Matcher Parameters
    distance_variance_penalty: 0.5      
    angle_variance_penalty: 1.0    

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349   
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true

    # Simulation
    use_sim_time: true
```
Drive robot around to create map.

#### Expected behavior

Map is being created.

#### Actual behavior

After few minutes, `slam_toolbox` freeze. Pose is not published but map publihser is still active.

#### Additional information

I have tried to run slam with valgrind, the issue did not occur after more than hour.

SteveMacenski, 2020-04-27:
Froze doesn’t say much, I’ll need more to go on. The first thing that hits me about your config file is you want updates every 5cm or super small changes in angle. That’s not a good recipe for scan matching quality but wouldn’t result in “freezing”.

lukaszmitka, 2020-05-02:
I changed the parameters to update less often, the issue does not occur now.

closed 2020-05-02 by SteveMacenski