SteveMacenski/slam_toolbox #202: Localization mode

RodBelaFarin, 2020-05-18:
#### Feature description

Would be nice to have an example launch file for the localization mode showing how to enable localization and how to parameterize the map location / map file.

#### Implementation considerations

In ROS2 Eloquent even the sample config file _mapper_params_localization.yaml_ enables the mapping mode.

SteveMacenski, 2020-05-18:
Dis you read the description or watch the ROSCon talk on this subject?

RodBelaFarin, 2020-05-18:
Well, yes I did. I did even watch the ROSCon talk live in Macau.

Again: Even the config file, which is called _localization_ runs the mapping mode by default. That is confusing to me.

> The localization mode will automatically load your pose graph, take the first scan and match it against the local area to further refine your estimated position, and start localizing.

And from reading the description, it says my pose graph is automatically loaded, BUT how does slam toolbox know where to find my pose graph? I guess that I need to specify this.

SteveMacenski, 2020-05-18:
From the readme:

> ...  mode: `localization` in the configuration file to allow for the Ceres plugin to set itself correctly to be able to quickly add and remove nodes and constraints from the pose graph, but isn't strictly required, but a performance optimization. The localization mode will automatically load your pose graph, take the first scan and match it against the local area to further refine your estimated position, and start localizing.

and please review the parameters section of the readme: https://github.com/SteveMacenski/slam_toolbox#toolbox-params on the configs for filenames/poses.

RodBelaFarin, 2020-05-18:
my .yaml file looks like this:
```
   # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_footprint
    scan_topic: /scan
    mode: localization #mapping

    # if you'd like to immediately start continuing a map at a given pose
    # or at the dock, but they are mutually exclusive, if pose is given
    # will use pose
    map_file_name: map_1589815244
    #map_start_pose: [0.0, 0.0, 0.0]
    #map_start_at_dock: true
```
But I don't receive a map. So what's wrong?

SteveMacenski, 2020-05-18:
Please ask on ROS Answers

closed 2020-05-18 by SteveMacenski

WilliamWoo45, 2020-11-20:
Hi @RodBelaFarin !
I'm kind of having a similar issue with you. 
Firstly, I managed to save the map by `rosservice call /slam_toolbox/save_map "name:
  data: 'test_map'"`.

There is also correct info showing in the command line as follows:
`[ INFO] [1605856325.722527948, 1562036683.907657948]: SlamToolbox: Saving map as test_map.
[ INFO] [1605856325.985398475]: Waiting for the map
[ INFO] [1605856326.181920720]: Received a 1789 X 737 map @ 0.050 m/pix
[ INFO] [1605856326.181994265]: Writing map occupancy data to test_map.pgm
[ INFO] [1605856326.222453671]: Writing map occupancy data to test_map.yaml
[ INFO] [1605856326.222636871]: Done`

But when I launch the localization file, the _localization_ actually runs the _mapping_ mode by default and there is no map showing in `Rviz`. Besides, there is no error message showing in the command line.

I'm wondering did you managed to solve this problem?

Thanks and all the best 
William

SteveMacenski, 2020-11-20:
@WilliamWoo45 please do not comment on old tickets with unrelated questions. Ask on ROS Answers.