SteveMacenski/slam_toolbox #208: libceres-dev ubuntu 20.04 no install candidate RaspberryPi 4 8GB Noetic

bgilreath, 2020-06-10:
**Required Info:**

- Operating System: Ubuntu 20.04
  - 
- Installation type: from source
  - 
- ROS Version ROS1 Noetic
  - 
- Version or commit hash: noetic-devel 
  - 

#### Steps to reproduce issue

1. cd catkin_ws
2. rosdep install --from-paths src --ignore-src -r -y
2a. This is first warning if you take this path
3. catkin_make
3a. Second warning

#### Expected behavior

#### Actual behavior

#### Additional information
Just pointing out this issue.

SteveMacenski, 2020-06-10:
>  E: Package 'libceres-dev' has no installation candidate

Your issue is that ceres hasn't been successfully released in 20.04 yet on ARM64 (and I suppose rosdep). This is actually a noted issue https://github.com/ceres-solver/ceres-solver/issues/599 https://bugs.launchpad.net/ubuntu/+source/ceres-solver/+bug/1882626 to me recently and its being slowly dealt with.

Since this isn't actually an issue with slam toolbox, but rather an unreleased dependency (also why you dont see a Noetic or Foxy release for this yet), I'm going to close this ticket. But I do really appreciate you bringing this to my attention, but for the moment, there's nothing I can do about it. We're both in a waiting pattern for Ceres support for 20.04, after which I will test and release slam toolbox as well. For now, I recommend reverting to Melodic. There's a number of other packages not available in Noetic yet that I think you should revert if this blocks you (because other things will certainly block you too).

SteveMacenski, 2020-06-10:
You know what, I lied, I'm going to keep this open so I have a note about this to remind me to deal with this once its been released.

closed 2020-07-13 by SteveMacenski