SteveMacenski/slam_toolbox #211: /slam_toolbox/karto_scan_visualization laser scan displayed in incorrect frame

MikeWrockAIS, 2020-06-17:
First off let me start by saying this is a very promising looking SLAM package, thank you for your work. I am just starting out with this package so I'm not sure if I'm doing something incorrectly or not, but I have the feeling the laser scan of the node that is displayed when using interactive markers to manually adjust the pose graph is being displayed in the sensor's coordinate frame.

In my particular gazebo simulation, I am using 2 lidars and combining them using ira_laser_tools. The two lidars are mounted at 45deg angles to the base_link, and I noticed the laserscans displayed when using the interactive markers were also incorrect by 45deg. The static map appears fine, which leads me to believe it is a visualisation error rather than a SLAM error.

If you think this is likely the cause I will look in to fixing it and submitting a PR.

SteveMacenski, 2020-06-17:
That seems like it could be the issue, I don't think that tool takes into account the `yaw` information from TF perhaps when reprojecting into rviz. I welcome a PR to resolve

SteveMacenski, 2020-08-03:
@MikeWrockAIS checking in on this - have you made progress / resolved it using the yaw?

MikeWrockAIS, 2020-08-04:
Sorry I never got back to you, the issue was I had use_scan_barycenter set
to true. When disabled the tf to the scan frame was used. The package works
great, thanks a lot for your contribution!

SteveMacenski, 2020-08-04:
Thanks for the followup

closed 2020-08-04 by SteveMacenski