SteveMacenski/slam_toolbox #215: base class type rviz::Panel does not exist.

ZOUJIASHUAI, 2020-07-01:
[ERROR] [1593583801.221850214, 1593415606.581352811]: PluginlibFactory: The plugin for class 'slam_toolbox/SlamToolboxPlugin' failed to load.  Error: According to the loaded plugin descriptions the class slam_toolbox/SlamToolboxPlugin with base class type rviz::Panel does not exist. Declared types are  rviz_plugin_tutorials/Teleop slam_toolbox::SlamToolboxPlugin

can you tell me how to fix this problem?

ZOUJIASHUAI, 2020-07-01:
WARNING: Target "ceres_solver_plugin" requests linking to directory "/usr/local/include".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "ceres_solver_plugin" requests linking to directory "/usr/include/eigen3".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "ceres_solver_plugin" requests linking to directory "/usr/include".  Targets may link only to libraries.  CMake is dropping the item.

SteveMacenski, 2020-07-01:
```
Declared types are rviz_plugin_tutorials/Teleop slam_toolbox::SlamToolboxPlugin
```

You need a `::` not `/`. Please redirect questions rather than bugs to ROS Answers

closed 2020-07-01 by SteveMacenski