SteveMacenski/slam_toolbox #217: Different base_frame defaults for online and offline mapping

NikolasE, 2020-07-07:
Frames in configs should be consistent (and there is a lot of shared params in the files, so it's hard to see which variables change between the different modes)

#### Actual behavior
no suprises when switching between online and offline mode

#### Additional information
mapper_params_offline.yaml:
base_frame: base_link

mapper_params_online_sync.yaml:
base_frame: base_footprint

SteveMacenski, 2020-07-07:
Ah, I think it should be `_footprint` in all cases. Mind updating those?

NikolasE, 2020-07-13:
I created a sophisticated PR :) [224](https://github.com/SteveMacenski/slam_toolbox/pull/224)

But I think the shared parameters should be in a base-configuration and only the changed parameter in their own files.

closed 2020-07-13 by SteveMacenski