SteveMacenski/slam_toolbox #218: ROS2 Eloquent  + Ouster Help

NathanDodson, 2020-07-08:
Hey there. i'm using the ros2_ouster_drivers package to interface with my os1 lidar. Trying to use the /scan topic published by this in slam_toolbox. I replace the base_frame with laser_sensor_frame for testing, but I get this in the command line. Any help with this would be much appreciated!

[slam_toolbox-1] [INFO] [slam_toolbox]: Message Filter dropping message: frame 'laser_data_frame' at time 646.885 for reason 'Unknown'

    odom_frame: odom
    map_frame: map
    base_frame: laser_sensor_frame
    scan_topic: /scan
    mode: mapping #localization

SteveMacenski, 2020-07-08:
Do you have odometry data coming in? if not, then there's no valid TF tree.

NathanDodson, 2020-07-08:
No I do not have odometry data. I thought this package only needed data from /scan to slam. I guess I misunderstood. Thank you for the help!

closed 2020-07-08 by NathanDodson