SteveMacenski/slam_toolbox #221: Support mapping without odometry

tim-fan, 2020-07-09:
Hello, I'd like to use slam_toolbox without an odometry source. From what I can tell odom is currently required as an odom->base tf, so I'm submitting this issue as a feature request.

#### Feature description
I'd like to run slam_toolbox using only laser scans - no odometry input. 
Ideally slam_toolbox would provide the odom->base tf itself through scan matching, and perhaps publish to /odom as well.
This may be valuable in some use-cases as it would enable mapping/localising without needing to set up an odometry source and calibrate the base->laser link, 
Looks like this would have helped #218 

#### Implementation considerations
I've managed to get this use-case running using [laser_scan_matcher](http://wiki.ros.org/laser_scan_matcher), but this seems like duplicated effort as I understand slam_toolbox is already running a scan matcher. Also laser_scan_matcher doesn't appear to be supported beyond kinetic.

SteveMacenski, 2020-07-09:
That's not in the scope of this project, you must provide yourself some form of odometry, whether from wheels, IMUs, laser scan matchers, RGB, T265, etc. Its not in the scope of this project or a SLAM library to do so. Your system should be composed of nodes to accomplish specific nodes and its not in the design of ROS-based systems to tie a SLAM library to an odometry method. The types of methods used to compute odometry vs matches for SLAM do not have the same requirements and should not be used as analogs.

closed 2020-07-09 by SteveMacenski

tim-fan, 2020-07-09:
Ok, I'll keep using an external scan matcher. Thanks for the comments :thumbsup:

NikolasE, 2020-07-14:
Would you merge a PR with a launch file that uses the laser_scan_matcher? I also ran into this issue and solved it like @tim-fan

daviddavo, 2022-12-09:
Is it possible to use mapping with "perfect odometry" (i.e. using just the `pose` of the robot?)