SteveMacenski/slam_toolbox #229: Error combining robot_localization and slam_toolbox

NathanDodson, 2020-07-16:
**Required Info:**

- Operating System: Ubuntu 18.04
- Installation type: source
- ROS Version: ROS2 eloquent
- Version or commit hash: Output of git rev-parse HEAD: fc3966c381527146d8f51c42746e0555bd743fe9

#### Steps to reproduce issue
Combine imu from os1 lidar with odom_raw from openrover_demo's using robot localization then run online_sync_launch.py

#### Expected behavior
Create a map of the area

#### Actual behavior
Dropping these errors

[slam_toolbox-1] [INFO] [slam_toolbox]: Message Filter dropping message: frame 'laser_data_frame' at time 1152.868 for reason 'OutTheBack'
[slam_toolbox-1] [WARN] [tf2_ros_message_filter]: MessageFilter [target=odom ]: Message dropped because: Failed to transform from laser_data_frame to odom

#### Additional information
When running ros2 run tf2_ros tf2_echo odom laser_data_frame a transform is echoed.

SteveMacenski, 2020-07-16:
I'll be honest that error (`OutTheBack`) I have literally never seen before. The description is `The timestamp on the message is more than the cache length earlier than the newest data in the transform cache. ` which I think means that the timestamp it received is so old its basically useless. 

Given you're doing a pretty unique thing and I'm not seeing this, I think its due to your setup. I can help within reason to resolve but largely its going to be on you to figure out where in your pipeline things are failing before getting to SLAM Toolbox. Are the timestamps from the OS1 & rover good / accurate to the ROS time? I'd check there first. If its dropping laser -> odom transforms, that would be the laser -> base and base -> odom, which means that either your TF tree for your base isn't published (or fast enough) or your odometry transforms from the inputs to Robot Localization is messed up.  If your robot was working before, then its probably then the odometry transforms from RL.

NathanDodson, 2020-07-16:
Ah yes I think I may know the issue then. I have heard that the os1 lidar defaults to the first time point in computing, beginning of 1970, but thankfully I know someone who has synced the timing with the system time. This would explain why the data seems to be too old to be useful. The current tf tree is odom -> base_footprint -> laser_sensor_frame -> laser_data. Robot_localization published the odom to base_footprint transform. I run a static transform from base_footprint to laser_sensor_frame. Then, ros2_ouster creates the laser_sensor_frame to laser_data_frame transform. Should that tree work for slam_toolbox?

SteveMacenski, 2020-07-16:
I believe so, yes, but that's overall part of your system and not this package

closed 2020-07-27 by SteveMacenski