SteveMacenski/slam_toolbox #242: Map topic is quiet

Jbwasse2, 2020-07-29:
**Required Info:**

- Operating System:
Ubuntu 18.04
- Installation type:
Binaries
- ROS Version
ROS1 Melodic
- Version or commit hash:
97409f5dffb89a28215053dc8a62757137214c8b
#### Steps to reproduce issue
1) Download .bag [here](https://drive.google.com/file/d/1eub_ycKyZMJ3I2wmI_1vZe2ungvYR_mA/view?usp=sharing), this comes from the [JRDB test dataset](https://jrdb.stanford.edu/dataset/about)
2) roscore
3) No tf_static is published in the rosbag, so I followed [this tutorial](http://wiki.ros.org/tf2/Tutorials/Writing%20a%20tf2%20static%20broadcaster%20(Python)) to make a publisher so that way I can try to get a map built.
4) roslaunch slam_toolbox online_sync.launch
5) rosrun rviz rviz 
6) Open slam_toolbox_default.rviz in rviz 
7) rosbag play quarry-road-2019-02-28_0.bag --clock /segway/filtered_front_scan:=/scan /map:=/na

#### Expected behavior
/map, /scan, /tf, /tf_static echo data when performing "rostopic echo".
A map is created in rviz

#### Actual behavior
/tf and /scan echo without issue
/tf_static echos data if I follow step (3)
/map never echos data
rviz does not create a visualization of the map (I am assuming this is because /map never echos data)

I get a lot of debug messages such as below:
[image]

#### Additional information
I am using default configs.
I have verified the topics exist, and that nodes are subscribing/publishing to the topic. Should the slam_toolbox both subscribe and publish to /map?

SteveMacenski, 2020-07-29:
Show a picture of your complete TF tree.

Given you're hacking this dataset, I'm inclined to think its your use of this that's a problem not the package itself. I run my systems on the melodic branch here and I have seen no problems recently.

SteveMacenski, 2020-08-03:
Closing from inactivity.

closed 2020-08-03 by SteveMacenski