SteveMacenski/slam_toolbox #243: Double walls / Artifacts at walls

AndreasAZiegler, 2020-07-30:
**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- ROS Version
  - ROS2, Foxy
- Version or commit hash:
  - 2.3.0

#### Steps to reproduce issue
Run
``` ros2 launch slam_toolbox online_sync_launch.py```

with

```
slam_toolbox:
  ros__parameters:

    # Plugin params
    solver_plugin: solver_plugins::CeresSolver
    ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
    ceres_preconditioner: SCHUR_JACOBI
    ceres_trust_strategy: LEVENBERG_MARQUARDT
    ceres_dogleg_type: TRADITIONAL_DOGLEG
    ceres_loss_function: HuberLoss

    # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_link
    scan_topic: /scan
    mode: mapping #localization

    # if you'd like to immediately start continuing a map at a given pose
    # or at the dock, but they are mutually exclusive, if pose is given
    # will use pose
    #map_file_name: test_steve
    map_start_pose: [0.0, 0.0, 0.0]
    map_start_at_dock: true

    debug_logging: true
    throttle_scans: 1
    transform_publish_period: 0.02 #if 0 never publishes odometry
    map_update_interval: 5.0
    resolution: 0.025
    max_laser_range: 20.0 #for rastering images
    minimum_time_interval: 0.25
    transform_timeout: 0.2
    tf_buffer_duration: 30.
    stack_size_to_use: 40000000 #// program needs a larger stack size to serialize large maps
    enable_interactive_mode: true

    # General Parameters
    use_scan_matching: true
    use_scan_barycenter: true
    minimum_travel_distance: 0.2
    minimum_travel_heading: 0.35
    scan_buffer_size: 70
    scan_buffer_maximum_scan_distance: 20.0
    link_match_minimum_response_fine: 0.6
    link_scan_maximum_distance: 5.0
    loop_search_maximum_distance: 4.0
    do_loop_closing: true 
    loop_match_minimum_chain_size: 10
    loop_match_maximum_variance_coarse: 0.16
    loop_match_minimum_response_coarse: 0.7
    loop_match_minimum_response_fine: 0.7

    # Correlation Parameters - Correlation Parameters
    correlation_search_space_dimension: 0.3
    correlation_search_space_resolution: 0.01
    correlation_search_space_smear_deviation: 0.03

    # Correlation Parameters - Loop Closure Parameters
    loop_search_space_dimension: 8.0
    loop_search_space_resolution: 0.05
    loop_search_space_smear_deviation: 0.03

    # Scan Matcher Parameters
    distance_variance_penalty: 0.3
    angle_variance_penalty: 0.35

    fine_search_angle_offset: 0.00349     
    coarse_search_angle_offset: 0.349
    coarse_angle_resolution: 0.0349        
    minimum_angle_penalty: 0.9
    minimum_distance_penalty: 0.5
    use_response_expansion: true
```

#### Expected behavior
[image]

#### Actual behavior
[image]

#### Additional information

I want to move from an old OpenKarto wrapper to the slam_toolbox. I used the Karto parameters from our old setup and expected a similar output. It's probably only a parameter tuning and I assume that the odometry gets not enough penalized (compared to the scan match). Which parameters could be most promising to achieve the result of the previous system?

SteveMacenski, 2020-07-30:
Something small like that is just parameter tuning. Sorry, I don’t have the cycles to help with individual parameter optimization. The defaults I provide are “good” starting points but the rest is on users.

closed 2020-07-30 by SteveMacenski

AndreasAZiegler, 2020-07-30:
@SteveMacenski Sorry, I did not intend to bother you. I just thought that due to your experience you might know from the top of your head how to give a bigger penalty to the odometry errors and/or a smaller penalty to scan matching errors.

SteveMacenski, 2020-07-30:
Your update distance looks a little short to me. Usually matcher approaches want 70% new and 30% last scan to match well. That often its closer to the opposite. 

Otherwise, i’d look at the correlation & penalty params, but I think you knew that from your comment.