SteveMacenski/slam_toolbox #244: [ROS2] Convert nodes to Lifecycle nodes

Michael-Equi, 2020-07-31:
#### Feature description
I feel like there is a lot to gain in this repository by using something like a life-cycle node over the standard node interface. I haven't dug too deep into this repository yet but integrating it into my own project makes me see how it could be very useful. If this is something others are interested in and there are no objection I may be able to contribute this feature in a month or so.

SteveMacenski, 2020-07-31:
No objections here, though all this repos launch files need to have lifecycle autotransition then. We can’t tie out of the box behavior of this to Nav2’s lifecycle manager since this will be used often outside that context.

SteveMacenski, 2020-07-31:
Keep in mind SLAM has alot of dynamic memory allocations in it either way - so lifecycle is nice for integrating and consistency, but you wont be able to isolate the memory allocations to configuration and construction.

SteveMacenski, 2021-03-02:
@Michael-Equi any plans still to contribute back a lifecycled-version of SLAM Toolbox?

Michael-Equi, 2021-03-02:
Likely not any time soon, the progress I made on it can be found [here](https://github.com/BytesRobotics/slam_toolbox/tree/lifecycle_node) but I hit some problems and was not able to get past them fast enough so my team settled for a less elegant solution for the time being and I havent had the time to go back and finish it.

closed 2024-01-04 by SteveMacenski