SteveMacenski/slam_toolbox #245: Interactive mode for deserialized maps

Tobias-Fischer, 2020-07-31:
#### Feature description

I serialized a map after running `online_sync.launch`. I would like to manually close a loop in this map after deserializing it. As far as I understand, this is currently not possible. It would be great if this could be done somehow.

#### Implementation considerations

Tobias-Fischer, 2020-07-31:
The error which is thrown currently is:
`  what():  vector::_M_range_check: __n (which is 533) >= this->size() (which is 0)`

Or alternatively in localization mode that `[ERROR] [1596176922.932154679]: Interactive mode is invalid outside processing mode.`

SteveMacenski, 2020-07-31:
PRs appreciated

Tobias-Fischer, 2020-07-31:
Any hints on where to start with a PR?

SteveMacenski, 2021-02-23:
#329 covers the same scope

closed 2021-02-23 by SteveMacenski