SteveMacenski/slam_toolbox #277: Parameter to adapt initial map orientation

macsmitty, 2020-08-07:
Hi!
I'm looking for a way to control the initial orientation of the map when initially creating it. The desire for this function comes from the circumstance that I currently want to use this framework and cartographer to create maps (with the same hardware and in the same environment). Using the same rosbag I currently obtain two maps that are offset by about 45°. I would like to align these two maps. I already played around with the map_start_position parameter without any luck. Please note that it does not help me to adjust the transformation of the map within the world frame as all objects that are defined w.r.t. the map would then change as well. Also changing the tf rotation of the laser scanner is not an option since this would of course also lead to inconsistent maps. 
Any help or pointer in the right direction would be highly appreciated.
Thanks!
Marius

SteveMacenski, 2020-08-07:
The orientation is based on the orientation you start in relative to your environment. That won't always be true with cartographer because of how it deals with submaps. As you move around and you have some 10 (or whatever number) of buffered scans into a submap that is being then matched, the orientation of that initial submap isn't really in your control. The orientation of the map in this library should be determinstic on the orientation you start in.

But no, there's no way and would be no way to control the initial orientation of the map because how would the library _know_ what's the global 0-degree direction? It hasn't started mapping the space yet.

macsmitty, 2020-08-10:
Thank you @SteveMacenski. This cleared things up for me as I wasn't aware that the orientation of the initial map in cartographer is not deterministic. I wouldn't expect the library to know or determine the global 0-degree direction. Rather was I looking for a way to set a rotational offset to align the two maps.
However, considering the non-deterministic behavior of cartographer, manual manipulation of the posegraph is a feasible solution for us.

closed 2020-08-10 by macsmitty