SteveMacenski/slam_toolbox #293: Turtlebot Incorrect Number of Measurement Readings

danwblackburn, 2020-09-16:
**Required Info:**

- Operating System:
  - Ubuntu 20.04  JetPack 4.4
- Installation type:
  - Source
- ROS Version
  - ROS2 Foxy
- Version or commit hash:
  - d9a8a8689251b96386e14812cf5a9b410b55145f
- Laser unit:
  - Turtlebot3 Simulator
  
#### Steps to reproduce issue
Assuming that Gazebo9 is already installed then install Turtlebot3 repos
```
mkdir -p ~/turtlebot3_ws/src
cd ~/turtlebot3_ws
wget https://raw.githubusercontent.com/ROBOTIS-GIT/turtlebot3/ros2/turtlebot3.repos
vcs import src < turtlebot3.repos
```
Clone d9a8a8689251b96386e14812cf5a9b410b55145f in to src
```
git clone https://github.com/SteveMacenski/slam_toolbox.git src/slam_toolbox
```
Build/Source
```
colcon build --symlink-install
source ~/turtlebot3_ws/install/setup.bash
```
Set Turtlebot 3 Variables
```
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/models
export TURTLEBOT3_MODEL=waffle_pi
```
Launch
```
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py 
ros2 launch slam_toolbox online_sync_launch.py 
```
#### Expected behavior
Should launch with no issue and start generating a map.

#### Actual behavior
Errors on runtime with this message
```
LaserRangeScan contains 360 range readings, expected 359
```

#### Additional information
This is a similar issue to #278 except its happening with the turtlebot simulator so it seems to be affecting more than just the RPLIDAR driver. This was also referenced by someone else on the comments of https://github.com/SteveMacenski/slam_toolbox/commit/6dcab997bb531d695fca401ee3f17b2298ef1f46#commitcomment-41268207. But as far as I can tell he never opened an issue. I figured it was worth doing since turtlebot3 should mostly "run out of the box" with slam toolbox, but I'm not sure if it's an issue with turtlebot or with this repo.

Here is a sample LaserScan message from turtlebot:
```
header:
  stamp:
    sec: 35
    nanosec: 401000000
  frame_id: base_scan
angle_min: 0.0
angle_max: 6.28000020980835
angle_increment: 0.01749303564429283
time_increment: 0.0
scan_time: 0.0
range_min: 0.11999999731779099
range_max: 3.5
```
And here's the model.sdf from gazebo for the lidar model
```
<sensor name="hls_lfcd_lds" type="ray">
      <always_on>true</always_on>
      <visualize>true</visualize>
      <pose>-0.064 0 0.121 0 0 0</pose>
      <update_rate>5</update_rate>
      <ray>
        <scan>
          <horizontal>
            <samples>360</samples>
            <resolution>1.000000</resolution>
            <min_angle>0.000000</min_angle>
            <max_angle>6.280000</max_angle>
          </horizontal>
        </scan>
        <range>
          <min>0.120000</min>
          <max>3.5</max>
          <resolution>0.015000</resolution>
        </range>
        <noise>
          <type>gaussian</type>
          <mean>0.0</mean>
          <stddev>0.01</stddev>
        </noise>
    </ray>
```

I also had the same issue when using https://github.com/ros-perception/pointcloud_to_laserscan to convert from a PointCloud2 to a LaserScan. It was fixed when I changed the Laser Scan projection settings from
```
Angle minimum : -3.14
Angle maximum: 3.14
Angle increment: .0058
```
to 
```
Angle minimum : -3.14159
Angle maximum: 3.14159
Angle Increment: 0.005796296
```
But I figured it was just a rounding error. It may actually be an issue with simulated Laser Scans and how slam toolbox is deciding whether or not they are 360 laser scans. Or it could be an issue with the simulated Laser Scans not following the correct standard, but this is beyond my knowledge. 

Also this issue doesn't occur when using the binary from apt-get so this isn't an issue in 2.3.0 but it is in the foxy-devel branch.

SteveMacenski, 2020-09-16:
Please ask on ROS Answers. I don't support turtlebot3 as an individual, if they've not done something correctly in their URDF, that's an issue for them to resolve.

danwblackburn, 2020-09-16:
I'm sorry but I don't think you read my issue correctly. I know that you don't support turtle bot individually. I believe that slam toolbox has an issue with determining if simulated/projected LaserScans are 360 laser scans for the karto sdk.

Especially because this is a regression of slam toolbox because it works on tag 2.3.0 but not on the current foxy-devel branch.

Specifically line 108 of src/laser_utils.cpp is what is causing this:
```
bool is_360_lidar = false;
  if (std::fabs(scan_.angle_max - scan_.angle_min - 2.0 * M_PI) < (scan_.angle_increment - (std::numeric_limits<float>::epsilon() * 2.0f*M_PI))) {
    is_360_lidar = true;
  }
```
changed from 
```
 bool is_360_lidar = false;
 if (std::fabs(scan_.angle_max - scan_.angle_min - 2.0 * M_PI) < 1e-1) {
     is_360_lidar = true;
}
```

I'm opening this issue in response to this comment where you said you'd be happy to help someone debug this. https://github.com/SteveMacenski/slam_toolbox/commit/6dcab997bb531d695fca401ee3f17b2298ef1f46#commitcomment-41288325. Would you be willing to reopen this issue so that we could find a more robust way to create the karto::LaserRangeFinder that doesn't cause the off by one range issue that people seem to be having?

SteveMacenski, 2020-09-16:
You are correct, apologies. I tend to skim TB3 related issues. 

@justinIRBT thoughts? Sounds like something introduced via your PR https://github.com/SteveMacenski/slam_toolbox/pull/288.

Doing some quick math...

`6.28000020980835 - 2 * PI = -0.00318509737`

`0.00318509737 < 0.01749303564429283 * 1.3951473992034527e-15` would be a false statement, so I see where that would be a problem. 

(where `1.3951473992034527e-15` is 2 * PI * float epsilon)

Perhaps this should really be something like  ` < angle_increment - epsilson` rather than multiplication, to show that the difference between the 360 wrap around is less than an additional potential reading.

justinIRBT, 2020-09-16:
Maybe the most robust thing to do for subtle laser configuration issues is just have the "is_360_lidar" in slam_toolbox instead check for "needs_residual", effectively doing the math that Karto will do 
```
int residual = 1;
    if (GetIs360Laser()) {
      // residual is 0 by 360 lidar conventions
      residual = 0;
    }

    m_NumberOfRangeReadings = static_cast<kt_int32u>(math::Round((GetMaximumAngle() -
      GetMinimumAngle()) /
      GetAngularResolution()) + residual);
  }
```
and determine whether or not Karto will need the +1 residual to match the numbers of range scan readings.  Then setting the SetIs360Laser based on whether it will need the residual or not.

SteveMacenski, 2020-09-18:
>  So 0.00318509737 < (0.01749303564429283 - 1.3951473992034527e-15) should be true

Correct, but that's not what the code is right now. The PR linked added `0.00318509737 < (0.01749303564429283 * 1.3951473992034527e-15)`, which resolves to `0.003... < 2.44e-17`. That number is so hilariously small, I don't think there's much case where any error in `pi` would probably resolve well. 

The block you added is identical to that already in karto.h, what do you mean?

justinIRBT, 2020-09-18:
The checked in code is (increment - scaled epsilon) not *

https://github.com/SteveMacenski/slam_toolbox/blob/d9a8a8689251b96386e14812cf5a9b410b55145f/src/laser_utils.cpp#L109
I am guessing the referenced turtlebot code has been broken since Aug 3 when the threshold was changed from:
https://github.com/SteveMacenski/slam_toolbox/commit/6dcab997bb531d695fca401ee3f17b2298ef1f46
```
  -if ((std::fabs(scan_.angle_min + M_PI) < 1e-3) &&
  -   (std::fabs(scan_.angle_max - M_PI) < 1e-3))
  - {
  + if (std::fabs(scan_.angle_max - scan_.angle_min - 2.0 * M_PI) < 1e-1) {
```
Because its using angle_min 0  angle_max 6.28, it would have returned false with
```
(std::fabs(scan_.angle_min + M_PI) < 1e-3) &&
      (std::fabs(scan_.angle_max - M_PI) < 1e-3))
```
(leading to Karto expecting 360)

What I am proposing as a more robust solution is to do the exact same math as Karto in slam_toolbox's laser_utils instead of the threshold check so you know you are giving Karto what it wants, so effectively:
```
// Remove this
//bool is_360_lidar = false;
//  if (std::fabs(scan_.angle_max - scan_.angle_min - 2.0 * M_PI) < (scan_.angle_increment - //(std::numeric_limits<float>::epsilon() * 2.0f*M_PI))) {
//    is_360_lidar = true;
//  }

//  laser->SetIs360Laser(is_360_lidar);

// Replace with
uint32_t karto_expected = static_cast<uint32_t>(math::Round((scan_.angle_max - scan_.angle_min) / scan_.angle_increment);
if (karto_expected == scan_.ranges.size()) {
  laser->SetIs360Laser(true);
} else if ((karto_expected+1) == scan_.ranges.size()) {
  laser->SetIs360Laser(false);
} else {
  // ERROR message that Karto is not going to be happy with your range scan
}
```
Essentially Karto's is360Laser check isn't really about whether a laser is a 360 laser or not, its about how many range readings are expected, where what Karto calls 360 means the angle min/max reporting wraps around [-M_PI,M_PI] and the scan at -M_PI/M_PI isn't populated twice (once at min angle -M_PI and again at max angle M_PI).

SteveMacenski, 2020-09-18:
You are correct, I misread. Wow, I'm 0/2 on this ticket. Now I'm unclear about why this didn't work for @danwblackburn, I agree that the expression should evaluate to `true` for their message listed. @danwblackburn please log `std::fabs(scan_.angle_max - scan_.angle_min - 2.0 * M_PI) < (scan_.angle_increment - (std::numeric_limits<float>::epsilon() * 2.0f*M_PI)` inline and verify for us that it results in `false` just to sanity check us before we continue. 

mhm. I need to think about that a day. I would not call that a "more robust solution" but it is a way of hiding it. A more robust way would be a mathematical expression rather than "guess and check".

danwblackburn, 2020-09-18:
It is the other way around actually. The statement is correctly evaluating to ``true`` which sets ``laser->SetIs360Laser(true)`` and in Karto that means that ``residual = 0`` but the residual needs to be ``residual = 1`` because the laser scan actually contains 360 ranges instead of 359 ranges. I think that issue here comes from whether or not ``angle_max`` represents the angle value of the last range in the scan or the maximum angle value of the scan itself.

I kinda typed this out for my own understanding, but I'll keep it because it makes my explanation more clear. Imagine a super low resolution laser scan that can only scan every 90°. There are two ways to represent a full 360° scan. One way with only 4 ranges, with the last range being at 270°. In this case if you set angle max to the angle of the last range then ``angle_max=270``. However, it still represents a full 360° scan because one of the ranges is at 0°. I believe the convention is to just set ``angle_max=360`` anyway.
The second way gets rid of this quirk.  It creates 5 ranges, but the last range is the same as the first range. Now the angle of the last range is 360°. It adds an extra range, but makes more sense because ``angle_max`` and the angle of the last range and the angles of the scan are the same.

Karto is trying to determine which representation the LIDAR is using with the is360Laser check. I think that the convention around 360° LIDARs is to send the scan using the first representation so it removes the residual to account for the difference in the number of ranges. However, laser filters and simulators don't always send a 360° scan so they don't follow the first representation when creating 360° scans. When people want a 360° scan they input 0° to 360° instead of 0° to (360° - angle increment) and create the second representation by accident. The current ``std::fabs(scan_.angle_max - scan_.angle_min - 2.0 * M_PI) < (scan_.angle_increment - (std::numeric_limits<float>::epsilon() * 2.0f*M_PI)`` picks up that the scan is actually a 360° and tells Karto to not add the residual it normally would causing it to expect one less range than the simulated scan is giving it.

Technically this is not an error with your code as it only occurs when people aren't following the 360° laser convention. However, it's going to cause confusion and isn't readily apparent when setting up your laser scan for slam toolbox so I think that it's probably worth addressing. Especially if even Turtlebot is making this mistake. It may have been causing the RPLIDAR issues too because of how the drivers do weird things with angle compensation. I think that @justinIRBT 's solution works best as a coding solution. Or you could put a notice in documentation that this convention needs to be followed. Ironically, it maybe something wrong in Turtlebot's code :sweat_smile:

SteveMacenski, 2020-09-21:
> because the laser scan actually contains 360 ranges instead of 359 ranges

Ah, that's a problem. If you have a 360-deg lidar, then you can't have both the 0-th and 360-th measurement, because that would be the starting scan of the next rotation. You're correct on the interpretation of the max angle. For a non-360 lidar, its inclusive, but for 360, its exclusive (e.g. `[0, 360)`) because that last measurement now belongs to the next scan.

From the gazebo XML configuration, did you find a way to update it to react more appropriately? 

So, I don't really like @justinIRBT's solution, that feels like kicking the can. But, I think a compromise that would be a bit more robust that I could live with would be the following:
- Get the total angular range the way we do now (max - min)
- Check if this angular range is reasonably 360 (e.g. the difference is very close to 360, as opposed to 270 or any other config)
- In the case that its close to 360, we do the logic @justinIRBT suggests. 

What that buys us is the following:
- Without the check for "closeness" any arbitrary lidar could have +1/-1 measurements and completely hides the issue. This limits the gerrymandering to only our 360 case. 
- Silently hides these annoying 360 issues that people keep mis-configuring. Hopefully the last I'll have to hear of it. 

Thoughts?

justinIRBT, 2020-09-23:
Works for me

SteveMacenski, 2020-09-23:
@justinIRBT @danwblackburn please verify this resolves your issues 
Foxy: https://github.com/SteveMacenski/slam_toolbox/pull/296
Melodic: https://github.com/SteveMacenski/slam_toolbox/pull/298
Noetic: https://github.com/SteveMacenski/slam_toolbox/pull/299

Once you confirm, I'll merge

SteveMacenski, 2020-10-01:
Merging - confirmed by both users