SteveMacenski/slam_toolbox #294: Issues running with Gazebo turtlebot simulation

aslyansky-m, 2020-09-16:
**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- ROS Version
  - melodic
- Version or commit hash:
  - latest melodic-devel
- Laser unit:
  - turtlebot sim
  
#### Steps to reproduce issue
Follow tutorial:
https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/
Run simulation in gazebo, run slam_toolbox

#### Expected behavior
slam_toolbox running, **/map** is being updated  
#### Actual behavior
**/map** is not being updated

Hi,
Thanks for your work.
I assume that this issue is a result of some misunderstanding from my side.
Basically I am trying to get the algorithm running with turtlebot3_sim and I validated that **/scan** topic is valid.
When slam_toolbox is run:
`
roslaunch slam_toolbox online_sync.launch
`
There's no visible indication of error but **/map** topic stays silent.
What could explain this behavior?  Issues with tf frames?
Also it unfortunate that the repo is missing a demo rosbag as a minimal example.

Thanks,
Maxim

SteveMacenski, 2020-09-16:
Please ask on ROS Answers.

closed 2020-09-16 by SteveMacenski

aslyansky-m, 2020-09-17:
> Please ask on ROS Answers.

By that you mean that I run the algorithm correctly?
Could you share the commands to run the toolbox with turtlebot simulation?